Koenkk/zigbee2mqtt #21078: sonoff dongle-P passthrough into LXC after reflash not started in Version 1.35.1

arteck, 2024-01-28:
### What happened?

[image]

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230922

### Adapter

sonoff -P

### Setup

zigbee2mqtt in LXC

Nerivec, 2024-02-03:
It was working in 1.35.0, but not in 1.35.1 nor 1.35.2? Are you sure it wasn't 1.34.0 that worked before? 1.35.0>1.35.1 didn't change anything at the herdsman level (serialport). That error is outputted directly from [serialport](https://github.com/serialport/node-serialport).

Are you using the `/dev/serial/by-id` path for the port? If not, try that. 
Can you post your `configuration.yaml` otherwise?

arteck, 2024-02-06:
yes with 1.34.x works correct.. i use /dev/serial/by-id direct path to the sonoff..

```

homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1883
  user: xxxx
  password: yyyyyyyyyyyyyy
serial:
  port: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_b0d7a514acc9eb11be3b8b4f1d69213e-if00-port0
frontend: true
advanced:
  log_level: error
  pan_id: 6757
  ext_pan_id:
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
  channel: 15
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 1
    - 2
    - 5
    - 6
    - 8
    - 10
    - 12
    - 12
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /var/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  transmit_power: 10
  last_seen: ISO_8601_local
  legacy_api: false
  legacy_availability_payload: false
  log_output:
    - file
  cache_state: false
device_options:
  legacy: false
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
map_options:
  graphviz:
    colors:
      fill:
        enddevice: '#fff8ce'
        coordinator: '#e04e5d'
        router: '#4ea3e0'
      font:
        coordinator: '#ffffff'
        router: '#ffffff'
        enddevice: '#000000'
      line:
        active: '#009900'
        inactive: '#994444'
devices: devices.yaml
groups: groups.yaml
blocklist:
  - '0x04cf8cdf3c75b805'

```

closed 2024-09-04 by github-actions[bot]