Koenkk/zigbee2mqtt #21083: Wrong BRidge/Device Message after update to 1.35.1

Burki24, 2024-01-28:
### What happened?

After updating Z2M to V. 1.35.1 some devices don't send exposes, when I try to get them using 'zigbee2mqtt/bridge/devices'

After analyzing the problem, I found, that only devices using converter typ: 'exposes: (device, options) => {const exps: Exposes [] ....' aren't  able to send exposes within the asked Payload. All devices using the old form: 'exposes:[]' are sending there exposes well.

[image]
Converter build like this style doesn't send any exposes. 

[image]
Converter build like this are sending correctly their exposes.

Tried with different devices.

The payload from the affected device is attached on this issue.

### What did you expect to happen?

When publishing (using an extension) 'symcon/zigbee2mqtt/getDevice' with data '0xa4c13855e2d4363a' get the 'zh' payload as described, with payloads inside.

The extension also is attached to this issue.

### How to reproduce it (minimal and precise)

pair a device, which is using:''exposes: (device, options) => {const exps: Exposes [] ....'' in it's converter and try tu get the device-exposes using 'exposes: (device, options) => {const exps: Exposes [] ....' (as example Ts0203 -  _TZ3000_au1rjicn - tuya window-sensor)

### Zigbee2MQTT version

1.35.1-dev commit: a879382c

### Adapter firmware version

EZSP V8 - 7.0.2.0 build 406

### Adapter

sonoff-zigbee-3-0-usb-dongle-plus-e

### Setup

plain on Pi

### Debug log

[_TZ3000_au1rjicn.json](https://github.com/Koenkk/zigbee2mqtt/files/14075716/_TZ3000_au1rjicn.json)
[Debug_TS0203_TZ3000_au1rjicn.txt](https://github.com/Koenkk/zigbee2mqtt/files/14075729/Debug_TS0203_TZ3000_au1rjicn.txt)
[symcon-extension.txt](https://github.com/Koenkk/zigbee2mqtt/files/14075739/symcon-extension.txt)

Koenkk, 2024-01-28:
The issue is in your extension, not in the z2m code. You should call the [`device.exposes()`](https://github.com/Koenkk/zigbee2mqtt/blob/3c9620428ea00d759bd1001322b0ead55eca2285/lib/model/device.ts#L24) function to retrieve the exposes

Burki24, 2024-01-29:
> [`device.exposes()`](https://github.com/Koenkk/zigbee2mqtt/blob/3c9620428ea00d759bd1001322b0ead55eca2285/lib/model/device.ts#L24)

THANKS!!

Now it's running well.

Greetings
Burkhard

closed 2024-01-29 by Koenkk