Koenkk/zigbee2mqtt #21088: No converter available for ZG-204ZL

Pferdebockwurst, 2024-01-28:
### What happened?

I have several motion sensors ZG-204ZL. They are working, but I get the following error messages in the log:

`Debug 2024-01-28 02:38:07 No converter available for 'ZG-204ZL' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'

Debug 2024-01-28 02:38:08 No converter available for 'ZG-204ZL' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'

Debug 2024-01-28 02:38:14 No converter available for 'ZG-204ZL' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":0}'

Debug 2024-01-28 02:38:15 No conver ter available for 'ZG-204ZL' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'

Debug 2024-01-28 02:38:16 No converter available for 'ZG-204ZL' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":0}'

Debug 2024-01-28 02:38:17 No converter available for 'ZG-204ZL' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'

Debug 2024-01-28 02:39:22 No converter available for 'ZG-204ZL' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'`

### Zigbee2MQTT version

newest

### Adapter firmware version

newest

### Adapter

Slaesh's CC2652RB stick

### Setup

Add-on on Home Assistant

antonrez89, 2024-02-21:
Hi! I have the same warn logs.

BassT23, 2024-07-28:
+1 
sad, that I cant see, what trigger this issue :(

bzyx, 2024-11-29:
I'm having very similar problem with my presence sensor
```
[2024-11-29 14:52:43] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:52:48] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:52:53] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:52:58] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:53:03] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:53:08] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:53:13] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:53:18] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}'
[2024-11-29 14:53:23] debug: 	z2m: No converter available for 'ZG-205Z/A' with cluster 'msIlluminanceMeasurement' and type 'attributeReport' and data '{"measuredValue":1}
```

closed 2025-02-05 by github-actions[bot]