Koenkk/zigbee2mqtt #21092: Aqara Door Contact Sensor naming in Homeassistant is not correct

peteh, 2024-01-28:
### What happened?

Somehow the naming of Aqara Door Contact sensors in Homeassistant  is a bit messed up. 

[image]

With the new naming scheme changes in Homeassistant Entities are now named
[device name] [sensor]. 

I have a contact sensor on my Washing machine. Therefore I named the sensor in z2m to "Washing Machine Door". Unfortunately the sensor entity that shows if it's closed or not is named "Door" instead of being the main entity of the device and taking its name. 

Thus Homeassistant names the sensor entity to "Washing Machine Door Door". 

With the new naming scheme the name of the contact sensor entity should be set to None/null so that Homeassistant will name it to [device name] removing the duplicate "Door". 

Not sure how to better express that, I hope it's clear what the issue is. 

Z2M Version: [1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1) commit: [9eaaa0f](https://github.com/Koenkk/zigbee2mqtt/commit/9eaaa0f)

### What did you expect to happen?

When I name the device to "Washing Machine Door", the entity that shows open/closed should also be named to "Washing Machine Door" and not to "Washing Machine Door Door"

### How to reproduce it (minimal and precise)

Pair an Aquara door sensor and name it to "Something Door". 

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Setup

Z2M and HA in docker on a mini pc

peteh, 2024-02-07:
If this is not the right place, can someone point me to where I should report this to please?

PaulHam211, 2024-02-25:
Issue for me too. I have contact sensors on the windows, I name it 'Bedroom Window', in Home Assistant it gets named "Bedroom Window Door"

csszokolai, 2024-04-06:
Same issue here.

greendog99, 2024-07-11:
I agree the way this is designed makes naming things difficult / impossible. Only partial solution I've come up with is to manually rename every door (or window) entity to "Contact" so it shows up as "Bedroom Door Contact" instead of "Bedroom Door Door".

peteh, 2024-07-13:
I doubt it's in HA, it's more likely in z2m. 

The main entity should report null as name but it probably responds with "Door". If it would respond with null, it would get the name of the device. If it responds with "Door" it gets the name of the device + Door. 

So it should be fixable in z2m when it names the entities. However, this code is a bit messy for this device type and I haven't found the obvious fix yet.

peteh, 2024-09-02:
I tried again to debug it but I could not find out where the additional "Door" comes from. 
In https://github.com/Koenkk/zigbee-herdsman-converters lumi.sensor_magnet.aq2 looks fine to me.

scorpio1441, 2025-01-24:
Is there a solution to this?

[image]

peteh, 2025-05-16:
not stale! There is still no fix

peteh, 2025-07-16:
Unstale because still broken

encryptix, 2026-10-04:
Unstale. Still an issue? Not sure with #32752