Koenkk/zigbee2mqtt #21108: Tongou TO-Q-SY1-JZT - Issue after reset sum energy

timewolf, 2024-01-29:
### What happened?

i've reset sum energy of a Tongou TO-Q-SY1-JZT with endpoint 1 group 0x00 command 0 but now seems that the "sum energy" will never raise, still 0 after 2 days.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

6.10.3.0 

### Adapter

sonoff zb

### Setup

docker

customstudioyandexru, 2024-02-21:
Та же проблема.

nireins, 2024-03-01:
Same Problem here. How can i full reset this device?

customstudioyandexru, 2024-03-03:
The only thing that helped me was reconnecting the device.

git-greatguy, 2024-04-06:
I added this device to my network and got troubles with "energy" counter as well.
this counter seems to not be read correctly after a reset command is sent.

- Tried to reset from 2 different command.  should trigger same code behavior, i believe.
1st
`'{"reset":""}'`
2nd
`'{"command": {"cluster": 0,"command": 0,"payload": {}}}'`

Both case ended with a new announce from device and values reading.

`Info 2024-04-06 17:44:10MQTT publish: topic 'zigbee2mqtt/[INT] Alim EDF', payload '{"current":1.382,"energy":0.03,"indicator_mode":"off_on","last_seen":"2024-04-06T15:44:10.643Z","linkquality":72,"power":263,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'
Info 2024-04-06 17:44:11MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"[INT] Alim EDF","ieee_address":"0x385cfbfffed18343"},"type":"device_announce"}'
Info 2024-04-06 17:44:13MQTT publish: topic 'zigbee2mqtt/[INT] Alim EDF', payload '{"current":1.425,"energy":0.03,"indicator_mode":"off_on","last_seen":"2024-04-06T15:44:13.345Z","linkquality":102,"power":273,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":235}'`

After some minutes, energy still showing  value "0.03".

`Info 2024-04-06 17:46:15MQTT publish: topic 'zigbee2mqtt/[INT] Alim EDF', payload '{"current":1.388,"energy":0.03,"indicator_mode":"off_on","last_seen":"2024-04-06T15:46:15.932Z","linkquality":72,"power":262,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'`

"forcing" a read from dev console from attribute "currentSummDelivered" / cluster "semetering", make it updates again

now 0.02

`2024-04-06 17:48:02MQTT publish: topic 'zigbee2mqtt/[INT] Alim EDF', payload '{"current":1.365,"energy":0.02,"indicator_mode":"off_on","last_seen":"2024-04-06T15:48:02.612Z","linkquality":76,"power":258,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":233}'`

but still not updating automatically. after some minutes

`Info 2024-04-06 17:51:16MQTT publish: topic 'zigbee2mqtt/[INT] Alim EDF', payload '{"current":1.703,"energy":0.02,"indicator_mode":"off_on","last_seen":"2024-04-06T15:51:16.609Z","linkquality":72,"power":379,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'`

If i update again from dev console, new values reported

`Info 2024-04-06 17:51:53MQTT publish: topic 'zigbee2mqtt/[INT] Alim EDF', payload '{"current":1.703,"energy":0.05,"indicator_mode":"off_on","last_seen":"2024-04-06T15:51:53.803Z","linkquality":72,"power":379,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'
`

if i hit "reconfigure" button from z2m, energy counter go back to refresh values by itself like before reseting.

hope this info will help resolve this case ;)

thanks !

git-greatguy, 2024-04-06:
Found a workaround for this weird behavior for people interested in reset energy sum.
From node red, after sending "reset" command to the device, i send a configure command on z2m bridge "configure" topic to make it report counting again.

Topic is "zigbee2mqtt/bridge/configure" and parameter is your z2m device name.
example from my precedent post, name  is : "[INT] Alim EDF"

Enjoy ! :)

closed 2024-11-04 by github-actions[bot]