Koenkk/zigbee2mqtt #21115: [New device support]: Ikea Tretakt Switch 805.403.49

DL4DP, 2024-01-30:
### Link

https://csa-iot.org/csa_product/tretakt-smart-plug/

### Database entry

Device '0x8c6fb9fffe143e7b' with Zigbee model 'TRETAKT Smart plug' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### Comments

![Unbenannt](http[image]

[image]

[image]

[image]

[image]

### External defintion

```shell
TRETAKT Smart plug

Firmware date 20230206
Firmware version 2.4.4

IKEA of Sweden
Support status Unsupported

Device '0x8c6fb9fffe143e7b' with Zigbee model 'TRETAKT Smart plug' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

According to the instructions I couldn't do it, unfortunately!

Received message from unsupported device with Zigbee model 'TRETAKT Smart plug' and manufacturer name 'IKEA of Sweden'

Hello everyone,

Since today I am the proud owner of the Tretract socket from Ikea 805.403.49.

This is probably a brand new socket from Ikea, unfortunately zigbee2mqtt says it doesn't support it. Can you insert them?

The new socket cannot even be found in the Ikea online shop.

Thanks! For your help, I suspect the devices will soon replace the old socket and then will definitely be sold and used a lot.

I would be happy if the socket was supported!

Thank you Dirk!
```

Koenkk, 2024-01-31:
Please check if it works out-of-the-box with z2m 1.35.1

DL4DP, 2024-01-31:
Hello and thank you for the feedback.

I updated my zigbee2mqtt to 1.35.1, I was really a bit behind in the version. Unfortunately the sockets are still not installed correctly. I deleted them and retrained them. Unfortunately it still doesn't work properly. I don't know if it's due to the firmware version of the socket. Maybe someone else has another solution to my problem.

Thank you Dirk

Koenkk, 2024-02-01:
What do you get in the about screen with 1.35.1?

DL4DP, 2024-02-01:
Hello and thank you for the feedback, do you mean this information?

- Zigbee2MQTT-Version:  [1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1)  commit: [4b6f50da](https://github.com/Koenkk/zigbee2mqtt/commit/4b6f50da)
- Coordinator-Typ:  EZSP v12
- Coordinator-Version: 7.3.1.0 build 176
- Coordinator IEEE Adresse: 0xe0798dfffeaabf1d
- Frontend-Version: 0.6.151
- Zigbee-herdsman-converters version: 18.9.0
- Zigbee-herdsman version: 0.30.0

Koenkk, 2024-02-01:
I mean of the z2m frontend -> click on device -> about screen.

DL4DP, 2024-02-01:
- Gerätename: 0x8c6fb9fffe143e7b

- Zuletzt gesehen:
 N/A
- Verfügbarkeit: [Deaktiviert](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
- Geräte-Typ: Router
- Zigbee-Modell: TRETAKT Smart plug
- Zigbee-Hersteller: IKEA of Sweden
- Unterstützungsstatus: Nicht unterstützt [(How to add support)](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)
- IEEE-Adresse:  0x8c6fb9fffe143e7b
- Netzwerk-Adresse: 0x108D
- Firmware-Datum: 20230206
- Firmware-Version: 2.4.4
- Interview erfolgreich: Trifft zu

Koenkk, 2024-02-02:
What is shown under the exposes tab?

DL4DP, 2024-02-06:
Hello and thank you for the response, unfortunately I was away for a few days and am only now replying.

I took some pictures of the screen, I hope you find the information you are looking for.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Thank you for your help, support and time.

Dirk

Koenkk, 2024-02-06:
Does it work when changing the state under the exposes tab?

DL4DP, 2024-02-07:
[image]

[image]

[image]

[image]

The socket is also switched, and the "linkquality" values also change again and again.

Koenkk, 2024-02-07:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-07 by Koenkk