Koenkk/zigbee2mqtt #21116: [New device support]: TONGOU ‎TOQCB2 3P - Zigbee Smart Switch Measurement Circuit Breaker

tarikcosmo, 2024-01-30:
### Link

https://elcb.net/product/wifi-smart-switch-measurement-circuit-breaker-toqcb2-3p/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x943469fffe12cf81","nwkAddr":37101,"manufId":4098,"manufName":"_TZE204_tzreobvu","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_tzreobvu","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1706649549334,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
Please add this is 3Pole Zigbee Smart Switch Measurement Circuit Breaker.
Thank you!

### External defintion

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_tzreobvu',
    description: 'Automatically generated definition',
    extend: [],
};

module.exports = definition;
```

fox11012024, 2024-03-09:
[image]
I suggest you try it, that is
*quickly corrected from someone else’s example.
It worked for me, although there are also errors in the log.

[image]

tarikcosmo, 2024-03-13:
Thank you for help. Working fine.

closed 2024-10-11 by github-actions[bot]