Koenkk/zigbee2mqtt #21117: CC2652P Router not reporting link quality

DrTron, 2024-01-30:
### What happened?

I successfully flashed a CC2652P USB stick with the CC1352P2_CC2652P_launchpad_router_20221102.hex firmware.
Almost everything is working as expected, device could be joined, shows up on the network map and seems to be working.

However, it does not report link quality like my CC2530 routers do. When I change transmit power (the only adjustable setting), then both mqtt topics are populated. But no reports otherwise.

Since I use the link quality of the routers to determine if Z2M is working correctly, it is something that I could use.

### What did you expect to happen?

Router would start reporting link quality periodically to the appropriate Z2M topic

### How to reproduce it (minimal and precise)

Flash CC2652P with router firmware.

### Zigbee2MQTT version

1.35.0 commit: 3d8f0fc3

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB

### Setup

plain from Linux Debian 12 system on PC hardware

DrTron, 2024-02-01:
Edit: when I manually request the transmission power as documented here (https://www.zigbee2mqtt.io/devices/ti.router.html#custom%2520devices%2520(diy)-ti.router) by sending to the "get" topic, both transmission power and link quality topics get updated.

So technically I could send that to the device once a minute or so, but that's obviously a pretty dirty hack.

jerrm, 2024-02-01:
Try restarting z2m.

May or may not be relevevant, but I have several router flashed ZBDongle-P sticks.  z2m doesnpt see to correctly communicate with them until z2m has been restarted.

DrTron, 2024-02-01:
> Try restarting z2m.

Already did, no improvement.
The funny thing about it is that z2m does seem to communicate with it. Every time I manually request transmission power setting, both it and LQI are updated without delay.

I wonder if I need to configure reporting, but can't seem to find anything related to that there. Besides, every other device doesn't have an entry there, either.

DrTron, 2024-07-31:
Issue is still unresolved, commenting to remove stale label.

DrTron, 2024-12-31:
Issue still relevant, no fix found yet.

kizovinh, 2025-01-11:
Same issue for me

closed 2025-03-21 by github-actions[bot]