Koenkk/zigbee2mqtt #21123: Exception while calling fromZigbee converter

001Tycoon, 2024-01-31:
### What happened?

Hello, after the last update I often get this error message.
```
Zigbee2MQTT:error 31-01-2024 17:30:07: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'find')}
Zigbee2MQTT:debug 31-01-2024 17:30:07: TypeError: Cannot read properties of undefined (reading 'find')
```

### Zigbee2MQTT version

1.35.1-dev commit: 630b4ec

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Add-on on  Home Assistant x86-64

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14114630/log.txt)

mrmaximas, 2024-01-31:
same problem
20230922
1.35.1-dev commit: [630b4ec5](https://github.com/Koenkk/zigbee2mqtt/commit/630b4ec5)

Koenkk, 2024-01-31:
Pushed a fix! Let me know if it's ok now

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

mrmaximas, 2024-01-31:
> Pushed a fix! Let me know if it's ok now
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

1.35.1-dev commit: [1cee011c](https://github.com/Koenkk/zigbee2mqtt/commit/1cee011c) - same issue.

mrmaximas, 2024-01-31:
@Koenkk зкщифидн 1.35.1-dev commit: [0b59a51c](https://github.com/Koenkk/zigbee2mqtt/commit/0b59a51c) fix this issue

001Tycoon, 2024-02-01:
The error message no longer exists 1.35.1-dev commit: 0b59a51 thanks for all the work.
​

closed 2024-02-01 by 001Tycoon