Koenkk/zigbee2mqtt #21124: After restart Homeassistant zigbee2Mqtt needs a restart

huuub, 2024-01-31:
### What happened?

After a restart of homeassistant I always need to restart zigbee2mqtt to get everything working again.
I run mosquitto, zigbee2mqtt and homeassistant in docker in an LXC container in proxmox. I passed through the USB device.
I am adding all my config files.

After the restart only these kinds of messages appear in z2m (when homeassistant tries to send something):

Zigbee2MQTT:info  2024-01-31 16:58:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'lcd_glo_hal1' failed: 'Error: Command 0x00124b001ee9bf85/11 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (28))'","meta":{"friendly_name":"lcd_glo_hal1"},"type":"zigbee_publish_error"}'Zigbee2MQTT:error 2024-01-31 16:58:38: Publish 'set' 'state' to 'lcd_glo_hal1' failed: 'Error: Command 0x00124b001ee9bf85/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (28))'

[105.conf.txt](https://github.com/Koenkk/zigbee2mqtt/files/14114634/105.conf.txt)
[docker-compose.yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/14114635/docker-compose.yaml.txt)
[moquitto.conf.txt](https://github.com/Koenkk/zigbee2mqtt/files/14114636/moquitto.conf.txt)
[proxmox.105.conf.txt](https://github.com/Koenkk/zigbee2mqtt/files/14114637/proxmox.105.conf.txt)
[z2m.configuration.yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/14114639/z2m.configuration.yaml.txt)
  
Zigbee2MQTT versie
1.35.1-dev commit: [630b4ec](https://github.com/Koenkk/zigbee2mqtt/commit/630b4ec)
Coördinator type
ConBee2/RaspBee2
Coordinator revisie
0x26720700
Coordinator IEEE adres
0x00212effff0716f8
Versie frontend
0.6.152
Zigbee-herdsman-converters versie
18.25.0
Zigbee-herdsman versie
0.33.5

### How to reproduce it (minimal and precise)

restarting homeassistant container

### Zigbee2MQTT version

1.35.1-dev

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

docker container on proxmox on x86-64

Nerivec, 2024-01-31:
I don't know much about the deconz side, but do you have the logs immediately after startup (from `Preparing to start...`)? I'll see if I can spot something.

bckthomas, 2024-07-30:
Hi @huuub 
I had the same problem as you.
It seems to be resolved with the last Homeassistant version 2024.7.4

closed 2025-01-09 by github-actions[bot]

vsatmydynipnet, 2025-02-05:
It looks like this is still there with 2025 02. After restart it needs a manual Zigbee2mqtt restart.

KotMorderca, 2025-02-15:
2025.2.4 and same problem since few versions (few weeks? months? not sure)