Koenkk/zigbee2mqtt #21125: Clicked red `Check all` button for OTA updates, crash with a timeout

kucharskim, 2024-01-31:
### What happened?

Clicked to check for updates for all devices (`Check all` button) and found nodejs crashed like this:
```
Zigbee2MQTT:info  2024-01-31 17:58:39: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"niko-room-power-plug"},"error":"Failed to check if update available for 'niko-room-power-plug' (OTA: Device didn't respond to OTA request)","status":"error","transaction":"deyms-35"}'
Zigbee2MQTT:error 2024-01-31 17:58:39: Failed to check if update available for 'niko-room-power-plug' (OTA: Device didn't respond to OTA request)
Error: Timeout - 25515 - 1 - null - 25 - 1 after 60000ms
    at Timeout._onTimeout (/data/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

### What did you expect to happen?

Not crash.

### How to reproduce it (minimal and precise)

It's random and rare, but it does happen.

Coordinator type: EZSP v8
Coordinator revision: 6.10.3.0 build 297
Zigbee-herdsman-converters version: 18.9.0
Zigbee-herdsman version: 0.30.0
Frontend version: 0.6.151

### Zigbee2MQTT version

1.35.1 commit: 3c962042

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITEAD SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

PC Engines APU3, OpenBSD -current, node-18.19.0

Koenkk, 2024-01-31:
Does z2m crash completely?

kucharskim, 2024-01-31:
Yes, node.js exists.

kucharskim, 2024-01-31:
I actually hit another problem after deleting (unpairing a device):
```
Zigbee2MQTT:info  2024-01-31 19:19:09: Removing device '0xa4c1389250e329e0' (block: false, force: false)
Error: {"address":22109,"clusterId":32773,"sequence":167} after 10000ms
    at Timeout._onTimeout (/data/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

All those crashes look similar in nature. I request an action, action times out because device is offline or not that easily reachable and z2m/node.js crashes with a timeout.

Koenkk, 2024-01-31:
Please check again with tomorrows new release, if it still crashes, provide the herdsman debug log of it.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

kucharskim, 2024-01-31:
Sure, will wait for a release and check again.

Muxa1L, 2024-02-01:
@Koenkk seems like I have same problem - when trying to (re-)join some devices - z2mqtt just crashes. Using HASS addon.
`Zigbee2MQTT:info  2024-02-01 13:41:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410004afbb9","ieee_address":"0x54ef4410004afbb9"},"type":"device_leave"}'
2024-02-01T10:41:50.799Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,2,0): 02d3b1a9112a15b259954a25aa1593499c6a8b6c615c9874f8726389fc7e3da20f7e
2024-02-01T10:41:50.799Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2024-02-01T10:41:50.800Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2024-02-01T10:41:50.800Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 3
2024-02-01T10:41:50.800Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9190014500000000010000004001000024acc78c92ffff05b40000000002
2024-02-01T10:41:50.802Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":0,"sequence":36,"clusterId":1,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":320},"lastHopLqi":172,"lastHopRssi":-57,"sender":37516,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[180,0,0,0,0]}}
2024-02-01T10:41:50.803Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":0,"sequence":36,"clusterId":1,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":320},"lqi":172,"rssi":-57,"sender":37516,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[180,0,0,0,0]}}
2024-02-01T10:41:50.804Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,2,0): 12d3b1a9702a5960e06f0025ba117d1d9d4ebf6dd57b7e
2024-02-01T10:41:50.804Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2024-02-01T10:41:50.805Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2024-02-01T10:41:50.806Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 3
2024-02-01T10:41:50.806Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 91900124004cd2b9fb4a001044ef54010098c6
2024-02-01T10:41:50.807Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":53836,"newNodeEui64":{"type":"Buffer","data":[84,239,68,16,0,74,251,185]},"status":1,"policyDecision":0,"parentOfNewNodeId":50840}
2024-02-01T10:41:50.807Z zigbee-herdsman:adapter:ezsp:ezsp ==> findKeyTableEntry: {"address":{"type":"Buffer","data":[84,239,68,16,0,74,251,185]},"linkKey":true}
2024-02-01T10:41:50.808Z zigbee-herdsman:adapter:ezsp:driv handleNodeJoined: change ManufacturerCode for ieee 54ef4410004afbb9 to 4447
2024-02-01T10:41:50.808Z zigbee-herdsman:adapter:ezsp:ezsp ==> setManufacturerCode: {"code":4447}
2024-02-01T10:41:50.809Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 53836 54ef4410004afbb9
2024-02-01T10:41:50.809Z zigbee-herdsman:controller:log Device '0x54ef4410004afbb9' joined
2024-02-01T10:41:50.810Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,2,0): 22d3b1a9702a5960e06f0025ba117d1d9d4ebf6d688e7e
2024-02-01T10:41:50.810Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2024-02-01T10:41:50.814Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2024-02-01T10:41:50.814Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 3
2024-02-01T10:41:50.814Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 91900124004cd2b9fb4a001044ef54010098c6
2024-02-01T10:41:50.815Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":53836,"newNodeEui64":{"type":"Buffer","data":[84,239,68,16,0,74,251,185]},"status":1,"policyDecision":0,"parentOfNewNodeId":50840}
2024-02-01T10:41:50.815Z zigbee-herdsman:adapter:ezsp:ezsp ==> findKeyTableEntry: {"address":{"type":"Buffer","data":[84,239,68,16,0,74,251,185]},"linkKey":true}
2024-02-01T10:41:50.816Z zigbee-herdsman:adapter:ezsp:driv handleNodeJoined: change ManufacturerCode for ieee 54ef4410004afbb9 to 4447
2024-02-01T10:41:50.817Z zigbee-herdsman:adapter:ezsp:ezsp ==> setManufacturerCode: {"code":4447}
2024-02-01T10:41:50.817Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 53836 54ef4410004afbb9
2024-02-01T10:41:50.818Z zigbee-herdsman:controller:log Device '0x54ef4410004afbb9' joined
2024-02-01T10:41:50.818Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,2,1): 7ad3b1a9702abb0be06f0025ba117d1d9e4da3899eba7e
2024-02-01T10:41:50.819Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2024-02-01T10:41:50.819Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2024-02-01T10:41:50.819Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 3
2024-02-01T10:41:50.820Z zigbee-herdsman:adapter:ezsp:uart Skipping the packet as repeated (0)
2024-02-01T10:41:50.820Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,2,1): 0ad3b1a9112a15b259954a25aa1593499c6a8b6c615c9874f8726389fc7e3d93c37e
2024-02-01T10:41:50.820Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2024-02-01T10:41:50.821Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2024-02-01T10:41:50.821Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 3
2024-02-01T10:41:50.821Z zigbee-herdsman:adapter:ezsp:uart Skipping the packet as repeated (1)
2024-02-01T10:41:50.822Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,2,1): 1ad3b1a9702a5960e06f0025ba117d1d9d4ebf6d68c27e
2024-02-01T10:41:50.822Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2024-02-01T10:41:50.822Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
[... cut ...]
2024-02-01T10:41:58.171Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x64: {"_cls_":"setMulticastTableEntry","_id_":100,"_isRequest_":false,"status":0}
2024-02-01T10:41:58.171Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
2024-02-01T10:41:58.172Z zigbee-herdsman:adapter:ezsp:cast Set MulticastTableEntry #7 for 2948 multicast id: 0
Error: {"address":53836,"clusterId":32773,"sequence":31} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)`

kucharskim, 2024-02-03:
Upgraded to Zigbee2MQTT version [1.35.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.2) commit: [fdc9f7d4](https://github.com/Koenkk/zigbee2mqtt/commit/fdc9f7d4) and I'm currently testing.

kucharskim, 2024-02-10:
With Zigbee2MQTT version [1.35.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.2) commit: [fdc9f7d4](https://github.com/Koenkk/zigbee2mqtt/commit/fdc9f7d4) this works very stable, zero crash so far.

closed 2024-09-08 by github-actions[bot]