Koenkk/zigbee2mqtt #21131: Zigbee2MQTT not updating Zigbee-Herdsman version

GitCharlieHub, 2024-01-31:
### What happened?

Latest release of Zigbee-Herdsman-converters 18.25.1 fixes a firmware OTA with my device. On looking within Zigbee2MQTT, I’m still on version 18.9.1.

It seems this doesn’t update automatically as the version I have is more than 2 months old.

### What did you expect to happen?

Either that zigbee2mqtt updates Zigbee-Herdsman-converters automatically in the background after a reboot, or that there is a button to force update.

### How to reproduce it (minimal and precise)

Reboot home assistant.
Go to Zigbe2mqtt -> settings -> about
Zigbee-herdsman-converters version is 18.9.0 and not latest available on GitHub as a release

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff USB dongle

### Setup

HomeAssistant add on - RPI4

mrmaximas, 2024-01-31:
Please switch to dev branch https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html or wait for release

Nerivec, 2024-01-31:
`zigbee-herdsman` & `zigbee-herdsman-converters` update whenever Zigbee2MQTT gets a new release.
When Z2M got [its last release (1.35.1)](https://github.com/Koenkk/zigbee2mqtt/blob/4b6f50daf3d850b6c6a8c51a805da2cd42e3e643/package.json#L3) , it was with the fixed versions listed [here](https://github.com/Koenkk/zigbee2mqtt/blob/4b6f50daf3d850b6c6a8c51a805da2cd42e3e643/package.json#L64).

This, among other things, allows daily changes to be tested in the DEV version by users who are comfortable dealing with potential bugs until a new release (1st of every month) publishes those changes along with the new Z2M version.
If you want faster updates, you'd have to [change to the DEV version](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), but beware of the aforementioned caveats.

Although you're in luck since tomorrow is the 1st, you should get an update prompt then (depending on your timezone of course).

GitCharlieHub, 2024-01-31:
@Nerivec & @mrmaximas thanks for the quick replies. 

Clear - thought the releases would be separate but makes sense to come with the core zigbee2mqtt release.

closed 2024-01-31 by GitCharlieHub

mrmaximas, 2024-01-31:
> `zigbee-herdsman` & `zigbee-herdsman-converters` update whenever Zigbee2MQTT gets a new release. When Z2M got [its last release [...]

I'm already on the 1st of February, but still no release ;)