Koenkk/zigbee2mqtt #21143: Automation using the TuYa SH-SC07 button does not work

Fierys, 2024-02-01:
### What happened?

After updating Zigbee2MQTT from 1.35.1 to 1.35.2, automation using TuYa SH-SC07 buttons stopped working.
Zigbee2MQTT has buttons, clicks are processed. I tried to rebind, HA restarted.
After rolling back to Zigbee2MQTT 1.35.1 everything works fine.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

20210319

### Adapter

dongle СC2652P. zStack3x0

### Setup

HAOS on x86-64 hardware (NUC)

Jonno12345, 2024-02-02:
Similarly my TS0041 automations have stopped working. I think this is related as both our devices should have actions of 'single', 'double' or 'hold'. Since this update mine is reporting '1_single', as per multi button devices. I'm not sure if this was an intentional breaking change, simple enough to fix my automations but without knowing if this'll be reverted I don't know if I should..!

coubi64, 2024-02-02:
Same here with Aqara WXKG11LM. Automatisation on "single action" for example (i didn't tried others) is not working anymore since update of Zigbee2MQTT to 1.35.2

Fierys, 2024-02-02:
[image]
[image]

Chartreusito, 2024-02-02:
Same problem here.
Most likely related to https://github.com/Koenkk/zigbee2mqtt/issues/21158

Koenkk, 2024-02-03:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/21158

closed 2024-02-03 by Koenkk

mysleinic, 2024-02-03:
Folks,

I have the same exact trouble - problem is that when using "Device" as a trigger it no longer discovers "single", "double", etc. events on the list of the triggers... In my case just a battery state or button mode. It was working for some of my buttons for some time after upgrade and gradually all buttons gone! What is f... -> really annoying.

However - working workaround is to use "State" as trigger:
[image]

Try it for yourself - watchout as action entity state names are case sensitive.

ps. I'm using Tuya Zigbee buttons via zigbee2mqtt

Chartreusito, 2024-02-03:
@mysleinic please follow up in the [linked issue](https://github.com/Koenkk/zigbee2mqtt/issues/21158)