Koenkk/zigbee2mqtt #21144: v1.35.2 build failing on homeassistant.ts

syphernl, 2024-02-01:
### What happened?

Building v1.35.2 is failing:

```
zigbee2mqtt:~$ npm run build                                               
> zigbee2mqtt@1.35.2 build                                                 > tsc && node index.js writehash
lib/extension/homeassistant.ts:1063:25 - error TS2339: Property 'category' does not exist on type 'Expose'.                                             Property 'category' does not exist on type 'Numeric'.
1063         if (firstExpose.category === 'config') {
                             ~~~~~~~~                                      
lib/extension/homeassistant.ts:1065:32 - error TS2339: Property 'category' does not exist on type 'Expose'.
  Property 'category' does not exist on type 'Numeric'.                    
1065         } else if (firstExpose.category === 'diagnostic') {                                               ~~~~~~~~

Found 2 errors in the same file, starting at: lib/extension/homeassistant.ts:1063
```

The v1.35.1 branch is working just fine without error.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

-

### Adapter

-

### Setup

Plain

Koenkk, 2024-02-01:
You probably forgot to do a `npm ci` after updating

syphernl, 2024-02-01:
> You probably forgot to do a `npm ci` after updating

Ah yes! I indeed forgot to run that.
Thanks for your quick reply :-)

closed 2024-02-01 by syphernl