Koenkk/zigbee2mqtt #21151: Lixee: Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'activeEnergyOutD01'

merlinpimpim, 2024-02-01:
### What happened?

_Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'activeEnergyOutD01'_ very regularly in  the log since 1.35.2-1

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

7.3.1.0 build 0

### Adapter

Skyconnect

### Setup

HA supervised, Khadas

V1nc3, 2024-02-02:
Same issue here since this afternoon : Lixee data not more updated in HA thrue zigbee2mqtt.
In the log, a lot of error like : Failed to read zigbee attributes: Error: Read 0x00158d0005d246c5/1 seMetering(["activeRegisterTierDelivered","meterSerialNumber"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (128))

titom43, 2024-02-04:
Same thing here. Zigbee payload not update since Z2M 1.35.2-1 upgrade...

2024-02-04 20:31:20Failed to configure 'Compteur Linky', attempt 1 (Error: ConfigureReporting 0x00158d0006a40061/1 seMetering([{"attribute":"currentTier1SummDelivered","minimumReportInterval":60,"maximumReportInterval":900,"reportableChange":100}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (11)) at DeconzAdapter.sendZclFrameToEndpoint ...

renaiku, 2024-02-05:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6866 could this be linked ? a lot has been renamed from ernerFy to enerGy [mildis](https://github.com/mildis) @mildis

renaiku, 2024-02-05:
indeed, the state of the device has this:

```
"active_enerfy_out_d01": 20234.584,
"active_enerfy_out_d02": 22505.523,
...
"active_energy_out_d01": null,
"active_energy_out_d02": null,
    ```

mildis, 2024-02-05:
It has already been corrected and merged with https://github.com/Koenkk/zigbee-herdsman/pull/903
zigbee-herdsman and zigbee-herdsman-converters live side-by-side but depends on each other.

renaiku, 2024-02-05:
Ok thanks. Do you have an idea why we keep having enerFy in the device status reported ?

romarysonrier, 2024-02-05:
add the same issue , just removed from tic_command_whitelist param the following items :
- EASD01
- EASD02
- EASD03
- EASD04
And the issue has disapeared.

renaiku, 2024-02-05:
My whitelist is empty. Anyway, my zlinky_tic is reporting null for the two new enerGy and keep filling enerFy

mildis, 2024-02-05:
If you want to correct the issue manually before the next release, you can try the following (adjust to your setup).
This connect to the z2m container, do the replacement in the required files, check nothing is left out, repack the asset.
The index-nnnnn.js file name might differ on your setup.
```shell
$ docker exec -it zigbee2mqtt /bin/sh
/app # sed -i -e 's/activeEnerfyOut/activeEnergyOut/g' node_modules/zigbee-herdsman/dist/zcl/definition/cluster.js
/app # sed -i -e 's/activeEnerfyOut/activeEnergyOut/g' node_modules/zigbee2mqtt-frontend/dist/assets/index-83109297.js
/app # grep -Rl activeEnerfyOut *
/app # cd node_modules/zigbee2mqtt-frontend/dist/assets/
/app/node_modules/zigbee2mqtt-frontend/dist/assets # rm index-83109297.js.gz 
/app/node_modules/zigbee2mqtt-frontend/dist/assets # gzip -k index-83109297.js 
``` 
then restart your z2m container.

mildis, 2024-02-08:
This issue can be closed with the release of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3

closed 2024-09-06 by github-actions[bot]