Koenkk/zigbee2mqtt #21153: Aqara switch Z1 The middle switch of the triple rocker switch cannot be used.

gresgtehrtsher, 2024-02-02:
### What happened?

I found that the code for the triple rocker switch is almost the same as the two-key switch. Did you forget to write the code for the middle key switch after copying the triple rocker switch code?

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

latest

### Adapter

docker

### Setup

docker

waterlou, 2024-02-04:
An update is submit and should be available in next update.  The reason why old external converter not working is the major name change from Xiaomi to lumi in the recent update.  In the meantime, you can use the following external converter.

https://gist.github.com/waterlou/88d4f1e15a31db9edfcefb0dd4e71304

leeaash, 2024-02-05:
I have made a lot of helper to change switch type to light on old external converters, they use button_1,2,3,4, I have tried this update, but I saw that all those helper were gone after I upgraded z2m to 1.35-2
Is there a way to keep my setting? or I have to re-change them one by one.

closed 2024-09-04 by github-actions[bot]