Koenkk/zigbee2mqtt #21154: Aqara Thermostat E1 external sensor wrong temperature

gbcatrinoiu, 2024-02-02:
### What happened?

Aqara Thermostat E1 set to external sensor shows the wrong temperature. My temperature of my Tuya sensor is 17 and Aqara sees as it 27.

In the Z2M dashboard I also get this error:

`Error 2024-02-02 07:55:08No converter available for 'sensor_temp' (17)`

The **(17)** is the actual is the actual temperature.

### What did you expect to happen?

The TRV see the temperature correct

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

7.3.1.0 build 176

### Adapter

home assistant skyconnect

### Setup

Add-on on Home Assistant on Pi 4

### Debug log

Error 2024-02-02 08:03:32No converter available for 'sensor_temp' (17)

mionut, 2024-02-02:
I looked in the documentation and it seems they changed the name from "sensor_temp" to "external_temperature_input". In my automation in node_red i changed to the new name and it works.

balubone, 2024-02-02:
> I looked in the documentation and it seems they changed the name from "sensor_temp" to "external_temperature_input". In my automation in node_red i changed to the new name and it works.

Thank you very much mionut! You saved my day. Thank you really.

silfax, 2024-02-03:
Wouldn't it be nice to have these changes listed somewhere? I wouldn't mind doing some reading if necessary, but many/most/some of these changes are not mentioned anywhere and we have to discover them the hard way.

closed 2024-09-01 by github-actions[bot]