Koenkk/zigbee2mqtt #21158: TuYa IH-K663 cheating statuses from single to 1_single

DivanX10, 2024-02-02:
### What happened?

The TuYa IH-K663 button in Zigbee2MQTT version 1.35.2-1 had the statuses single, double

Documentation of the TuYa IH-K663 device in Zigbee2MQTT https://www.zigbee2mqtt.io/devices/IH-K663.html

```
Action (enum)
Triggered action (e.g. a button click). Value can be found in the published state on the action property. It's not possible to read (/get) or write (/set) this value. The possible values are: single, double
```

In version Zigbee2MQTT 1.35.2, for some reason, single, double changed to 1_single, 2_double in Action. Is it possible to return it as it was? Why did this button have its statuses renamed to 1_single, 2_double? If not, please correct the description in the documentation, it is misleading about the correct status when clicking

[image]

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

20230507

### Adapter

CC2652P

### Setup

Home Assistant Add-on: Zigbee2MQTT x86-64

skrepkaq, 2024-02-02:
I have the same issue with [TS0041](https://www.zigbee2mqtt.io/devices/TS0041.html#tuya-ts0041) and [MINI-ZSB](https://www.zigbee2mqtt.io/devices/MINI-ZSB.html#tuya-mini-zsb)
[image]
[image]

zvirja, 2024-02-02:
Same issue for https://www.zigbee2mqtt.io/devices/MINI-ZSB.html#tuya-mini-zsb

martinclaus1, 2024-02-02:
I have the same issue with [HG08164](https://www.zigbee2mqtt.io/devices/HG08164.html).

Koenkk, 2024-02-03:
Could you provide the data/database.db entry of your device?

mysleinic, 2024-02-03:
Folks,

I have the same exact trouble - problem is that when using "Device" as a trigger it no longer discovers "single", "double", etc. events on the list of the triggers... In my case just a battery state or button mode. It was working for some of my buttons for some time after upgrade and gradually all buttons gone! What is f... -> really annoying.

However - working workaround is to use "State" as trigger:
[image]

Try it for yourself - watchout as action entity state names are case sensitive.

ps. I'm using Tuya Zigbee buttons via zigbee2mqtt

skrepkaq, 2024-02-03:
Here is [TS0041](https://www.zigbee2mqtt.io/devices/TS0041.html#tuya-ts0041)

```json
{"id":4,"type":"EndDevice","ieeeAddr":"0x70ac08fffea1aa89","nwkAddr":5833,"manufId":4098,"manufName":"_TZ3000_axpdxqgu","powerSource":"Battery","modelId":"TS0041","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":1,"65534":0,"modelId":"TS0041","manufacturerName":"_TZ3000_axpdxqgu","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":192,"batteryVoltage":29}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1706959504243,"defaultSendRequestWhen":"immediate"}
```

Chartreusito, 2024-02-03:
> Could you provide the data/database.db entry of your device?

Funny enough i could not help you with that right now, my database backup seems truncated at id 47 :/

iqubik, 2024-02-03:
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/14151484/database.db.zip)

Koenkk, 2024-02-03:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-03 by Koenkk

iqubik, 2024-02-06:
1.35.3
thats not fixed
[image]

zvirja, 2024-02-06:
Can confirm that its NOT fixed MINI-ZSB

Nisbo, 2024-02-07:
Still not fixed for MINI-ZSB
Zigbee-Modell    TS0041
Zigbee-Hersteller    _TZ3000_qgwcxxws

DivanX10, 2024-02-07:
I don’t understand, will it now be `1_single` instead of `single` or has it not been fixed again? It should be `single`, not `1_single`.

Koenkk, 2024-02-07:
`"modelId":"TS0041","epList":[1,2,3,4]`, this is interesting, what I did was to not prefix the endpoint when it has 1 endpoint, somehow this device has 4. Pushed a fix.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Koenkk, 2024-02-08:
> I've got a TS0044, a four button Tuya remote. Single press of button number one sends 1_single, double press of button 1 sends 1_double. Pressing button number 2 one time sends 2_single etc.

That's the expected behaviour, right?

Calimerorulez, 2024-02-08:
Well, then I'm literally in the dark, because after upgrading to the most recent version of Z2M, Home Assistant does not see the actions of these remotes anymore in my existing automations ("Unknown trigger").

I can't choose the button actions anymore in HA, even if I have pressed the buttons on the remote to make them try to show up in HA.

I though it was related with this issue, because it only involves my two Tuya remotes.

Koenkk, 2024-02-10:
@Calimerorulez what actions did it send before?

APMXGH, 2024-02-13:
For TS0044 I do experience also weird behaviour since the update to version 1.35.3-1 as @Calimerorulez also describes.
Behaviour of the button sometimes works, sometimes the action event is not recognized.

If it is recognized it sends:

**z2m log:**
info 2024-02-13 10:24:49MQTT publish: topic 'zigbee2mqtt/0x3425b4fffef4dfae', payload '{"action":"3_single","battery":100,"linkquality":191,"voltage":3000}'
info 2024-02-13 10:24:49MQTT publish: topic 'zigbee2mqtt/0x3425b4fffef4dfae', payload '{"action":"","battery":100,"linkquality":191,"voltage":3000}'
info 2024-02-13 10:24:49MQTT publish: topic 'zigbee2mqtt/0x3425b4fffef4dfae/action', payload '3_single'

**Automation trace:**
platform: device
domain: mqtt
device_id: xxxxxxx
type: action
subtype: 3_single
discovery_id: 0x3425b4fffef4dfae action_3_single

Records also show the same log from button 1, however the automation does not get triggered anymore.
No automation trace is available after the click.

**z2m log**
info 2024-02-13 10:26:29MQTT publish: topic 'zigbee2mqtt/0x3425b4fffef4dfae', payload '{"action":"1_single","battery":100,"linkquality":167,"voltage":3000}'
info 2024-02-13 10:26:29MQTT publish: topic 'zigbee2mqtt/0x3425b4fffef4dfae', payload '{"action":"","battery":100,"linkquality":167,"voltage":3000}'
info 2024-02-13 10:26:29MQTT publish: topic 'zigbee2mqtt/0x3425b4fffef4dfae/action', payload '1_single'

Chartreusito, 2024-02-16:
> @Calimerorulez what actions did it send before?

@Koenkk It did send "single" and "double", at least for _TZ3000_mrpevh8p [SH-SC07](https://www.zigbee2mqtt.io/devices/SH-SC07.html#tuya-sh-sc07)

Chartreusito, 2024-02-16:
Just realized your fix is named "Fix single_1 instead of single action for various TuYa device", but here devices sends 1_single not single_1

mysleinic, 2024-02-17:
Guys, I understood you trying to fix this issue permamently and for wide
range of buttoms.. to make you old automation works, as they were working..
however (I hope) I posted under this issue quick work around.. Use state
change trigger.. then you can configure any specyfic states/antions your
random cheap Tuya button sends :) At least it works perfectly fine for all
of my cheap buttons and I have few of them :)

Koenkk, 2024-02-18:
@Chartreusito

- Can you provide the debug log from starting z2m until these actions are send?
- Can you provide the data/database.db entry of your device?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Chartreusito, 2024-02-18:
I'm using HAOS with z2m addon.
My database.db.backup is weirdly truncated (i'm gonna open an issue).
The device is this one : interrupteur_tuya_01 (0xa4c1388aa2adf49f): SH-SC07 - TuYa Button scene switch (EndDevice)
Here is my startup log up until i single and double click the device :
[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/14323280/log.zip)

Koenkk, 2024-02-18:
@Chartreusito the fix is only available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) yet

Chartreusito, 2024-02-18:
Sorry, i can confirm it's fixed on the dev branch, thanks.

NeoFax75, 2024-03-09:
Still have issues as only single and double work.  Hold does not register properly.

zulufoxtrot, 2024-08-23:
I have the same issue (or at least the same as https://github.com/Koenkk/zigbee2mqtt/issues/21205) on Z2M 1.39.1.

```
Zigbee2MQTT version
    [1.39.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    EZSP v8

Coordinator revision
    6.10.3.0 build 297

Coordinator IEEE Address
    0xe8e07efffeef260f

Frontend version
    0.7.4

zigbee-herdsman-converters version
    19.72.0

zigbee-herdsman version
    0.55.3
```
[image]

Ramss3s, 2024-09-08:
still dosen't wok ... SH-SC07

janfuhlrott, 2024-10-07:
Zigbee-Modell   TS0043
Zigbee-Hersteller    _TZ3000_kkahwiyu

Still same Problem with this device...

SvetlovSS, 2024-10-27:
> I have the same issue (or at least the same as #21205) on Z2M 1.39.1.
> 
> ```
> Zigbee2MQTT version
>     [1.39.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
> 
> Coordinator type
>     EZSP v8
> 
> Coordinator revision
>     6.10.3.0 build 297
> 
> Coordinator IEEE Address
>     0xe8e07efffeef260f
> 
> Frontend version
>     0.7.4
> 
> zigbee-herdsman-converters version
>     19.72.0
> 
> zigbee-herdsman version
>     0.55.3
> ```
> 
> <img alt="Screenshot 2024-08-23 at 19 18 26" width="331" src="https://private-user-images.githubusercontent.com/1007033/361012565-2c4aeaf6-0457-4e30-952d-a046b71c2541.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAwMjM4NzUsIm5iZiI6MTczMDAyMzU3NSwicGF0aCI6Ii8xMDA3MDMzLzM2MTAxMjU2NS0yYzRhZWFmNi0wNDU3LTRlMzAtOTUyZC1hMDQ2YjcxYzI1NDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjdUMTAwNjE1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2I1MWJjNzM3ZGZjYzNiYTI1ZWUzZTdkYTVhY2ExZDQzMjEwOTNlYjBiMjYxNGQ1ZWUyZjI2NjYxMTMwYjdmYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.hNfVNjRR9Dub4D2mg6lucx_hrheBz_ZZ-9sKFqj8TLg">

the same problem