Koenkk/zigbee2mqtt #21159: Zigbee2MQTT gets unresponsive out of the blue

webnar, 2024-02-02:
### What happened?

My network has been running stable for some years and suddenly for a few days, it gets unresponsive multiple times a day.
The front end is still working but the backend generates a lot of errors.has been

### What did you expect to happen?

I was able to control my lights etc.

### How to reproduce it (minimal and precise)

just wait

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20220219

### Adapter

Sonoff P

### Setup

Z2M in docker on a X86-64 NUC

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14138838/log.txt)

igalita, 2024-02-02:
I confirm I have the exact same problem !!!

webnar, 2024-02-02:
I went back from docker image 1.35.1 to image 1.33.2 and the problem still occurs, it looks like it is less often with 1.33.2.

webnar, 2024-02-02:
I just bought a new stick and updated that one to coordinator version 20230507 and replaced the current one (don't know if the 2-year-old coordinator version can be of influence on this).

webnar, 2024-02-08:
Seems to work now with the updated stick

closed 2024-02-08 by webnar