Koenkk/zigbee2mqtt #21160: OTA Firmware Update keeps failing

andyblac, 2024-02-02:
### What happened?

started OTA firmware update, but it keeps failing to complete, and restarts to try again.

### What did you expect to happen?

firmware to complete and update.

### How to reproduce it (minimal and precise)

just start a Firmware update of device:

Zigbee Model: DoubleSocket50AU
Zigbee Manufacturer: Aurora
Description: Double smart socket UK

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee Dongle-P

### Setup

add-on on Home Assistant OS on NUC

### Debug log

Zigbee2MQTT:info  2024-02-02 10:48:33: MQTT publish: topic 'zigbee2mqtt/Lounge Sockets 2', payload '{"brightness":50,"last_seen":"2024-02-02T10:48:33+00:00","linkquality":120,"power_left":0,"power_right":0,"state_left":"OFF","state_right":"OFF","update":{"installed_version":20,"latest_version":22,"progress":2.98,"remaining":33295,"state":"updating"},"update_available":null}'
Zigbee2MQTT:debug 2024-02-02 10:48:34: OTA: Image block response failed (CommandResponse 0x00158d000256fa88/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4636,"imageType":5129,"fileVersion":22,"fileOffset":5854,"dataSize":50,"data":{"type":"Buffer","data":[32,71,33,0,56,224,19,13,128,216,224,4,192,140,96,65,8,0,32,140,224,14,8,0,32,216,128,47,64,152,224,5,97,128,0,0,168,140,224,54,8,0,32,140,0,38,8,0,32,140]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:info  2024-02-02 10:48:35: MQTT publish: topic 'zigbee2mqtt/Lounge Sockets 2', payload '{"brightness":50,"last_seen":"2024-02-02T10:48:35+00:00","linkquality":120,"power_left":0,"power_right":0,"state_left":"OFF","state_right":"OFF","update":{"installed_version":20,"latest_version":22,"progress":2.98,"remaining":33295,"state":"updating"},"update_available":null}'
Zigbee2MQTT:debug 2024-02-02 10:48:35: OTA: Request offsets: fileOffset=5854 pageOffset=0 dataSize=60
Zigbee2MQTT:debug 2024-02-02 10:48:35: OTA: Payload offsets: start=5854 end=5904 dataSize=50

closed 2024-09-01 by github-actions[bot]

KidA001, 2024-10-27:
This has been an issue for years, multiple GH Issues but no responses

andyblac, 2024-10-27:
I got it to update by unplugging devices from both sockets, and then try update, it seems that some devices stop the update from completing no idea why.

SCCMOG, 2024-11-03:
Having a nightmare with this at the minute also.
Bought one and it had no issues during my tests, so bought another 4 for the kitchen and now I can't update any. All are timing out from 29% to 50%. Driving me nuts as other devices, not even routers, are updating fine.
Any more information would be a great help.