Koenkk/zigbee2mqtt #21162: Firmware update 1.35.2 breaks link with zigbee devices

Malvadis, 2024-02-02:
### What happened?

Update 1.35.2 was applied by auto update last night, I was previously running on version 1.35.1.
After the update the link to all devices was lost.

### What did you expect to happen?

I expected the update to apply and all device joins to be working

### How to reproduce it (minimal and precise)

Update to version 1.35.2 from 1.35.1, I use the Conbee 2 dongel.
HomeAssistant is running on a promox VM, the conbee stick is directly passed to the VM using USB redirection.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

Unsure

### Adapter

Conbee 2

### Setup

Addon on home assistant OS, running Home assistant OS on a virtual machine, Proxmox hypervisor, conbee stick directly given to the OS.

### Debug log

[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/14139195/log.1.txt)

Andrewletsdig18, 2024-02-02:
hi i also have same problem

donagrejja, 2024-02-02:
Many reports of similar issues all over the place. I also have the same issue so I had to roll back.

janwing, 2024-02-02:
Same setup, same problem, rollback to 1.35.1 fixed it.

solarssk, 2024-02-02:
The same situation.

Hardware: Synology DS920+
Home Assistant version: 2024.1.5
Zigbee2MQTT version: worked on 1.35.1
Adapter: Conbee2
How HA is working: VM

Gazoonky, 2024-02-02:
Same issue here.

petterw33, 2024-02-02:
Same issue for me as well

ExMacro, 2024-02-02:
Same thing with Conbee II and latest HA version. Restore from backup to 1.35.1 returned all the lights.

tduong-cloud, 2024-02-02:
Same here with HA on docker, and skyconnect. I had to roll back to 1.35.1 as well.

Rempert, 2024-02-02:
No problems here with 1.35.2. Using Z2M and UZG-01.

linloen, 2024-02-02:
+1 Zigbee2mqtt broken after latest update. Conbee2 not supported.

BarryH1972, 2024-02-02:
No problems, sonoff stick, HA running direct on a thinclient

aceindy, 2024-02-02:
Here too issues with ConBee-II

julllleee, 2024-02-03:
Im using a sonzoff and my aqura devices working great but my Tuya Model TS0042 Singel button stopped working. Solved by downgrading

z-master42, 2024-02-03:
Same problem here, with ConBee 2 on a Home Assistant Yellow. Connection is established for approx. 15 minutes, after which all devices are no longer available. The same after an add-on restart.

BurningTheSky, 2024-02-03:
Similar issue here. Conbee 2 on Unraid via docker. Works for a while then starts erroring talking to any device. Works for a time when container is restarted but then stops again.

pbathuk, 2024-02-03:
I'm on HASS OS with conbee 2 and upgraded last night (via add-on), v1.35.2 is working for me today. I did downgrade the conbee stick a while ago to the firmware mentioned on the zigbee2mqtt website to ensure other bugs did not happen (I forget the version). 
Just wanted to add this, as can see it appears to all be conbee 2 sticks only affected from the thread.

Koenkk, 2024-02-03:
Is anyone running a bare metal setup here (without docker)?

leopoldwang, 2024-02-03:
Also affecting conbee 1.

Koenkk, 2024-02-03:
Can someone provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

fraban, 2024-02-03:
Same here - auto update witch Conbee1 and Conbee 2 ( I run severals HA-Systems) all breaking the connection to all Aqara temp sensors ...

is there some howto  downgrade the zigbee2mqtt to older version... 
running HA in a docker system on raspberry 4

BurningTheSky, 2024-02-03:
> Can someone provide the herdsman debug log of this?
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Not sure if this is what you're looking for?

```
error 2024-02-03 08:43:08 Publish 'set' 'state' to 'Toilet and Under Stairs Lights' failed: 'Error: Command 0x54ef44100076d47e/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (73))'
debug 2024-02-03 08:43:08 Error: Command 0x54ef44100076d47e/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (73)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
info 2024-02-03 08:43:08 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Toilet and Under Stairs Lights' failed: 'Error: Command 0x54ef44100076d47e/2 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (73))'","meta":{"friendly_name":"Toilet and Under Stairs Lights"},"type":"zigbee_publish_error"}'
```

janfuhlrott, 2024-02-03:
Same here with Conbee II.

Koenkk, 2024-02-03:
Please share more about the setup, HA addon, z2m docker image or bare metal. Otherwise I cannot help

BurningTheSky, 2024-02-03:
I'm running z2m as a docker container through Unraid with the ConBee 2 mapped to the container

vsatmydynipnet, 2024-02-03:
Everything was not working here after this auto-update to 1.35-2:
HA latest

Core 2024.1.6
Supervisor 2023.12.1
Operating System 11.4
Frontend 20240104.0

Current version: 1.35.2-1

Conbee II

IKEA Sensors

Power Switches, Motion Sensor and temperature Sensors failed. After 2 reboots th epower switches works again, motion sensor too, but temperature sensors do not work. they have a old value, but do not update. I think i will roll back the VM to 1.35.1 and block auto update

fraban, 2024-02-03:
Core 2024.1.6
Supervisor 2023.12.1
Operating System 11.4
Frontend 20240104.0

Version not working with Aqara-Tempsensors  1.35.2-1
Conbee 2
with IKEA-Buttons and Aqara Sensors 
All Aqara-sesnors - stopped updating after installing 1.35.2 

Downgrading from Backup to 1.35.0 ...   
now Aqara-sensors are back 

I blocked AUTO-Update now

Gazoonky, 2024-02-03:
Interestingly I had this issue after the first attempt at update. I then rolled back my entire setup to before that update. I forgot to disable auto-update for zigbee2mqtt so it’s updated again. Now it seems to work normally.

stomko11, 2024-02-03:
Same issue here on 2 installs. Both VM on undraid (HAOS), conbee stick passed to VM.
On my install, reverting to 1.35.1 seemed to works. On the other not?
At first, I did not notice it, because devices were present, just not updating. I have aqara opplo that isn't really 100% responsive, I thought battery problem. But then later saw the error.
Also, my tuya temp/humidity meters were just sitting at one value since updating to 1.35.2

fraban, 2024-02-03:
[image]
May be this helps a little bit with the aqare-Temp sensor. 
As you can see - when autoupdate starts to 1.35.2 - the sensor frozes the last value and no more updates from the device are received. Then you can see the roll-back to version 1.35.0 and update starts again and is working.

vsatmydynipnet, 2024-02-03:
Really strange:

- Restored a full VM with 1.35.1
- blocked Auto-Updates

Then no Aquara Temp Sensors are filled with value. 
Restarting Z2B fills the values, but some old ones, not updating anymore.

Ikea Power Outlet are working and the Aquara motion detector too. So the alarm for the cat, when coming home works again :-)

If I need 1.35.0 i must to restore from a external backup......

vsatmydynipnet, 2024-02-03:
Restored to 

Core 2024.1.5
Supervisor 2023.12.1
Operating System 11.4
Frontend 20240104.0

1.35.1-1

The Aquara [Xiaomi](https://www.zigbee2mqtt.io/supported-devices/#v=Xiaomi)	[WSDCGQ11LM](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#xiaomi-wsdcgq11lm) 

are not updating, keeping an old value.  Motion detect and power outlet works.

Unsure why this one worked since the update to 1.35.2, now not even rollback does help.

SilverPhoenix99, 2024-02-03:
> Interestingly I had this issue after the first attempt at update. I then rolled back my entire setup to before that update. I forgot to disable auto-update for zigbee2mqtt so it’s updated again. Now it seems to work normally. HA addon Conbee II

I had the same experience. After downgrading and upgrading they seem to be working fine.

Here are the logs nonetheless: https://gist.github.com/SilverPhoenix99/1f0d051557e416accf6e87e0850e1f51

What I noticed different in the logs are the errors with `Failed to configure`.

NB: Ignore `Kitchen Under Sink Leak Detector`, it's been offline for a while.

My setup:
* Docker CE 25.0.1 on Linux Mint 21.2
* Conbee II - firmware version 0x26780700

Docker compose definition, if it helps:
```yaml
zigbee2mqtt:
  container_name: zigbee2mqtt
  image: koenkk/zigbee2mqtt
  network_mode: host
  privileged: true
  restart: unless-stopped
  environment:
    TZ: Europe/London
    DEBUG: zigbee-herdsman*
  volumes:
    - /home/homey/home-assistant/zigbee2mqtt:/app/data
    - /run/udev:/run/udev:ro
  devices:
    - /dev/ttyACM0:/dev/ttyACM0
```

If there's anything else I can include to help trace the issue, please let me know.

fraban, 2024-02-03:
> I had the same experience. After downgrading and upgrading they seem to be working fine.

Hi all ... ok I tried the same:

1. Downgrading  to 1.35.0-1  --- then the aqara-temp sensors started working again  ...   except 2 ... they are marked OFFLINE  now ... I don´t know why. 
2. Upgrade (again) to 1.35.2-1 - and the sensors are still working ....   except the 2 - OFFLINE ones - we lost by doing the first upgrade. Guess I have to delete them and learning again communication with zigbee. ..

So now I let it run and will see what's happening in the next hours --- stable or unstable ... 
Actually it looks that ist stable .

janfuhlrott, 2024-02-03:
Update was installed by auto update 2 times with same result.
Rolled back to previous from backup and installed Update immediately manual and everything works.
Don‘t understand why…

Gazoonky, 2024-02-03:
> > > Interestingly I had this issue after the first attempt at update. I then rolled back my entire setup to before that update. I forgot to disable auto-update for zigbee2mqtt so it’s updated again. Now it seems to work normally. HA addon Conbee II
> > 
> > 
> > I had the same experience. After downgrading and upgrading they seem to be working fine.
> 
> Hi all ... ok I tried the same:
> 
> 1. Downgrading  to 1.35.0-1  --- then the aqara-temp sensors started working again  ...   except 2 ... they are marked OFFLINE  now ... I don´t know why.
> 2. Upgrade (again) to 1.35.2-1 - and the sensors are still working ....   except the 2 - OFFLINE ones - we lost by doing the first upgrade. Guess I have to delete them and learning again communication with zigbee. ..
> 
> So now I let it run and will see what's happening in the next hours --- stable or unstable ... Actually it looks to be "stable" .

Have you allowed enough time for the sensors to send information. I had a few aqara temp sensors that were marked as unavailable for about an hour before they sent some data and reappeared online.

vsatmydynipnet, 2024-02-03:
Finally it looks like things are working:

Rolled back
Made manual Update incl. Z2B to 1.35.2
Enabled join
startet join on the 3 missing Aquara temp sensors.
currently they are reporting again. produced some temp changes, looks good

fraban, 2024-02-03:
…
> > So now I let it run and will see what's happening in the next hours --- stable or un
> Have you allowed enough time for the sensors to send information. I had a few aqara temp sensors that were marked as unavailable for about an hour before they sent some data and reappeared online. You could just warm them up to see if that gets them to update data.

Yes thats the reason why I leave it untouched now. So I can see if the 2 missing aqara-sensors will reconnect again … Actually it seems that this won’t work.

jacekmarek, 2024-02-03:
> Is anyone running a bare metal setup here (without docker)?

@Koenkk I have HA running on the NUC:
Core 2024.1.5
Supervisor 2023.12.1
Operating System 11.4 
Z2M as addon

Zigbee2MQTT version:       [1.35.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type:    ConBee2/RaspBee2
Coordinator revision: 0x26720700
Zigbee-herdsman-converters version:    18.25.1
Zigbee-herdsman version:     0.33.5

I upgrade Z2M from 1.35.1-1 to 1.35.2-1, after 10 minutes later, no communication with devices:

```
debug 2024-02-03 19:06:40Saving state to file /config/zigbee2mqtt/state.json
debug 2024-02-03 19:06:46Received MQTT message on 'zigbee2mqtt/dining room light/set' with data '{"state_l1":"OFF"}'
debug 2024-02-03 19:06:46Publishing 'set' 'state' to 'dining room light'
error 2024-02-03 19:06:57Publish 'set' 'state' to 'dining room light' failed: 'Error: Command 0xa4c13895232f30db/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (43))'
debug 2024-02-03 19:06:57Error: Command 0xa4c13895232f30db/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (43)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1116:32) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
info 2024-02-03 19:06:57MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'dining room light' failed: 'Error: Command 0xa4c13895232f30db/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (43))'","meta":{"friendly_name":"dining room light"},"type":"zigbee_publish_error"}'
```
How can I help you debug the problem?

Koenkk, 2024-02-03:
Can you see if the following fixes the problem:
- Make sure you are running 1.35.2
- ssh into your system and execute:

```bash
docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) /bin/sh
apk add npm
npm install zigbee-herdman@0.30.0
exit
docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
```
- See if the issue is solved now

BurningTheSky, 2024-02-03:
> Can you see if the following fixes the problem:
> 
>     * Make sure you are running 1.35.2
> 
>     * ssh into your system and execute:
> 
> 
> ```shell
> docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) /bin/sh
> apk add npm
> npm install zigbee-herdsman@0.30.0
> exit
> docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
> ```
> 
>     * See if the issue is solved now

I tried that and I seem to be still having issues

```
2024-02-03 21:11:38Failed to configure 'Toilet and Under Stairs Lights', attempt 1 (Error: Cluster with name 'manuSpecificLumi' does not exist at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:61:19) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:363:35) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:1137:41) at Configure.configure (/app/lib/extension/configure.ts:120:37) at /app/lib/extension/configure.ts:85:62 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23) at EventEmitter.emit (node:events:529:35) at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:51:22) at Controller.<anonymous> (/app/lib/zigbee.ts:72:27) at Controller.emit (node:events:517:28))
```

jacekmarek, 2024-02-03:
> See if the issue is solved now
@Koenkk I did as you wanted:
Frontend version:    0.6.152
Zigbee-herdsman-converters version:     18.25.1
Zigbee-herdsman version:     0.30.0
After 10 minutes later, everything works correctly, the problem is solved.

Gazoonky, 2024-02-03:
> Can you see if the following fixes the problem:
>
> * Make sure you are running 1.35.2
> * ssh into your system and execute:
>
> ```shell
> docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) [...]

Is it possible that downgrading and then reinstalling achieves the same thing - some of us have found the problem goes away if we do this? 
Do we all need to apply this fix, or only if there is an ongoing problem?
Could you explain what caused this problem, there didn’t seem to be any error messages which is a bit worrying. If entities are not responding then surely there should have been some sort of error message in Home assistant that highlighted that there was a big problem.

SilverPhoenix99, 2024-02-04:
@Koenkk I've applied v0.30.0, but I'm seeing similar logs.

```
zigbee2mqtt  | Zigbee2MQTT:error 2024-02-04 00:27:00: Failed to configure 'Hall Entrance Motion Sensor', attempt 1 (Error: Cluster with name 'manuSpecificLumi' does not exist
```

I'm wondering, the Aqara/Xiaomi devices were merged as Lumi. I almost exclusively have Aqara devices, and maybe the 2 might be related (or a red-herring), especially since the message above mentions that it can't find the `manuSpecificLumi` cluster.

SilverPhoenix99, 2024-02-04:
Update: v0.30.0 seems to have broken a few things.

The devices are still online, and I can manually turn on the lights through Home Assistant or the frontend, but the motion detectors aren't picking up any motion. Interestingly, this matches the errors in logs. Almost all failures are about configuring motion sensors.

Logs: https://gist.github.com/SilverPhoenix99/b431fbda28c7041eeb75dc0994b8c0c9

Koenkk, 2024-02-04:
- `Cluster with name 'manuSpecificLumi' does not exist is` is already solved in the `latest-dev` (so use that  Docker tag)
- If 0.30.0 fixes it, can you upgrade the zigbee-herdsman version and see which one breaks? An overview of the version can be found [here](https://github.com/Koenkk/zigbee-herdsman/releases?page=2)

Albruaje, 2024-02-04:
Same issue for me, but wirh Phoscon Raspbee II

jhurbain, 2024-02-04:
Hi, same issue for me

BurningTheSky, 2024-02-04:
> * `Cluster with name 'manuSpecificLumi' does not exist is` is already solved in the `latest-dev` (so use that  Docker tag). Double check that `latest-dev` does not fix the issue completely already.
> 
>     * If `latest-dev` doesn't fix it, can you upgrade the zigbee-herdsman version and see which one breaks (starting from zigbee-herdsman 0.30.0? An overview of the version can be found [here](https://github.com/Koenkk/zigbee-herdsman/releases?page=2)

Switched to latest-dev and so far no errors.

Koenkk, 2024-02-04:
@BurningTheSky latest-dev without any changes completely fixes this issue?

BurningTheSky, 2024-02-04:
> @BurningTheSky latest-dev without any changes completely fixes this issue?

I had tried the npm install that you suggested previosuly, but other than that, no changes. I've had one command fail but that's to a valve that might be slight of range anyway. One in range worked fine

Koenkk, 2024-02-04:
@BurningTheSky the `npm install` was before you switched to `latest-dev` right?

BurningTheSky, 2024-02-04:
That's correct, and I'd switched back to 1.34.0 after that.

This is the error for the one that I think is maybe out of range:
```
error 2024-02-04 08:53:25Publish 'set' 'occupied_heating_setpoint' to 'Small Radiator' failed: 'Error: Write 0x54ef441000787c04/1 hvacThermostat({"occupiedHeatingSetpoint":1350}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (23))'
debug 2024-02-04 08:53:25Error: Write 0x54ef441000787c04/1 hvacThermostat({"occupiedHeatingSetpoint":1350}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (23)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1561:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
info 2024-02-04 08:53:25MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupied_heating_setpoint' to 'Small Radiator' failed: 'Error: Write 0x54ef441000787c04/1 hvacThermostat({\"occupiedHeatingSetpoint\":1350}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (23))'","meta":{"friendly_name":"Small Radiator"},"type":"zigbee_publish_error"}'
```

One other thing I've noticed is that since the issues the network map no longer shows connections from the coordinator but a router still shows. It also gives `Failed to execute LQI for xxx` messages.

SilverPhoenix99, 2024-02-04:
@Koenkk on `latest-dev`, everything seems to be working properly - with no changes to the container, i.e., `zigbee-herdsman` is v0.33.7.

Koenkk, 2024-02-04:
great! if some more people can confirm I will publish a new release

BurningTheSky, 2024-02-04:
I've just got home and found I'm still getting the errors:

```
Zigbee2MQTT version
    1.35.2-dev commit: [3d37892](https://github.com/Koenkk/zigbee2mqtt/commit/3d37892)

Coordinator type
    ConBee2/RaspBee2

Coordinator revision
    0x26720700

Coordinator IEEE Address
    0x00212effff085a0a

Frontend version
    0.6.154

Zigbee-herdsman-converters version
    18.28.0

Zigbee-herdsman version
    0.33.7
```

```
error 2024-02-04 13:31:42Publish 'set' 'state' to 'Hall 3 Gang Lights' failed: 'Error: Command 0xa4c1386b29785531/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (47))'
debug 2024-02-04 13:31:42Error: Command 0xa4c1386b29785531/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (47)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
info 2024-02-04 13:31:42MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Hall 3 Gang Lights' failed: 'Error: Command 0xa4c1386b29785531/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (47))'","meta":{"friendly_name":"Hall 3 Gang Lights"},"type":"zigbee_publish_error"}'
```

Koenkk, 2024-02-04:
@BurningTheSky this looks like a occasional connection issue, does the system still freeze completely like with 1.35.2?

jonasrickert, 2024-02-05:
> great! if some more people can confirm I will publish a new release

latest-dev fixed it for me as well

BurningTheSky, 2024-02-05:
> @BurningTheSky this looks like a occasional connection issue, does the system still freeze completely like with 1.35.2?

I had a system hang where all sensors froze but I'm not sure of the cause and I took it as an opportunity to move to a Sonoff adapter to be honest!

aranw, 2024-02-05:
> great! if some more people can confirm I will publish a new release

Is it possible to upgrade to `latest-dev` in HAOS? I run Zigbee2MQTT via home assistant. Happy to give it a try if there's a way to get that to update to the `latest-dev` release

Nuuki9, 2024-02-05:
> great! if some more people can confirm I will publish a new release

The dev version is working for me, whereas 1.3.5.2 didn't.

[139 more replies]

closed 2024-02-13 by Koenkk