Koenkk/zigbee2mqtt #21165: Z2M 1.35.2 issues and errros with LYWSD03MMC OTA/Firmware

shirou93, 2024-02-02:
### What happened?

After update to 1.35.2 all LYWSD03MMC devices for errors with not supported features.

Default custom firmware on docs is devbis 
https://github.com/devbis/z03mmc
This firmware have support ota on z2m.

This pull is for differ firmware from (https://github.com/pvvx/ZigbeeTLc/)

https://github.com/Koenkk/zigbee-herdsman-converters/pull/6939

I flash via USB one device to support news features but this device got now OTA errors check because  ota od for devbis firmware.

### What did you expect to happen?

Add to docs new firmware info and differ other firmware. Get info which firmware is supported by ota.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

20230507

### Adapter

Sonoff P

### Setup

Debian supervised 

### Debug log

After reflash

[12:02:48] INFO: Preparing to start...
[12:02:49] INFO: Socat not enabled
[12:02:49] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:warn  2024-02-02 12:04:11: OTA: Images currently unavailable for device 'LYWSD03MMC', hardwareVersion '14', manufacturerName Xiaomi, {"fieldControl":0,"manufacturerCode":4417,"imageType":522,"fileVersion":18427905}'
Zigbee2MQTT:error 2024-02-02 12:05:37: Failed to configure '0xa4c1389794e737b5', attempt 1 (Error: Read 0xa4c1389794e737b5/1 hvacUserInterfaceCfg(["programmingVisibility"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:84:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:warn  2024-02-02 12:06:11: OTA: Images currently unavailable for device 'LYWSD03MMC', hardwareVersion '20', manufacturerName Xiaomi, {"fieldControl":0,"manufacturerCode":4417,"imageType":522,"fileVersion":18427905}'
Zigbee2MQTT:warn  2024-02-02 12:07:24: OTA: Images currently unavailable for device 'LYWSD03MMC', hardwareVersion '14', manufacturerName Xiaomi, {"fieldControl":0,"manufacturerCode":4417,"imageType":522,"fileVersion":18427905}'

XFNeo, 2024-02-02:
I also have a problem with LYWSD03MMC after update from 1.35.1 to 1.35.2:
```Error 2024-02-02 19:55:52
Failed to configure '0xa4c1388b14f755fb', attempt 1 (Error: Read 0xa4c1388b14f755fb/1 hvacUserInterfaceCfg(["programmingVisibility"], 
{"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,
"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') 
at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) 
at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:13) 
at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:84:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) 
at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

stomko11, 2024-02-02:
I am seeing something similar on other devices too (some zigbee plugs, aqara opplo, etc.). Not exactly the same, but started after last update. Reverted back and does not seem to be different. Here is an example:
'''
Zigbee2MQTT:error 2024-02-02 23:27:04: Failed to configure 'Dishwasher plug', attempt 1 (Error: Read 0xa4c138a8d0fc1437/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (4))
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processTimers (node:internal/timers:509:9)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:81:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:460:28)
    at Object.configureMagicPacket (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:373:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3398:13)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:72:17))
'''

VivantSenior, 2024-02-03:
> I also have a problem with LYWSD03MMC after update from 1.35.1 to 1.35.2:
> 
> ```
> Failed to configure '0xa4c1388b14f755fb', attempt 1 (Error: Read 0xa4c1388b14f755fb/1 hvacUserInterfaceCfg(["programmingVisibility"], 
> {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,
> "transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') 
> at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) 
> at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:63:13) 
> at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:84:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) 
> at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
> ```

I'm having the same issues with `hvacUserInterfaceCfg` but also with `genPowerCfg`. Logs below:

```
Failed to configure 'Kitchen environment screen', attempt 3
(Error: Bind 0xa4c138b73e4f0a87/1 genPowerCfg from '0x00124b0029de14c4/1' failed (AREQ - ZDO - bindRsp after 10000ms)
 at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
 at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:692:28
 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
 at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)
 at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:54:13)
 at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:84:17)
 at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
 at Configure.configure (/app/lib/extension/configure.ts:120:13)
 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

All of mine LYWSD03MMC have newest available firmware: [1.0.6 / 20231109](https://github.com/devbis/z03mmc/releases/tag/1.0.6).

shirou93, 2024-02-03:
> I'm having the same issues with `hvacUserInterfaceCfg` but also with `genPowerCfg`. Logs below:
>
> ```
> Failed to configure 'Kitchen environment screen', attempt 3
> (Error: Bind 0xa4c138b73e4f0a87/1 [...]

This issues is because z2m updated was changed listening to different firmware maked by pvvx.
On docs no info about its.

closed 2024-02-06 by Koenkk