Koenkk/zigbee2mqtt #21171: Aqara Wireless remote switch H1 (WXKG15LM)  errors

dann032, 2024-02-02:
### What happened?

Aqara	Wireless remote switch H1 (double rocker ) WXKG15LM,
the following errors appear in the log: 
error 2024-02-02 16:27:05Publish 'get' 'operation_mode' to 'Hallway remote' failed: 'Error: Read 0x54ef4410001b0a4c/1 manuSpecificLumi(["mode"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24844 - 1 - 45 - 64704 - 1 after 10000ms)'

error 2024-02-02 16:27:26Publish 'get' 'click_mode' to 'Hallway remote' failed: 'Error: Read 0x54ef4410001b0a4c/1 manuSpecificLumi([293], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24844 - 1 - 46 - 64704 - 1 after 10000ms)'

### Zigbee2MQTT version

1.35.2-dev commit: da1d0d8

### Adapter firmware version

7.3.2.0 build 212

### Adapter

Sonoff dongle-E

### Setup

Z2MQTT add-on, Raspberry pi 3B+, Latest HA OS

VivantSenior, 2024-02-03:
I have quite similar error (issue with `manuSpecificLumi`) in the logs for the [WXKG15LM](https://www.zigbee2mqtt.io/devices/WXKG15LM.html):

```
Failed to configure 'Upper hall light wireless switch', attempt 1
(Error: Write 0x54ef4410001b0206/1 manuSpecificLumi({"mode":1}, 
{
	"timeout": 10000,
	"disableResponse": false,
	"disableRecovery": false,
	"disableDefaultResponse": true,
	"direction": 0,
	"srcEndpoint": null,
	"reservedBits": 0,
	"manufacturerCode": 4447,
	"transactionSequenceNumber": null,
	"writeUndiv": false
})
failed (Timeout - 1920 - 1 - 107 - 64704 - 4 after 10000ms)
	at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
	at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47)
	at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
	at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
	at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28)
	at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2829:13)
	at Configure.configure (/app/lib/extension/configure.ts:120:13)
	at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

gatinhot, 2024-02-05:
Similar issue here
[image]

WarC0zes, 2024-02-07:
Hi,
with version 1.35.3-1, I get an error too.
```
2024-02-07 06:57:10Failed to configure 'Interrupteur Salon', attempt 1 (Error: Write 0x54ef44100044ca82/1 manuSpecificLumi({"mode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2830:13) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

closed 2024-09-05 by github-actions[bot]