Koenkk/zigbee2mqtt #21172: [New device support]: TuYa iH-F8260

kikher, 2024-02-02:
### Link

https://www.tuya.com/

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0x7cc6b6fffec06795","nwkAddr":64603,"manufId":4098,"manufName":"_TZ3290_gnl5a6a5xvql7c2a","powerSource":"Mains (single phase)","modelId":"TS1201","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":61440,"inClusterList":[0,4,5,3,60672,57348,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS1201","manufacturerName":"_TZ3290_gnl5a6a5xvql7c2a","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1706801740930}

### Comments

The external definition is obtained from this repository. https://github.com/Koenkk/zigbee2mqtt/discussions/19881

### External defintion

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const e = exposes.presets;
const ea = exposes.access;
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

const definition = {
    fingerprint: [
        {
            modelID: 'TS1201',
            manufacturerName: '_TZ3290_gnl5a6a5xvql7c2a',
        },
    ],
    model: 'iH-F8260',
    vendor: 'TuYa',
    description: 'Universal smart IR remote control',
	fromZigbee: [
		fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
		fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
	],
	toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send()],
};

module.exports = definition;
```

maddog87, 2024-02-03:
I have this device, How can I use the above code to use it in HA? Thanks I am new to using HA

Koenkk, 2024-02-03:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-03 by Koenkk

vinodmishra, 2024-04-24:
@Koenkk While this was based on iH-F8260  it's not quite the same as it's mains powered and has no battery so keeps giving me this error.

error 2024-04-24 03:19:09Failed to configure '0xb0c7defffe7e90c4', attempt 1 (Error: ZCL command 0xb0c7defffe7e90c4/1 genPowerCfg.read(["batteryVoltage","batteryPercentageRemaining"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:24) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:369:13) at Configure.configure (/app/lib/extension/configure.ts:122:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
info 2024-04-24 03:19:09MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0xb0c7defffe7e90c4"},"error":"Failed to configure (ZCL command 0xb0c7defffe7e90c4/1 genPowerCfg.read([\"batteryVoltage\",\"batteryPercentageRemaining\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"xhv48-1"}'

I can see the external converter has no battery stuff while the one where this fingerprint was added does