Koenkk/zigbee2mqtt #21175: Zigbee2MQTT crashes

TwoPlayer, 2024-02-02:
### What happened?

Zigbee2MQTT just crash after some time. I followed those steps: [Zigbee2MQTT crashes after some time](https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time)

The Adapter has a completly new cable (problem was also with a old one), the latest firmware and autosuspend is disabled:
[image]

### What did you expect to happen?

Zigbee2MQTT don't crash

### How to reproduce it (minimal and precise)

Let Zigbee2MQTT runnig and wait (sorry, there's no action I'm aware of, which will trigger the crash.

### Zigbee2MQTT version

1.35.2 

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-On on Home Assistant OS on Raspberry Pi 3

### Debug log

[core_mosquitto_2024-02-02T16-45-59.511Z.log](https://github.com/Koenkk/zigbee2mqtt/files/14142912/core_mosquitto_2024-02-02T16-45-59.511Z.log)
[host_2024-02-02T16-46-16.421Z.log](https://github.com/Koenkk/zigbee2mqtt/files/14142913/host_2024-02-02T16-46-16.421Z.log)
[supervisor_2024-02-02T16-44-43.318Z.log](https://github.com/Koenkk/zigbee2mqtt/files/14142914/supervisor_2024-02-02T16-44-43.318Z.log)
[zigbee2mqtt-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14142915/zigbee2mqtt-log.txt)

garry0garry, 2024-02-02:
After update to [1.35.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.2) commit: [06b3395](https://github.com/Koenkk/zigbee2mqtt/commit/06b3395):

[TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)
`error 2024-02-02 22:36:33Publish 'set' 'state' to '0xa4c138b0ae6701b2' failed: 'Error: Command 0xa4c138b0ae6701b2/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 28728 - 1 - 19 - 6 - 11 after 10000ms)'`

[RTCZCGQ11LM](https://www.zigbee2mqtt.io/devices/RTCZCGQ11LM.html#aqara-rtczcgq11lm)
`error 2024-02-02 22:37:47Failed to configure '0x54ef4410006492ac', attempt 3 (Error: Read 0x54ef4410006492ac/1 manuSpecificLumi([268], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47985 - 1 - 20 - 64704 - 1 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:486:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:466:28) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:1866:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

thueske, 2024-02-03:
Maybe related... Problems since 2 days.

`Error 2024-02-03 06:53:16Publish 'set' 'state' to 'Regallampe' failed: 'Error: Command 0x001788010b856bda/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (252))'`

`Debug 2024-02-03 06:53:16Error: Command 0x001788010b856bda/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (252)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1116:32) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)`

srouaix, 2024-02-03:
Same issue on my side, I downgraded to 1.35.1 and It works without crash.

cornholio1234, 2024-02-04:
Just a bump - same for me. 1.35.2 looses all connections after some minutes. 1.35.1 is fine.

VDRainer, 2024-02-04:
Had that issue after 1.35.2 first start.
Another restart fixed it for me.
No issues since 14 hours now. :crossed_fingers:

garry0garry, 2024-02-05:
> Another restart fixed it for me.

Rebooted the computer. It didn't help me.

cornholio1234, 2024-02-05:
> Had that issue after 1.35.2 first start. Another restart fixed it for me. No issues since 14 hours now. 🤞

Reboot did not help in my case. I am using an Deconz Stick - so ich guess it is not hardware related (?).
Running on a proxmox debian VM. Couldn't see anything else than timeouts in the logs ...

MoosJ80, 2024-02-05:
For me the same.
Happened 3 times now. I have restored to 1.35.2-1
Thx for the support

VDRainer, 2024-02-05:
Hmm...
Since the last restart yesterday it was stable for about 30 hour's.
After another restart it took 10 Minutes and crashed again.
[debug_log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/14172197/debug_log1.txt)
The next restart an it is running since 3 hours.

olo81, 2024-02-06:
At my place entities are online, still something is not working.  Interestingly, after the update it was ok, only a reboot revealed the problems. Reverting to the previous version didn't change anything :( One of the devices I unsealed and it no longer manages to reconnect. I don't really have an idea how to deal with this.

OddyAte, 2024-02-06:
Same issue, for me a restart fixes the issue for maybe 10 minutes. I am using Proxmox and deCONZ ConBee II. The issue is still present in v1.35.3-1

Koenkk, 2024-02-07:
If you still have this issue with 1.35.3, post the herdsman debug log from starting z2m until the crash

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

TwoPlayer, 2024-02-09:
> If you still have this issue with 1.35.3, post the herdsman debug log from starting z2m until the crash
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Had the problem again today, had to restart the whole system after this, because the addon wouldn't start.
This is the log from before the crash: https://drive.google.com/file/d/1cKMjBM1j3PEMshXbqCnV7rejBQUKwyfZ/view?usp=sharing (Sorry couldn't upload it to Github)

Zigbee2MQTT version
1.35.3-1

Adapter firmware version
20230507

Adapter
SONOFF Zigbee 3.0 USB Dongle Plus-P

Setup
Add-On on Home Assistant OS on Raspberry Pi 3

Home Assistant Core
2024.1.6

Home Assistant Supervisor
2024.01.1

Home Assistant Operating System
11.5

Hundhausen, 2024-02-09:
> If you still have this issue with 1.35.3, post the herdsman debug log from starting z2m until the crash
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

When you mean with STDOUT that what I can see on the Log Page on the Addon page, I can give you that. 

[log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/14225722/log1.txt)

Also the other Log, just in case:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14225738/log.txt)

"Luftqualitätsmesser Schlafzimmer" is currently unplugged. All other devices in the first log file, where powered on.

Before mentioning #9554 with the firmware version, I didn't had this kind of issues. Until this month, I had no big issues. Sometimes a button needing a 2nd invite to do stuff. I didn't changed my setup, except installing updates. The latest changes where removing the power report on battery operated stuff, because the batteries die so fast. And I got the IKEA Air quality sensor as new device and in the beginning I suspected it to cause the issues, because it reports every second and I had some crashes but not like every day. And unplugging it, should eliminate that issue.

Also to be specific: the addon doesn't crash, just every device does not respond (send and receive). 

Zigbee2MQTT version
1.35.3-1

Adapter firmware version
0x26780700

Adapter
ConBee2

Setup
Add-On - RPI 3

Home Assistant Core
2024.2.1

Home Assistant Supervisor
2024.01.1

Home Assistant Operating System
11.5

Koenkk, 2024-02-11:
@TwoPlayer in your case it might help to try the `20221226` firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin

@Hundhausen since you are using the Conbee, it might be related to https://github.com/Koenkk/zigbee2mqtt/issues/21162

Hundhausen, 2024-02-11:
I've now downgraded 1.34.0-1. It got quite out of hand and the restarts needed shortened. I tried to automate the restart but the restarts took to long and it restarted every hour, with my detection method (checking one device getting its update entity unavailable).
It looks like 1.34 does contain the same bug or at least similar. It does restart every hour, nearly on point every time but the watchdog does catch it crashing and the reboot is faster.

I will watch the other issue and try out some stuff and I might report back there, if I have additional information.

MoosJ80, 2024-02-11:
> For me the same. Happened 3 times now. I have restored to 1.35.2-1 Thx for the support

Fixed it for me. Stable for few days now.
Thx!

TwoPlayer, 2024-02-15:
> @TwoPlayer in your case it might help to try the `20221226` firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin

Hi @Koenkk, I changed my system from a raspberry Pi B to a Thin Client with more RAM and now everything is working. I think the problem was, that for some reason the RAM was full and then HA restarted, but Zigbee2MQTT failed to restart.