Koenkk/zigbee2mqtt #21179: New install with previous configuration.yaml file occur a empty dashboard (devices not found)

espeka44, 2024-02-02:
### What happened?

I reinstalled Z2M on a new machine (RPi 3B+ to RPi 5) and exported my configuration.yaml file with all the configurations of my adapter (ConBee II) and all my devices.
Zigbee2Mqtt successfully started but on the frontend I have nothing, no device appears.

### What did you expect to happen?

Backup my previous configuration, recognized automaticly all my devices

### How to reproduce it (minimal and precise)

Setup on a new machine with same coordinator. Rename default configuration.yaml to configuration_old.yaml and cut/paste the configuration.yaml of the old machine in zigbee2mqtt/data.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Raspberry Pi OS for Pi 5 8Go (Debian 12 Bookworm)

### Debug log

info  2024-02-02 22:29:17: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-02-02.22-29-17' filename: log.txt
info  2024-02-02 22:29:17: Starting Zigbee2MQTT version 1.35.2 (commit #fdc9f7d)
info  2024-02-02 22:29:17: Starting zigbee-herdsman (0.33.5)
info  2024-02-02 22:29:17: zigbee-herdsman started (resumed)
info  2024-02-02 22:29:17: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
info  2024-02-02 22:29:17: Currently 0 devices are joined:
warn  2024-02-02 22:29:17: `permit_join` set to  `true` in configuration.yaml.
warn  2024-02-02 22:29:17: Allowing new devices to join.
warn  2024-02-02 22:29:17: Set `permit_join` to `false` once you joined all devices.
info  2024-02-02 22:29:17: Zigbee: allowing new devices to join.
info  2024-02-02 22:29:17: Connecting to MQTT server at mqtt://localhost
info  2024-02-02 22:29:17: Connected to MQTT server
info  2024-02-02 22:29:17: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2024-02-02 22:29:17: Started frontend on port 8080
info  2024-02-02 22:29:17: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"fdc9f7d","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.35.2"}'
info  2024-02-02 22:29:17: Zigbee2MQTT started!

furieuxjk, 2024-02-03:
Did you copy the database file as well?

espeka44, 2024-02-03:
No. I did not know how do it, so i didnt it. Any other way to hope fix it with the original configuration.yaml file ?

furieuxjk, 2024-02-03:
File configuration.yaml is just a configuration. All devices are in the database.db file located in data folder. You can SSH into old RPi and copy database.db file with scp command to the new RPi.

espeka44, 2024-02-03:
Formatted my old device. :(

furieuxjk, 2024-02-03:
Oh, well in that case you have to pair all devices one by one again.

espeka44, 2024-02-03:
I will try to add them one by one. Keep in touch if another issue spot.

marinus61, 2024-03-06:
@furieuxjk 

I experienced same issue as mentioned above however I made a full backup of the RPi4 and restored it on PRi5, all Zigbee devices came online, but nothing showed on the GUI under zigbee2mqtt

Marinus

closed 2024-10-03 by github-actions[bot]