Koenkk/zigbee2mqtt #21185: [Wrong device]: Miboxer E2-ZR shown as TuYa TS0502B

Shellfishgene, 2024-02-03:
### Link

https://miboxer.com/product/2-in-1-led-controller-e2-zr/

### Model

E2-ZR

### Description

LED controller

### Vendor

Miboxer

### Picture (link)

https://miboxer.com/wp-content/uploads/45656.webp

### Database entry

{"id":56,"type":"Router","ieeeAddr":"0xa4c1382d152561fd","nwkAddr":1861,"manufId":4417,"manufName":"_TZB210_ayx58ft5","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1811148075},"lastSeen":1706958369967,"defaultSendRequestWhen":"immediate"}

pvyleta, 2024-02-22:
I think you might be able to add the device into the database yourself: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

vtolstov, 2024-04-06:
i have the same device, may be we can add it to database for all users ?

closed 2024-06-15 by Koenkk

andrew01144, 2024-07-19:
Hi - I have the same problem, but have not succeeded in fixing it.

I my E2-ZR successfully joined Z2M, but it appeared as Tuya TS0502B. (Z2M was from ~01-Jun, that's before the fix).
I upgraded Z2M; herdsman now shows as 19.62.0.
I removed the device. Re-joined the device, but it still appears as Tuya TS0502B.
Am I missing a step?

I tried bringing up a new instance of Z2M with a different adapter. Still appears as Tuya TS0502B.
 
Many thanks for any help!

Also - setting 'transition' seems to have no affect on the actual transition.