Koenkk/zigbee2mqtt #21190: DANFOSS TRV Connection broken - Link Quality 0 when paired successful

rubenkr, 2024-02-03:
### What happened?

- Valves were not controllable via MQTT messages, nor Z2M Frontend
- Those valves connected to the router directly show 0 Link Quality, however other Zigbee devices have not problems

[image]

### What did you expect to happen?

- Normal connections to the router
 - Interestingly other Danfoss valves connect but via a repeater (e.g. Hue Lights in this case)

### How to reproduce it (minimal and precise)

- Open Z2M -> Map -> Create
- Find Children displaying Link Quality 0 (all directly connected to the router)

### Zigbee2MQTT version

    1.35.1 commit: 3c962042

### Adapter firmware version

zStack3x0

### Adapter

Slaesh's CC2652R

### Setup

Proxmox Container with Home-Assistant

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14151959/log.txt)

Mikkooooo, 2024-02-03:
….same situation for me.

rubenkr, 2024-02-03:
thanks for the fast reply. have you already tried and usb extension cable as described here? https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html. But interestingly to hear that I am not the only one.

Mikkooooo, 2024-02-03:
There are a lot of  posts here describing the same behaviour.
I have tried many solutions but no luck.
Status messages FROM the TRV wor OK every time but controlling often fails.
The it‘s „only“ nessecary to unplug and replug the battery again.
Don‘t know why !

closed 2024-09-01 by github-actions[bot]

rholighaus, 2024-10-18:
Observing the same issue