Koenkk/zigbee2mqtt #21193: [Wrong device]:  TS011F_wall_outlet turned into TS011F_plug_3 after the update   zigbee2mqtt 1.35.2-1

AlexzeyK, 2024-02-03:
### Link

https://www.zigbee2mqtt.io/devices/TS011F_wall_outlet.html

### Model

BSEED Zigbee Socket

### Description

Zigbee EU Wall Socket

### Vendor

BSEED  (TuYa)

### Picture (link)

https://ae01.alicdn.com/kf/H8fddddf2e4974dbcaf1de394e9e99452t.jpg

### Database entry

"id":4,"type":"Router","ieeeAddr":"0xa4c138fb9c8d1217","nwkAddr":28733,"manufId":4417,"manufName":"_TZ3000_o1jzcxou","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_o1jzcxou","zclVersion":3,"appVersion":64,"modelId":"TS011F","powerSource":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1706971135418

### Notes

In the previous version zigbee2mqtt 1.35.1-1  the outlet was detected correctly and worked (controlled) correctly.
[image]

After the update 1.35.2-1 the socket appeared as a plug with energy monitoring.
[image]

now the control doesn't work. Only the signal level is correctly displayed.

    "child_lock": null,
    "current": null,
    "energy": null,
    "indicator_mode": null,
    "linkquality": 124,
    "power": null,
    "power_outage_memory": null,
    "state": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
 
    "update_available": null,
    "voltage": null,
    "last_seen": "2024-02-03T17:33:29+02:00"

is automatically mapped to the wrong converter.

Koenkk, 2024-02-04:
This is because of https://github.com/Koenkk/zigbee2mqtt/issues/20702 (@franele), does your device work correctly in 1.35.2?

AlexzeyK, 2024-02-04:
i ignored version 1.35.2 and upgraded from 1.35.1-1 to 135.2-1. I don't have a backup from 1.35.2 to check.
 Please tell me how I can downgrade to 1.35.2 in home assistant?

AlexzeyK, 2024-02-04:
this is a record of a working outlet from the version 1.35.1-1 database:
"id":3,"type":"Router","ieeeAddr":"0xa4c138fb9c8d1217","nwkAddr":58137,"manufId":4417,"manufName":"_TZ3000_o1jzcxou","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_o1jzcxou","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1707043545757,"defaultSendRequestWhen":"immediate"

franele, 2024-02-04:
> This is because of #20702 (@franele), does your device work correctly in 1.35.2?

Hello, the devices work correctly. I have tested the on/off function and it works correctly.
The energy monitoring still does not work, until today I have not been able to test it. I'm pairing the devices again...

s482dcaw, 2024-02-29:
Adding here that one side effect from this change is that now devices show a new firmware is available. 
The firmware cannot be installed and fails, which is probably expected as it is a different device.
[image]
[image]

rosicenko, 2024-03-18:
I have same device and it shows as TS011F_plug_1. Controlling functions well but metering was firstly showing only N/A and after OTA update I receive constant 0 for all parameters.
[image]

rosicenko, 2024-03-19:
So I just contacted BSEED and regarding to them they have 2 different sockets. One is with metering and another without any metering. They told me this specific one is without metering capabilities. That means it works in Z2MQTT just fine.

derkorte, 2024-04-08:
I've had a similar device as in wall socket without energy monitoring. Since updating it's also configured as plug with energy monitoring. Repair was no success. Here my log error:

Failed to configure 'Steckdose 4', attempt 1 (Error: Bind 0xa4c1380f62782395/1 haElectricalMeasurement from '0x00124b001cdcaac9/1' failed (Error: status: 130) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3771:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:21) at Configure.configure (/app/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Anyone knows how to fix?

closed 2024-11-06 by github-actions[bot]

martinw72, 2025-05-06:
any update as I was falling into the same issue!

martinw72, 2025-05-07:
> Friend, @martinw72 <https://github.com/martinw72>
> what problem do you have?
>
>    1. Does your outlet with energy monitoring look like a regular outlet?
>    2. Does your outlet without energy monitoring look like an outlet with
>    energy monitoring

It has energy monitoring but is not working. Also it looks like a plug, but
is a wall socket!

Martin Weilhartner
+436765339609

AlexzeyK, 2025-05-07:
The problem is that the manufacturer made 2 different devices (with energy monitoring and without energy monitoring) with the same ID _TZ3000_o1jzcxou.
To solve this problem, you need to manually select the converter and add it to the folder for external converters.
I don't see any other option(

s482dcaw, 2025-05-07:
> The problem is that the manufacturer made 2 different devices (with energy monitoring and without energy monitoring) with the same ID _TZ3000_o1jzcxou. To solve this problem, you need to manually select the converter and add it to the folder for external converters. I don't see any other option(

can you explain this a bit more? Where do we find the "right" converter, and where does it need to be added?

AlexzeyK, 2025-05-08:
If your version of Z2M is greater than 2.0, it is not installed in the directory with HA.
Go to the Z2M location folder and create a folder named external_converters  .
Here is the path, for example: /addon_configs/45df7312_zigbee2mqtt/zigbee2mqtt/external_converters/

[image]
[image]

in the folder external_converters put the converters.

[image]

After adding or editing a converter, it is advisable to restart Z2M.
Note: after starting Z2M, look at the file name of your converter. If instead of .js at the end there is .invalid, then there is an error in the converter and Z2M did not load it for itself.