Koenkk/zigbee2mqtt #21199: Aqara WXCJKG12LM Error in Logfile after update to Zigbee2MQTT 1.35.2-1

alwillha, 2024-02-03:
### What happened?

error 2024-02-03 10:56:41: Failed to configure 'Aqara-Opple-2-2', attempt 2 (Error: Write 0x54ef4410003b3de4/1 manuSpecificLumi({"mode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2582:13)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

### What did you expect to happen?

no error messages after the add-on update

### How to reproduce it (minimal and precise)

Update Zigbee2MQTT Add-on from installed version 1.35.1-1 to latest version 1.35.2-1

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

20221226

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Zigbee2MQTT add-on on latest Home Assistant OS, odroid-n2

### Debug log

[log.txt ](https://github.com/Koenkk/zigbee2mqtt/files/14152533/log.txt)

alwillha, 2024-02-03:
The same for Aqara RTCGQ14LM  https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html#xiaomi-rtcgq14lm

error 2024-02-03 19:31:51: Failed to configure 'Aqara-Motion-P1', attempt 1 (Error: Read 0x54ef441000491aed/1 genPowerCfg(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:466:28)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:1777:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
    
    
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14152559/log.txt)

alwillha, 2024-02-03:
Is it possible that these changes (https://github.com/Koenkk/zigbee-herdsman-converters/pull/6969) could cause the error message?

VivantSenior, 2024-02-03:
> The same for Aqara RTCGQ14LM https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html#xiaomi-rtcgq14lm

I'm having the same issues (with `genPowerCfg(["batteryVoltage"]`) for my RTCGQ14LM:

```
Failed to configure 'Stairs up motion sensor', attempt 1
(Error: Read 0x54ef4410005ff950/1 genPowerCfg(["batteryVoltage"], 
{
	"timeout": 10000,
	"disableResponse": false,
	"disableRecovery": false,
	"disableDefaultResponse": true,
	"direction": 0,
	"srcEndpoint": null,
	"reservedBits": 0,
	"manufacturerCode": null,
	"transactionSequenceNumber": null,
	"writeUndiv": false
})

failed (Timeout - 36144 - 1 - 102 - 1 - 1 after 10000ms)
 at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
 at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47)
 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
 at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
 at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:466:28)
 at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:1777:13)
 at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
 at Configure.configure (/app/lib/extension/configure.ts:120:13)
 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

bricololo44, 2024-02-03:
Same probleme for me with 
- Version 1.35.2 commit: fdc9f7d4
and
- Aqara Opple WXCJKG13LM

lukonin, 2024-02-07:
Have the same issue with Aqara E1 double key wireless switch:

> Publish 'set' 'click_mode' to 'Room Switch' failed: 'Error: Write 0x54ef4410004f6c90/1 aqaraOpple({"293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'

and

> Failed to configure 'Kitchen Switch', attempt 1 (Error: Write 0x54ef4410004f6af4/1 manuSpecificLumi({"293":{"value":2,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:3126:13) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Reverting to 1.35.1-1 did not solve the problem.
How to fix it?

alwillha, 2024-02-07:
@lukonin 
> Reverting to 1.35.1-1 did not solve the problem. How to fix it?

I reverted to Add-On Backup addon_45df7312_zigbee2mqtt_1.35.1-1. This means I have no error messages in the log file with Z2M 1.35.1-1. I wouldn't have to do an HA restart.
Maybe HA Restart would help in your case.

lukonin, 2024-02-07:
@alwillha
I reverted to addon_45df7312_zigbee2mqtt_1.35.1-1 from system backups.
Unfortunately, HA restart didn't help too.
Tried to re-pair the wall switch, same result.

alwillha, 2024-02-07:
@lukonin 
After the revert I no longer have this effect. There are no error messages in /homeassistant/zigbee2mqtt/log/current/log.txt and in Z2M WebUI.
When I update to a new version, the error messages keep coming back at certain intervals.
Since I don't want to see the error messages, I use the old version.

Are the error messages in /homeassistant/zigbee2mqtt/log/current/log.txt from the timestamp new or from the old version?
Do the error messages constantly appear in the WebUI even after the revert to the 1.35.1-1 version?

lukonin, 2024-02-07:
@alwillha, I can't change the click_mode from multi to fast, so these errors appers in some seconds after I trying to change it.
click_mode was set to "fast" before upgrade and now it is set to "multi", so I'm waiting over a second before the lights on or off.

lukonin, 2024-02-10:
@alwillha, my mistake, I have tried "reconfiguring", not "re-pairing".
With 1.35.1-1 tried to force remove the switch, re-pairing, changing click_mode. And it works!
After updating to 1.35.3-1 the click_mode changed to null. Ok.
With 1.35.3-1 tried to force remove the switch, re-pairing, changing click_mode. And it works again!

Thank you!

sjorge, 2024-02-10:
@Koenkk my guess here is, since these seem to be mostly battery device, they need the quirkCheckinInterval added. 

The lumi rename probably wants some of them te reconfigure and they are now probably using send immediately.

Koenkk, 2024-02-10:
@sjorge normally configure only triggers on pairing, now also on start (you will get an error once), but after the device sends a message configure is re-attempted and should succeed, isnt this the case?

sjorge, 2024-02-10:
> @sjorge normally configure only triggers on pairing, now also on start (you will get an error once), but after the device sends a message configure is re-attempted and should succeed, isnt this the case?

The sendWhen stuff got removed in ZH in this release, which defaults to immediately sending when the genPollCtrl cluster is not present, even on battery devices. Hence the quirkCheckinInterval was also needed for the TVOC sensor I worked on.

My assumption is that none of the Xiaomi/Aqara/Maji devices have the genPollCtrl cluster so they are probably defaulting to sending immediately. And of course if the device is sleeping it will hit a timeout as most of the errors here are (or no route, if it has not been seen in a while).

SchoutenHans, 2024-02-12:
Same issue with 1.35.3  Aqara Xiaomi switch  WXCJKG13LM  lumi.remote.b686opcn01.

-----
Z2M Server: Failed to configure 'Switch-3', attempt 3 (Error: Bind 0x54ef44100045620e/1 genOnOff from '0x00124b00239325b4/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\utils\waitress.ts:63:23) at C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:699:28 at Queue.execute (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\utils\queue.ts:35:20) at Endpoint.bind (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:551:13) at Object.bind (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\reporting.ts:43:9) at Object.configure (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\devices\lumi.ts:2611:13) at Configure.configure (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\lib\extension\configure.ts:120:13) at EventEmitter.wrappedCallback (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\lib\eventBus.ts:174:17))
------
AND  Aqara  Xiaomi  Temp/VOC sensor VOCKQJK11LM lumi.airmonitor.acn01

Z2M Server: Failed to configure 'AirQ-2', attempt 1 (Error: ConfigureReporting 0x54ef4410001a0b2c/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49029 - 1 - 230 - 1 - 7 after 10000ms) at Timeout._onTimeout (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\utils\waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))

lukonin, 2024-03-04:
> @alwillha, my mistake, I have tried "reconfiguring", not "re-pairing". With 1.35.1-1 tried to force remove the switch, re-pairing, changing click_mode. And it works! After updating to 1.35.3-1 the [...]

The same situation after updating to 1.36.0-1...

closed 2024-06-02 by Koenkk