Koenkk/zigbee2mqtt #21201: Tuya 2-button scene remote (LKWSZ211) no longer work with 1.35.2 docker image

frederickcwong, 2024-02-03:
### What happened?

Tuya [LKWSZ211](https://www.zigbee2mqtt.io/devices/LKWSZ211.html#tuya-lkwsz211) no longer working after updated to docker image 1.35.2 (was working in 1.35.1). Please see the debug log below from 1.35.1 and 1.35.2 when a button is pushed. I also confirmed the dev branch docker image has the same issue. Is the message converter missing from the docker build?

### What did you expect to happen?

I expected that the Tuya buttons continue to work after the upgrade from 1.35.1 to 1.35.2

### How to reproduce it (minimal and precise)

Click the buttons and no response from Home Assistant, review the log and there is no message sent to the MQTT.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Docker with Home Assistant (latest stable release) and MQTT (Eclipse broker) on x86-64 nuc

### Debug log

**Debug Log from 1.35.2:**
```
2024-02-03 11:41:40 Received Zigbee message from 'Office Desk Button', type 'raw', cluster 'genOnOff', data '{"data":[1,55,253,0],"type":"Buffer"}' from endpoint 2 with groupID null
2024-02-03 11:41:40 No converter available for 'LKWSZ211' with cluster 'genOnOff' and type 'raw' and data '{"data":[1,55,253,0],"type":"Buffer"}'
```

**Debug Log from 1.35.1:**
```
2024-02-03 11:47:08 Received Zigbee message from 'Office Desk Button', type 'raw', cluster 'genOnOff', data '{"data":[1,58,253,0],"type":"Buffer"}' from endpoint 1 with groupID null
2024-02-03 11:47:08 MQTT publish: topic 'zigbee2mqtt/Office Desk Button', payload '{"action":"1_single","battery":null,"linkquality":255}'
2024-02-03 11:47:08 MQTT publish: topic 'zigbee2mqtt/Office Desk Button', payload '{"action":"","battery":null,"linkquality":255}'
2024-02-03 11:47:08 MQTT publish: topic 'zigbee2mqtt/Office Desk Button/action', payload '1_single'
```

reflectivecode, 2024-02-03:
I can confirm. I am having the same issue on version 1.35.2.

Koenkk, 2024-02-04:
This should already be fixed in the  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

BitWuehler, 2024-02-04:
I had a similar problem. The "TuYa IH-K663" Button never worked fully with home assistant, so that i used the MQTT Payloads. The last update changed the payload from "single" and "double" to "1_single" and "1_double". In my installation easy to fix but in other you have to change the config here.

frederickcwong, 2024-02-04:
> This should already be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hi, fyi the package update does not fix the problem with the Tuya 2-button remote. Thanks.

Koenkk, 2024-02-04:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/21196

closed 2024-02-04 by Koenkk

frederickcwong, 2024-02-07:
Confirm 1.35.3 fixed the issue for the device. Thanks!