Koenkk/zigbee2mqtt #21202: [New device support]: Nedis ZBHTR20WT

nilsmahlstaedt, 2024-02-03:
### Link

https://nedis.de/de-de/product/haushalt-und-wohnen/klima/heizung/550786152/smartlife-heizkorpersteuerung-zigbee-30-batteriebetrieben-led-android-ios

### Database entry

{   "id": 2,   "type": "EndDevice",   "ieeeAddr": "0xa4c138a9adee5729",   "nwkAddr": 51184,   "manufId": 4417,   "manufName": "_TZE200_ne4pikwm",   "powerSource": "Battery",   "modelId": "TS0601",   "epList": [     1   ],   "endpoints": {     "1": {       "profId": 260,       "epId": 1,       "devId": 81,       "inClusterList": [         4,         5,         61184,         0       ],       "outClusterList": [         25,         10       ],       "clusters": {         "genBasic": {           "attributes": {             "65503": "\u0000\u0000\u0000\u0000\u0011\u0004\u0000\u0000\u0000f~9Q-\u0012",             "65506": 55,             "65508": 0,             "65534": 0,             "stackVersion": 0,             "dateCode": "",             "manufacturerName": "_TZE200_ne4pikwm",             "zclVersion": 3,             "appVersion": 67,             "modelId": "TS0601",             "powerSource": 3           }         }       },       "binds": [],       "configuredReportings": [],       "meta": {}     }   },   "appVersion": 67,   "stackVersion": 0,   "hwVersion": 1,   "dateCode": "",   "zclVersion": 3,   "interviewCompleted": true,   "meta": {     "configured": -708457359   },   }

### Comments

Hi, today i spend the day trying to get a Nedis ZBHTR20WT Smart Thermostatic Radiator Value working.

As i don't have access to a tuya gateway, i cannot perform the recommended steps (see [supporting new tuya devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html)) and had to rely on guesswork and testing with implementations of other tuya converters for comparable (maybe whitelabel) products.

This is as far as i got with the adapted external definition (see below). 
[image]

I need help from someone who maybe has a working setup to get extended functions like Battery State, Anti-Scaling, Window Open Sensor going.

Testing State:
- Current Temperatur (works and verified)
- Setpoint Temp (works and verified)
- Child Lock (works and verified)
- Window Open (shown, could not be verified)
- Anti Feeeze (shown, could not be verified - it's currently not cold enough for that)
- Local Temperature Calibration (shown, not verified)

If someone is currently tinkering with these, I hope writing my current progress down will help you completing this integration. My current feeling is, that the Nedis TRV is just a whitelabel of one of the below products.
Sadly they seem to have scambled the datapoint id's. Once these are known, you can probably copy and paste all the converters from existing implementations.

Comparable Products used as sources and inspiration for testing:
- [Siterwell](https://github.com/Koenkk/zigbee-herdsman-converters/blob/1761d8bf9bfabb6143ec14e6790aae5aee7a3a69/src/devices/siterwell.ts)
- [Sonoff TRVZB](https://github.com/Koenkk/zigbee-herdsman-converters/blob/1761d8bf9bfabb6143ec14e6790aae5aee7a3a69/src/devices/sonoff.ts#L532)
- [Tuya TV02](https://github.com/Koenkk/zigbee-herdsman-converters/blob/1761d8bf9bfabb6143ec14e6790aae5aee7a3a69/src/devices/tuya.ts#L3062)
- [Tuya TS0601_thermostat_2 to 4](https://github.com/Koenkk/zigbee-herdsman-converters/blob/1761d8bf9bfabb6143ec14e6790aae5aee7a3a69/src/devices/tuya.ts#L3165)

PS: I had some Datapoints I could not make heads or tails from. Log Lines from Debug log are in Definition as comments.

### External defintion

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
        fingerprint: tuya.fingerprint('TS0601', [
            '_TZE200_ne4pikwm', /* modell: 'ZBHTR20WT', vendor: 'Nedis' */
        ]),
        model: 'ZBTHR20WT',
        vendor: 'Nedis',
        description: '(TESTING) Thermostat radiator valve',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetLocalTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.battery_low(), 
            e.child_lock(), 
            e.open_window(), 
            //e.open_window_temperature().withValueMin(5).withValueMax(30),
            e.climate()
                .withLocalTemperatureCalibration(-5, 5, 0.1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET),
            tuya.exposes.frostProtection('When Anti-Freezing function is activated, the temperature in the house is kept '+
                    'at 8 °C, the device display "AF".press the pair button to cancel.'),
            // what about anti scale?
            e.binary('online', ea.STATE_SET, 'ON', 'OFF').withDescription('The current data request from the device.'),
            tuya.exposes.errorStatus(),
        ],
        meta: {
            tuyaDatapoints: [
                [2, 'preset', tuya.valueConverter.tv02Preset()],
                    //Datapoint 3 not defined for '_TZE200_ne4pikwm' with value 1
                [8, 'open_window', tuya.valueConverter.onOff],
                [10, 'frost_protection', tuya.valueConverter.TV02FrostProtection],
                //[24, 'local_temperature', tuya.valueConverter.divideBy10],
                [27, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
                //[35, 'battery_low', tuya.valueConverter.trueFalse0],
                [40, 'child_lock', tuya.valueConverter.lockUnlock],
                //[45, 'error_status', tuya.valueConverter.raw],
                [101, 'schedule_mode', tuya.valueConverter.onOff],
                [102, 'local_temperature', tuya.valueConverter.divideBy10],
                [103, 'current_heating_setpoint', tuya.valueConverter.divideBy10],    
                [107, 'system_mode', tuya.valueConverter.TV02SystemMode],
                [107, 'heating_stop', tuya.valueConverter.TV02SystemMode],
                    //Datapoint 110 not defined for '_TZE200_ne4pikwm' with value 
                    //"dpValues":[{"data":{"data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,21,15,0,0,0,0,0],"type":"Buffer"},"datatype":0,"dp":110}],"seq":3328}
                [115, 'online', tuya.valueConverter.onOffNotStrict],
                    //Datapoint 119 not defined for '_TZE200_ne4pikwm' with value
                    //{"dpValues":[{"data":{"data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,9,2,0,0,0,0,0],"type":"Buffer"},"datatype":0,"dp":119}],"seq":4352}

                // Datapoint 132 not defined for _ with value 8 {"dpValues":[{"data":{"data":[1,24,2,3,18,56,0,6],"type":"Buffer"},"datatype":0,"dp":132}],"seq":7680}
            ],
        },
    } 

module.exports = definition;
```

a42wg, 2024-03-13:
Hi I have used your recipe and it works fine (there are missing a ; after the last curly bracket in blue print above). I was primarily missing the battery level, so I hooked my device up to the Tuya platform to extract the data points.  

Here is a compiled list of all DP-codes from the Tuya developer page. Not all data points have entry's in my log, it have only been running for about 24 hours.
The data types in the table are somehow inconsistent, the types are extracted from the data point description, but  for instance DP 105 Battery level alarm are listed as a bitmap type, but the value in the log is "0".

Most of the historical data seems to share the same data point in the FW, but the names is mostly in chinese and are translated using google, so that may also introduce some uncertainty. 

br Anders

[ZBTHR20WT Codes.pdf](https://github.com/Koenkk/zigbee2mqtt/files/14591996/ZBTHR20WT.Codes.pdf)

fridtjof, 2024-04-14:
Also interested in this.
When I joined a thermostat with this code, it somehow enabled the schedule feature. I can see that supposedly z2m can expose a schedule on/off to Home Assistant, but when I tried adding it (through `e.schedule()`), it did not end up working because tuya's onOff converter deals in "ON"/"OFF", whereas e.schedule is a Binary with values true/false.

In the meantime, is there another way to manually send a value to a tuya datapoint (to turn the schedule off)? Whatever schedule's built into the device as a default is pretty annoying and I'd rather have it function as a "dumb" thermostat for now...

inFLOmatic, 2024-09-22:
A bit of issue necromancy on my side but I happend to buy he nedis thermostat a few days ago.
I made the schedule switch work by assigning `schedule_mode` to the data point 108
`[108, 'schedule_mode', tuya.valueConverter.onOff],`
and exposing it with 
`e.binary('schedule_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Should the device be on the heating schedule')`

Using the schedule is still not possible and the thermostat is clearly on a wrong time setting (the default schedule did things in the middle of the night). But thats no issue on my side as I'm using Home Assistant to do those things anyway

Aimeroo, 2024-11-18:
Hi there, I added it to my HA install and it works like a charm. Thanks for that!
One question though: 
The device has a push button on the wheel 
When you push it, it resets to 16°C.

Is there a way to configure that to 19°C? 
If yes, where? :)

AdrianSchmidt91, 2024-11-20:
Works fine.
I had to remove `const extend = require('zigbee-herdsman-converters/lib/extend');` in order to get it working.

Aimeroo, 2024-11-21:
@nilsmahlstaedt I have a question for the "Window Open" datapoint
Should it do anything when triggered? Because if it is on, the thermostat just keeps on doing what it did before

castn, 2024-11-24:
How did you get your thermostats paired with z2m? I am already struggling with this point. I didn't get to the point where I could enter the definition from the first message.

Aimeroo, 2024-11-24:
@nilsmahlstaedt ah ok. Thanks, I thought as much. 

I opted for the Sonoff TRV instead as they are natively supported :)

@castn
mount it on the radiator, turn it to "OF(F)", put z2m into joining mode and press the dial button for 5 seconds

Aimeroo, 2024-11-25:
@castn 
in the Device in z2m you click under "Dev console" tab on "generate_external_definition"
Then you take the code from above, add the semicolon that is missing on the last swirly bracket, remove that line: "const extend = require('zigbee-herdsman-converters/lib/extend');" and create a new .js file under /homeassistant/zigbee2mqtt/ and put the code in it
For example "/homeassistant/zigbee2mqtt/ZBHTR20WT.js"

Then in the configuration.yaml in the /homeassistant/zigbee2mqtt folder you add this:
external_converters:
  - ZBHTR20WT.js

castn, 2024-11-25:
@Aimeroo thanks. I did everything you explained, which is in the documentation as well. I think is messed something up in the file because the one from @KoenGoe worked for me.

castn, 2024-11-25:
Did anyone create a Pull Request with the working file.

z3roq, 2024-11-27:
I have this issue with this that i think is related to pre configured stuff in both thermostats. 

At two specific time of day temperature will set to 16C even if i have schedule off. Also whenever i try to change any options (schedule, child lock or else) thermostat will go to off mode and temperatures i set via z2m wont take effect until i remove child lock and manually screw thermostat on

z3roq, 2024-12-03:
I got figured out that one above. User error.

With some try and error i got thermostat state working

expose in e.climate:
.withRunningState(['idle', 'heat'], ea.STATE)

and datapoing:
[3, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],

Now it shows idle and heat correctly.

 @a42wg have you had a chance to map these further?

z3roq, 2024-12-07:
I have got almost everything working except schedule but who needs it anyway. Limescale function and battery level i have not been able to verify, but system mode and running states are now working correctly so if You put thermostat off, it goes off and if heating, running states will show when it heats or idles correctly ( it was flipped in above ). Also leave home is now added and drops temp to 16 when activated and returns to set when deactivated. Did not find any way to set home away temperature even with quite long logging time and i do not have TuYa gateway to map those further. 

Also cleaned that bit and made some descriptions. Now it is completely usable with HA

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'ZBTHR20WT',
    vendor: '_TZE200_ne4pikwm',
    description: '(TESTING) Thermostat radiator valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime,
    configure: tuya.configureMagicPacket,
    extend: [],
    exposes: [
        e.battery_low(),
        e.child_lock(), 
        e.open_window(), 
        e.climate()
            .withLocalTemperatureCalibration(-5, 5, 0.1, ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET),
        tuya.exposes.frostProtection('This function prevents freezing of the radiator. It automatically switches on the thermostat between 5°C and 8°C.'),
        e.binary('schedule_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Should the device be on the heating schedule'),
        e.binary('scale_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('The radiator can scale and become clogged if the valve is not opened regularly. This function opens the valve for 30 seconds every two weeks. The display shows “Rd” during this procedure.'),
        e.binary('leave_home', ea.STATE_SET, 'ON', 'OFF').withDescription('Temperature drops to 16°C when activated and restores when off' ),
        tuya.exposes.errorStatus(),
        
    ],
    meta: {
        tuyaDatapoints: [
            [3, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(1), 'idle': tuya.enum(0)})],
            [8, 'open_window', tuya.valueConverter.onOff],
            [10, 'frost_protection', tuya.valueConverter.TV02FrostProtection],
            [27, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [101, 'system_mode', tuya.valueConverterBasic.lookup({'heat': true, 'off': false})],
            [102, 'local_temperature', tuya.valueConverter.divideBy10],
            [103, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [105, 'battery_low', tuya.valueConverter.trueFalse1],  // Not sure if works but no null atm. DP is bitmap?
            [106, 'leave_home', tuya.valueConverter.onOff],
            [108, 'schedule_mode', tuya.valueConverter.onOff],
            [130, 'scale_protection', tuya.valueConverter.onOff], // Not verified, but DP should match
        ],
    },
};

module.exports = definition;
`

morph027, 2024-12-07:
Nice, good work. Just got 2 of those and can confirm it's working.

Regarding modes, you're only exposing "Off" and "Heat" ... Couldn't extract any useful information from https://cdn.nedis.com/manuals/html/ZBHTR20WT/?locale=en-GB if "Heat" means something like "Auto" on other HVACs, so it's only heating when temperature is below threshold (it should).

morph027, 2024-12-07:
Alright, no problem. I guess it's okay. I have a Mill heater which also behaves the same, so it only has "Off" and "Heat" but "Heat" will take care of this itself. So all good. Glad you did the work, so no complaining.

Will test drive the 2 HVACs further.

morph027, 2024-12-07:
Jip, HA is working in Heat mode.

[image]
[image]

TAKeanice, 2024-12-08:
> I have got almost everything working except schedule but who needs it anyway. Limescale function and battery level i have not been able to verify, but system mode and running states are now working [...]

That file is working for me, however I noticed some behavior I wasn´t able to explain, and I don´t know for sure where it came from:
- Pressing the button on the thermostat turns on schedule mode, previously it only turned on the display
- Window open mode activated today morning when ventilating, but didn´t turn off any more until now, a couple hours later
- The thermostats seem to randomly jump to 16°C from time to time, could not find a correlation to anything else. Looks similar to what @z3roq found:
> At two specific time of day temperature will set to 16C even if i have schedule off.

> Did anyone create a Pull Request with the working file.
@castn I could do that if you want. Do you all think this is at a sufficient level of quality for being included?

z3roq, 2024-12-08:
@TAKeanice I think that You somehow manage to switch schedule mode on because default schedule mode sets it to 16C couple times a day. Try to put child lock on, then switch schedule on and then after that again off and wait if it works then. This does not change the way how thermostat button will work so if you accidentally push it when You are adjusting temp manually, You could enable scheduling. First press wakes thermostat and when it is wake, button will enable schedule. DP is right for that at least with my thermostat when i put it on it shows clock on thermostat and so on. I suggest controlling thermostat 100% remote. 

For qualitywise, maybe if someone could verify that low battery detection and limescale function? As i told before, my experience with these are very limited and got that working just connecting dots.

TAKeanice, 2024-12-08:
Thank you for the advice, they have not reset themselves for a while now. What do you think about the open window state?

z3roq, 2024-12-08:
@TAKeanice That window open detection is happening inside device and can only be turned on or off. There was no other adjustments to it in datapoints provided by PDF above.

TAKeanice, 2024-12-09:
Thank you, I thought from another thermostat that there is another datapoint that tells if the window open detection was triggered and one that is for setting how long it should stop heating if it happens. Maybe this one really is steered via some hub-based automation.

@KoenGoe 
> ## Excessive heating
> 
> The problem that I have with this thing, is  that it seems to turn on even when its measured temperature is above the target temperature. 
> 
> Is anyone else seeing such behavior?
> 
> Or does anyone have tips on how to deal with this?
> 

I have seen that the thermostat heats well beyond its preset temperature (about 1°C). I will experiment with setting the offset one degree higher as a workaround.
I suspect there could be some setting to limit the amount of overshooting. @a42wg 's datapoints show some heating time (#104). Has anyone figured out what that could mean?

z3roq, 2024-12-09:
Here is my heating behaviour: 
[image]

Temperature is set to 20 and it starts heating well before it drops there. Water temp is around 80 so heating cycles are not very long even if it is cold outside. Temperature sensor in other side of room shows about 20,5 all day with little deviation so i say this is pretty accurate. Calibration is set to 0

z3roq, 2024-12-09:
Little update. I noticed that frost protection did not "answer" when put off so i changed it to binary and now it works like other switches. Also i tested temperature calibration and it did not work and had wrong range. It is also now fixed. I try to figure out that heating time what is it and if i do not get it figured out by tomorrow i will post updated code for those two fixes.

TAKeanice, 2024-12-17:
I have tried to contact the Nedis support and had some contact with friendly people (or chatbots, who knows nowadays), but unfortunately beyond pointing me to the Tuya Smart App they wouldn't give me any information on the datapoints.

TAKeanice, 2024-12-22:
I have created a PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8504 and would recommend that you submit your changes there if you want to suggest a fix or extension of the converter.

closed 2024-12-23 by Koenkk