Koenkk/zigbee2mqtt #21203: [ZB-SW03] Error after Z2M restart

andrejsoucek, 2024-02-03:
### What happened?

In the last couple of month I need to restart all of my [wall switches](https://www.zigbee2mqtt.io/devices/ZB-SW03.html#ewelink-zb-sw03) stop working after Z2M restart. Configuration fails and commands time out.

When I restart it by cutting of the power and powering it again, everything is good.

This might be related: https://github.com/Koenkk/zigbee2mqtt/issues/20600

### What did you expect to happen?

Device should configure automatically without any probelm after Z2M restart.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

HA OS

### Debug log

```
Failed to configure '<device name>', attempt 3 (Error: ConfigureReporting 0x00124b0024701621/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 61472 - 1 - 62 - 6 - 7 after 10000ms) 
at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) 
at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:486:47)
at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) 
at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:669:28) 
at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:55:13) 
at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:126:13) 
at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

andrejsoucek, 2024-02-04:
Looking at the fix in #20483 would it be fixed by adding `configureReporting: false`? The `extend` property is different for ZB-SW01 and the others (02, 03, 04, 05...).

Koenkk, 2024-02-05:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-05 by Koenkk

cipry85, 2024-02-09:
is it the same problem to #21233 ?