Koenkk/zigbee2mqtt #21206: [New device support]: Samotech SM325 (partially done). Zigbee Smart pull cord dimmer switch

alexeiw123, 2024-02-04:
### Link

https://www.samotech.co.uk/products/zigbee-ceiling-switch-pull-cord/

### Database entry

I'm not sure where to find this

### Comments

I have followed the instructions as best I can to the point that I have a working light entity with dimmer controls.
I have taken pieces of code from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/samotech.ts to extend the external definition so ythe dimmer controls work.

This device does say that it is supported here: https://zigbee.blakadder.com/Samotech_SM325.html
But it wasn't for me (running latest dev build).

Here are the contents of the my file SM325.js:

```
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SM325-ZG'],
    model: 'SM325-ZG',
    vendor: 'Samotech',
    description: 'Zigbee Smart Pull Cord Dimmer Switch',
    fromZigbee: extend.light_onoff_brightness().fromZigbee,
    toZigbee: extend.light_onoff_brightness().toZigbee,
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
    exposes: extend.light_onoff_brightness().exposes,
};

module.exports = definition;
```

State and brightness are working. There are other exposes such as effect and power-on behaviour which do not work, but I don't know how to remove them.
I also expect there is a way to set the minimum and maximum brightness levels, which I can't see how.

[image]

Would anyone be able to finish this off and get the picture added and merge? I've never done any pull/merge before and don't have the time to learn it at present. I feel that it is pretty close.

### External defintion

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SM325-ZG'],
    model: 'SM325-ZG',
    vendor: 'Samotech',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```
```

alexeiw123, 2024-02-18:
I've now got this to remove the exposes for power on behaviour and effects which weren't working.
This is fully functional, I just need to know how to add it to Z2MQTT for others.

There's probably a way to configure the min brightness level which can be done from the hardware directly. Beyond me though.

```
const { onOff } = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const lightOnOffBrightness = extend.light_onoff_brightness({
    disablePowerOnBehavior: true,
    disableEffect: true,
});

const definition = {
    zigbeeModel: ['SM325-ZG'],
    model: 'SM325-ZG',
    vendor: 'Samotech',
    description: 'Zigbee Smart Pull Cord Dimmer Switch',
    fromZigbee: lightOnOffBrightness.fromZigbee,
    toZigbee: lightOnOffBrightness.toZigbee,
    configure: async (device, coordinatorEndpoint, logger) => {
        await lightOnOffBrightness.configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
    exposes: lightOnOffBrightness.exposes,
};

module.exports = definition;
```

alexeiw123, 2024-02-19:
OK - so I've been trying to learn something new here. I _think_ I've done this correctly.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7088
https://github.com/Koenkk/zigbee2mqtt.io/pull/2581

Welcoming any help and/or feedback

alexeiw123, 2024-02-19:
support added and merged here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/7088

closed 2024-02-19 by alexeiw123

ukmgranger, 2025-03-01:
Not sure if this is the correct place to ask this, but I'm struggling to find answers.
I've just added this device to Z2M, but it doesn't seem to expose brightness in the Home Assistant entity - only on/off.  It has brightness in Z2M.
Any thoughts

alexeiw123, 2025-03-01:
> Not sure if this is the correct place to ask this, but I'm struggling to find answers. I've just added this device to Z2M, but it doesn't seem to expose brightness in the Home Assistant entity - only on/off. It has brightness in Z2M. Any thoughts

What version of HA core are you on?

It sounds very likely to be related to this:
https://github.com/Koenkk/zigbee2mqtt/issues/26520

ukmgranger, 2025-03-02:
> What version of HA core are you on?
>
> It sounds very likely to be related to this: [#26520](https://github.com/Koenkk/zigbee2mqtt/issues/26520)
Cheers for the heads up!  I'm on 2025.3.0b2

Looks very likely that is the issue.  Also spotted that many of my other dimmable lights are now just on/off too.

alexeiw123, 2025-03-02:
> Cheers for the heads up! I'm on 2025.3.0b2
>
> Looks very likely that is the issue. Also spotted that many of my other dimmable lights are now just on/off too.

There's a Z2M update released now that should fix it