Koenkk/zigbee2mqtt #21208: Bosch BTH-RA "window_open" entity not available in Z2M 1.35.2

ylieder, 2024-02-04:
### What happened?

In HA, the entity "window_open" is no longer available for my Bosch BTH-RA thermostats. In zigbee2mqtt it is still shown. Also in the MQTT messages, it is visible.

Worked in 1.35.0 and 1.35.1 docker images, but no longer in 1.35.2 and latest-dev.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

### What did you expect to happen?

The "window_open" switch entity is visible in HA.

### How to reproduce it (minimal and precise)

- Delete MQTT connection from HA (Deletes all devices and entities)
- Shutdown Z2M docker container
- Start Z2M docker container with specified version
- Setup MQTT integration in HA

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

20210708

### Adapter

SONOFF CC2652P ZBDongle-P

### Setup

Docker container

ylieder, 2024-02-04:
The behavior was already mentioned in this issue: https://github.com/Koenkk/zigbee2mqtt/issues/20707#issuecomment-1920140304

danielpetrovic, 2024-02-05:
It seems to be an error in the config topic provided by Zigbee2MQTT, I have the Bosch BTH-RM which had the same switch and every reboot of HA I get this log now:

`Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x001e5e0902910e5b/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'command_topic': 'zigbee2mqtt/Woonkamer Thermostaat/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x001e5e0902910e5b'], 'manufacturer': 'Bosch', 'model': 'Room thermostat II (Battery model) (BTH-RM)', 'name': 'Woonkamer Thermostaat', 'via_device': 'zigbee2mqtt_bridge_0x00124b0030eb5c72'}, 'device_class': 'window', 'object_id': 'woonkamer_thermostaat_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.2-dev', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Woonkamer Thermostaat', 'unique_id': '0x001e5e0902910e5b_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'

Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x001e5e09029103c1/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'command_topic': 'zigbee2mqtt/Badkamer Thermostaat/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x001e5e09029103c1'], 'manufacturer': 'Bosch', 'model': 'Room thermostat II (Battery model) (BTH-RM)', 'name': 'Badkamer Thermostaat', 'via_device': 'zigbee2mqtt_bridge_0x00124b0030eb5c72'}, 'device_class': 'window', 'object_id': 'badkamer_thermostaat_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.2-dev', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Badkamer Thermostaat', 'unique_id': '0x001e5e09029103c1_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'

Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x001e5e090290f15d/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'command_topic': 'zigbee2mqtt/Gameroom Thermostaat/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x001e5e090290f15d'], 'manufacturer': 'Bosch', 'model': 'Room thermostat II (Battery model) (BTH-RM)', 'name': 'Gameroom Thermostaat', 'via_device': 'zigbee2mqtt_bridge_0x00124b0030eb5c72'}, 'device_class': 'window', 'object_id': 'gameroom_thermostaat_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.2-dev', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Gameroom Thermostaat', 'unique_id': '0x001e5e090290f15d_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'

Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x001e5e090290f1cb/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'command_topic': 'zigbee2mqtt/Kantoor Thermostaat/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x001e5e090290f1cb'], 'manufacturer': 'Bosch', 'model': 'Room thermostat II (Battery model) (BTH-RM)', 'name': 'Kantoor Thermostaat', 'via_device': 'zigbee2mqtt_bridge_0x00124b0030eb5c72'}, 'device_class': 'window', 'object_id': 'kantoor_thermostaat_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.2-dev', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Kantoor Thermostaat', 'unique_id': '0x001e5e090290f1cb_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'

Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x001e5e090290e78d/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'command_topic': 'zigbee2mqtt/Sauna Thermostaat/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x001e5e090290e78d'], 'manufacturer': 'Bosch', 'model': 'Room thermostat II (Battery model) (BTH-RM)', 'name': 'Sauna Thermostaat', 'via_device': 'zigbee2mqtt_bridge_0x00124b0030eb5c72'}, 'device_class': 'window', 'object_id': 'sauna_thermostaat_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.2-dev', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Sauna Thermostaat', 'unique_id': '0x001e5e090290e78d_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}' `

The switch has since disappeared from HA. But works fine when switched within Z2M itself.

GatoPharaoh, 2024-02-05:
oh, that log sheds some light i guess. i think the issue is due to this commit [ac74fac](https://github.com/Koenkk/zigbee2mqtt/commit/ac74fac1db9b1ee6d473c5045989003548e451f6).

in line 654 `window_open: {device_class: 'window'}` was added - as window_open is a switch, it cannot be `device_class: 'window` as switches can only be `none`, `outlet `or `switch` if i unterstand the [docs ](https://www.home-assistant.io/integrations/switch/#device-class)correctly

GatoPharaoh, 2024-02-05:
for aqara devices, the "window switch" is called `window_detection`. for bosch devices it is called `window_open`

windows_detection in line 653 is defined as `window_detection: {icon: 'mdi:window-open-variant'}`

NicoDreamzZ, 2024-02-07:
> It seems to be an error in the config topic provided by Zigbee2MQTT, I have the Bosch BTH-RM which had the same switch and every reboot of HA I get this log now:
>
> `Error 'expected SwitchDeviceClass [...]

I have the same errors in homeassistant and the manual window open switch is marked as unavailable in homeassistant. In Z2M it is working fine.

```Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:270
Integration: MQTT (documentation, issues)
First occurred: 00:16:10 (8 occurrences)
Last logged: 00:23:09

Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x18fc26000005707b/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Thermostat Küche/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Thermostat Küche/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x18fc26000005707b'], 'manufacturer': 'Bosch', 'model': 'Radiator thermostat II (BTH-RA)', 'name': 'Thermostat Küche', 'sw_version': '3.05.09', 'via_device': 'zigbee2mqtt_bridge_0x00124b0021cc3eb1'}, 'device_class': 'window', 'object_id': 'thermostat_küche_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.3', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Thermostat Küche', 'unique_id': '0x18fc26000005707b_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'
Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x18fc26000006d941/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Thermostat Schlafzimmer/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Thermostat Schlafzimmer/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x18fc26000006d941'], 'manufacturer': 'Bosch', 'model': 'Radiator thermostat II (BTH-RA)', 'name': 'Thermostat Schlafzimmer', 'sw_version': '3.05.09', 'via_device': 'zigbee2mqtt_bridge_0x00124b0021cc3eb1'}, 'device_class': 'window', 'object_id': 'thermostat_schlafzimmer_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.3', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Thermostat Schlafzimmer', 'unique_id': '0x18fc26000006d941_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'
Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x18fc26000006f66f/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Thermostat Bad/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Thermostat Bad/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x18fc26000006f66f'], 'manufacturer': 'Bosch', 'model': 'Radiator thermostat II (BTH-RA)', 'name': 'Thermostat Bad', 'sw_version': '3.05.09', 'via_device': 'zigbee2mqtt_bridge_0x00124b0021cc3eb1'}, 'device_class': 'window', 'object_id': 'thermostat_bad_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.3', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Thermostat Bad', 'unique_id': '0x18fc26000006f66f_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'
Error 'expected SwitchDeviceClass or one of 'outlet', 'switch' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/switch/0x18fc2600000ac570/switch_window_open/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Thermostat Wohnzimmer/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Thermostat Wohnzimmer/set/window_open', 'device': {'identifiers': ['zigbee2mqtt_0x18fc2600000ac570'], 'manufacturer': 'Bosch', 'model': 'Radiator thermostat II (BTH-RA)', 'name': 'Thermostat Wohnzimmer', 'sw_version': '3.05.09', 'via_device': 'zigbee2mqtt_bridge_0x00124b0021cc3eb1'}, 'device_class': 'window', 'object_id': 'thermostat_wohnzimmer_window_open', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.35.3', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Thermostat Wohnzimmer', 'unique_id': '0x18fc2600000ac570_switch_window_open_zigbee2mqtt', 'value_template': '{{ value_json.window_open }}', 'platform': 'mqtt'}'

simplyRoba, 2024-02-09:
This PR should fix this https://github.com/Koenkk/zigbee-herdsman-converters/pull/7029

danielpetrovic, 2024-02-09:
> This PR should fix this [Koenkk/zigbee-herdsman-converters#7029](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7029)

No it does not. The PR you are linking to is not related to that. It is related to accidentally including the Heating Valve which is not present in the Battery version of the Room Thermostat.

danielpetrovic, 2024-02-11:
Fixed with next release. If you run Edge version now the fix is there.

dasEtwasmitPunkten, 2024-02-12:
> Fixed with next release. If you run Edge version now the fix is there.

Do you know, when the next release is? Or is there any roadmap about it?

danielpetrovic, 2024-02-17:
> The same probs here.

If you read above, you’d know it is fixed in next release (1.36). If you run Edge now, you already have the fix.

danielpetrovic, 2024-02-17:
> > Fixed with next release. If you run Edge version now the fix is there.
> 
> Do you know, when the next release is? Or is there any roadmap about it?

Not sure, the bug fix releases seem to be once a week. But the main release is once a month. In the Pull Requests I can see Koenkk is readying 1.36 release.

salopette, 2024-03-01:
Zigbee2MQTT-Version
[1.36.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator-Typ
zStack3x0
Coordinator-Version
20230507
Coordinator IEEE Adresse
0x001ccccscsdsdwsd8ea
Frontend-Version
0.6.159
zigbee_herdsman_converters_version
18.42.0
zigbee_herdsman_version
0.35.1

Failed to configure 'Bosch-TRV-Schlafzimmer', attempt 3 (Error: Bind 0x18fc2600000ec473/1 genPowerCfg from '0x00124b001ca168ea/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:692:28 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/bosch.ts:1111:13) at Configure.configure (/app/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

danielpetrovic, 2024-03-01:
This error has nothing to do with the other change. Please make a new bug report for it. I don't have the TRV so cannot test it for you.

danielpetrovic, 2024-03-01:
> Zigbee2MQTT-Version [1.36.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown) Coordinator-Typ zStack3x0 [...]

This error has nothing to do with the other changes. Please make a new bug report for it. I don't have the TRV so cannot test it for you.

NicoDreamzZ, 2024-03-02:
The new version is released and works fine, no more errors with the window switch, be aware that the name of the entity has changed, so check your automations in homeassistant :)

coxtor, 2024-03-08:
Hi, I am confused about the feature window detection in the bosh radiator 2 device. Does the window_detection option disable/enable window detection by the thermostat, or does enabling the switch enable window open mode that essentially disables the thermostat untill the switch is turned off?

danielpetrovic, 2024-03-08:
@ylieder can you close this issue? As it is solved.

coxtor, 2024-03-08:
thank you @danielpetrovic  . Just to clarify: For my automation a reasonable function would be to check wether the window is open and toggle the thermostat switch accordingly ?

danielpetrovic, 2024-03-08:
> thank you @danielpetrovic . Just to clarify: For my automation a reasonable function would be to check wether the window is open and toggle the thermostat switch accordingly ?

Yep, I would use it like:
If window sensor is open/on -> turn on window_open switch
If window sensor is closed/off -> turn off window_open switch

closed 2024-03-08 by ylieder