Koenkk/zigbee2mqtt #21210: [New device support]: Paulmann Gent 2 (501.40)

BenRyde, 2024-02-04:
### Link

https://de.paulmann.com/p/fernbedienung-gent-2-weiss/50140

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0x9035eafffe0ffdb5","nwkAddr":19218,"manufId":4632,"manufName":"Paulmann Licht GmbH","powerSource":"Battery","modelId":"501.40","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"modelId":"501.40","manufacturerName":"Paulmann Licht GmbH","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":65,"dateCode":"NULL","swBuildId":"2.7.6_r22"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd44b0","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd44b0","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}},"3":{"profId":260,"epId":3,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd44b0","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}},"4":{"profId":260,"epId":4,"devId":261,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd44b0","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":65,"dateCode":"NULL","swBuildId":"2.7.6_r22","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1707046717097}

### Comments

I thought it must be similar to this device:

https://www.zigbee2mqtt.io/devices/4512706.html

### External defintion

```shell
const {deviceEndpoints, batteryPercentage, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['501.40'],
    model: '501.40',
    vendor: 'Paulmann Licht GmbH',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4}}), batteryPercentage(), identify()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

EpicLPer, 2024-03-13:
Are there any potential updates on this? Currently considering to buy this remote myself as it has a massive assortment of buttons. It also seems to be one of the "rare" ones with this many for whatever reason.

corve001, 2024-03-14:
I allready have this remote, I could make it work for allmost all buttons, except the wheel color. I don't know why I can't see any debug information in the logfiles for this button. It's too complicated for me...

But the external file that is working for all other commands is :
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const {deviceEndpoints, batteryPercentage, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['501.40'],
    model: '501.40',
    vendor: 'Paulmann Licht GmbH',
    description: 'RGB Remote Control Gent 2',
	extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4}})],
  	fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_step_color_temperature,
		fz.command_recall, fz.command_move_to_color_temp, fz.battery, fz.command_move_to_hue,fz.command_move_to_saturation,],
    toZigbee: [],
    exposes: [e.battery(), e.action([
		'on', 'off', 'brightness_step_up', 'brightness_step_down', 'color_temperature_step_up',
        'color_temperature_step_down', 'recall_*', 'color_temperature_move',
        'move_to_hue_l1', 'move_to_hue_l2', 'move_to_hue_l3', 'move_to_hue_l4'])], // <-- this will define which fields will be exposed in the definition message to configure a frontend (e.g. the Z2M frontend, Home Assistant, Domoticz)
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

corve001, 2024-03-26:
Hi,

I can now access to the log of Zigbee-hermann.

The command of the RGB wheel that is not understand is "moveToHueAndSaturation".

here the log, when I send a color with the remote : 
`zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":45825,"clusterid":768,"srcaddr":14003,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":69,"securityuse":0,"timestamp":12713892,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[1,78,6,249,254,6,0,0,0]}} +3s`

`zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":78,"manufacturerCode":null,"commandIdentifier":6},"Payload":{"hue":249,"saturation":254,"transtime":6},"Command":{"ID":6,"parameters":[{"name":"hue","type":32},{"name":"saturation","type":32},{"name":"transtime","type":33}],"name":"moveToHueAndSaturation"}},"address":14003,"endpoint":1,"linkquality":69,"groupID":45825,"wasBroadcast":false,"destinationEndpoint":1}' +3s`

`zigbee-herdsman:controller:log Skipping command 'moveToHueAndSaturation' because it is missing from the lookup +1ms`

if someone who know can add this command to Zigbee-hermann-converter, it will be very usefull for this remote and the other who seems to use the same command ([this](https://github.com/Koenkk/zigbee2mqtt/issues/2766) and [this](https://github.com/Koenkk/zigbee2mqtt/issues/4393))

thenicnet, 2024-06-09:
I bought one of these without checking the support page first. Has anyone gotten it working? If someone needs some data to help get this working please let me know and I'll see what I can do.

Koenkk, 2024-06-10:
@corve001 move to hue and saturation should be exposed, could you check with 1.38.0?

corve001, 2024-06-12:
@Koenkk ,
after update, move to hue and saturation is not exposed.
But I updated with a lot of errors (node is not up to date), so I'm not sure the install is correct even tough zigbee2mqtt seem's to be fully functionnal. I think I need to update raspbian to the latest version to update cleanly.

blubberfasel, 2024-08-13:
With some tweaks to the script I got it mostly working going by the MQTT messages:
```
const {deviceEndpoints, battery, identify, commandsOnOff, commandsLevelCtrl, commandsColorCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['501.40'],
    model: '501.40',
    vendor: 'Paulmann Licht GmbH',
    description: 'RGB Remote Control Gent 2',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4}})], 
    fromZigbee: [fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.command_step, fz.command_step_color_temperature, fz.command_recall, fz.command_move_to_color_temp, fz.battery, fz.command_move_hue, fz.command_move_to_hue_and_saturation, fz.command_move_to_hue, fz.command_move_to_saturation],
    toZigbee: [],
    exposes: [e.battery(), e.action(['on', 'off', 'brightness_stop', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down', 'color_temperature_step_up', 'color_temperature_step_down', 'recall_*', 'hue_move', 'hue_stop', 'move_to_hue', 'move_to_hue_and_saturation'])],
    meta: {"multiEndpoint":true, "publishDuplicateTransaction":true},
};

module.exports = definition;
```

A few button presses don't work as far as MQTT publishing is concerned due to unmapped raw values (they are functional when binding to a lamp):
- The "Start/Pause RGB color loop" event: `Received Zigbee message from '...', type 'raw', cluster 'lightingColorCtrl', data '{"data":[1,39,71,0,0],"type":"Buffer"}' from endpoint 1 with groupID 2`
- The stop event for "color_temperature_move_down": `Received Zigbee message from '...', type 'raw', cluster 'lightingColorCtrl', data '{"data":[1,79,71,0,0],"type":"Buffer"}' from endpoint 3 with groupID 24835`
- The stop event for "color_temperature_move_up": `Received Zigbee message from '...', type 'raw', cluster 'lightingColorCtrl', data '{"data":[1,4,71,0,0],"type":"Buffer"}' from endpoint 3 with groupID 24835`

Hykario, 2024-09-09:
added support on
#https://github.com/Koenkk/zigbee-herdsman-converters/pull/7947
Please Close