Koenkk/zigbee2mqtt #21216: Zigbee 3.0 DALI CONV LI dimmer for DALI-based luminaires (with two devices) does not recognize channels correctly

Hubi-Swiss, 2024-02-04:
### What happened?

When pairing above device, it is correctly recognized as 4062172044776_3 with 2 devices after successfully interviewing, but the 2 devices can only be switched together on endpoint 10.

### What did you expect to happen?

After interviewing the endpoints 10, 11 and 242 are created, although from reading through the issue list and documentation I would expect l1 and l2 to be created.

### How to reproduce it (minimal and precise)

Pairing Dali device from scratch after reset of device. This is on fresh install of Homeasistent add on edge version.

### Zigbee2MQTT version

1.35.2-dev commit: 4b17217

### Adapter firmware version

20220219

### Adapter

	SLZB-06

### Setup

Add on Homeassistant OS Edge Version on Synology DS1621+

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14157363/log.txt)

Hubi-Swiss, 2024-02-04:
In case it helps here is the database entry for the device:
{"id":17,"type":"Router","ieeeAddr":"0x000d6f00170e21fc","nwkAddr":54290,"manufId":4364,"manufName":"Osram Sylvania\u0000","powerSource":"Mains (single phase)","modelId":"Zigbee 3.0 DALI CONV LI\u0000","epList":[10,11,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":257,"inClusterList":[0,3,4,5,6,8,769,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Zigbee 3.0 DALI CONV LI\u0000","manufacturerName":"Osram Sylvania\u0000","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":0,"hwVersion":5,"dateCode":"","swBuildId":"1.0.0.8-6.5.5.b"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030eb5f76","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0030eb5f76","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,769,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030eb5f76","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0030eb5f76","endpointID":1}],"configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":0,"hwVersion":5,"dateCode":"","swBuildId":"1.0.0.8-6.5.5.b","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1707061928420}

closed 2024-09-03 by github-actions[bot]

michnovka, 2025-03-26:
I have the same issue, have you figured out a solution? I have 2 devices connected on DALI, it works flawlessly on ZHA but z2m only allows me to control both devices at once (basically the l2 endpoint does nothing, and l1 controls both lights)