Koenkk/zigbee2mqtt #21223: Aqara Motion Sensor RTCGQ11LM will not pair. Works on Deconz just fine.

Ltek, 2024-02-04:
### What happened?

I have tried at least a dozen times to pair these. I have many of them paired to Deconz for about 2 years. No matter how many times I reset and try to pair to Z2M it never works... I dont even see any debug logs showing any comms between them. When I repair to Deconz it literally takes 2 seconds from button press to being paired.

I have many Tuya and Bosch devices paried to Z2M, quickly and without issue... there is no doubt a bug in Z2M causing the https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html not to pair

### What did you expect to happen?

should pair without issue, like it does in Deconz

### How to reproduce it (minimal and precise)

reset and then try to pair

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

 EZSP v8 6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Windows PC

### Debug log

literally doesnt show anything so not adding

szymucha94, 2024-02-07:
Had same exact problem with SONOFF Zigbee 3.0 USB Dongle Plus-E and RTCGQ11LM about half a year ago. It did pair after multiple restarts but just decided to give up and sold the adapter.
Hope this gets fixed tho.

Ltek, 2024-02-07:
> Had same exact problem with SONOFF Zigbee 3.0 USB Dongle Plus-E and RTCGQ11LM about half a year ago. It did pair after multiple restarts but just decided to give up and sold the adapter. Hope this gets fixed tho.

I'd bet the sonoff hardware is not the problem.

szymucha94, 2024-02-08:
> > Had same exact problem with SONOFF Zigbee 3.0 USB Dongle Plus-E and RTCGQ11LM about half a year ago. It did pair after multiple restarts but just decided to give up and sold the adapter. Hope this gets fixed tho.
> 
> I'd bet its a Z2M bug -- or gap in functionality; not a hardware problem - sonoff and Aqara work fine on lots of stuff.

I agree, although I was replacing existing EZSPv8 gateway with zigbee stick so I couldn't wait for the fix to be released. Keep in mind this platform is in experimental state in z2m so it may take long time for it to be fixed.

Ltek, 2024-02-08:
> I agree, although I was replacing existing EZSPv8 gateway with zigbee stick so I couldn't wait for the fix to be released. Keep in mind this platform is in experimental state in z2m so it may take [...]

not sure what you mean by experimental ... I've been in hardware and software for 35 yrs, experience has proven that 99% of all software never actually gets beyond beta quality *kinda kidding but its true

Ltek, 2024-02-21:
@Koenkk would appreciate if you can look at this issue. Those devices work 100% fine with Conbee/Deconz I cant imagine Z2M could not accommodate?  From the user testing (mine and others) it looks like maybe a comms timeout issue.

Koenkk, 2024-02-22:
@Ltek this is likely an issue with the EZSP firmware or integration, I would recommend trying out the [new driver](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)

Ltek, 2024-08-25:
not stale... and @Koenkk, I updated my FW and Z2M several months ago and it never fixed the issue. I had to go back to Conbee/Deconz for several of my devices since Z2M has problems both doing the initial pairing and often drops the device only a few mins-to-hours after it pairs. Yes, Z2M stats show 'last seen' gets updated when none of them work (trigger or update any properties). Pretty Sure 'last seen' is not actually reading the device since some of them dont even show a actual connected path on the map!

Ltek, 2024-12-30:
bummmmmp

closed 2025-05-10 by github-actions[bot]