Koenkk/zigbee2mqtt #21225: [New device support]: Tuya 6 gang smart switch

hanzala-123, 2024-02-04:
### Link

https://a.aliexpress.com/_EyQ1gsN

### Database entry

0xa4c13845c0b37a16 (0x97BA)

### Comments

I tried to follow the instructions on how to add support but failed. This is the exact same or very similar but it was not discovered as this device: 
https://www.zigbee2mqtt.io/devices/SWS6TZ-WHITE.html

### External defintion

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_cduqh1l0',
    description: 'Automatically generated definition',
    extend: [],
};

module.exports = definition;
```

mrmaximas, 2024-02-04:
> ### Link
> https://a.aliexpress.com/_EyQ1gsN
> 
> ### Database entry
> 0xa4c13845c0b37a16 (0x97BA)
> 

Please provide info from database.db file

hanzala-123, 2024-02-05:
How do I access it

mrmaximas, 2024-02-05:
> How do I access it

depending on OS but in the ~/zigbee2mqtt/data/database.db

Koenkk, 2024-02-05:
Does your device support power monitoring?

hanzala-123, 2024-02-05:
No

hanzala-123, 2024-02-05:
> > How do I access it
> 
> depending on OS but in the ~/zigbee2mqtt/data/database.db

My OS is home assistant on raspberry Pi

hanzala-123, 2024-02-22:
Hi, I’m struggling to find the database.db file on my home assistant OS

hanzala-123, 2024-02-22:
> > ### Link
> > https://a.aliexpress.com/_EyQ1gsN
> > ### Database entry
> > 0xa4c13845c0b37a16 (0x97BA)
> 
> Please provide info from database.db file

{"id":12,"type":"Unknown","ieeeAddr":"0x84fd27fffe4370e2","nwkAddr":4872,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1708501156121}

Koenkk, 2024-02-24:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

> {"id":12,"type":"Unknown","ieeeAddr":"0x84fd27fffe4370e2","nwkAddr":4872,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1708501156121}

Make sure to re-pair again after updating, interview failed in this case (just retry until it succeeds)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-24 by Koenkk

hanzala-123, 2024-02-24:
Thank you so much man ❤️Kind Regards, Hanzala Chaudry

hanzala-123, 2024-02-24:
Hi one of the switches works but the other ones say no converter available