Koenkk/zigbee2mqtt #21226: Z2M generate mass MQTT messages

mrmaximas, 2024-02-04:
### What happened?

@Koenkk A mass of (14-16) MQTT messages are generated when the device is switched on or off for this device
https://www.zigbee2mqtt.io/devices/SA-003-Zigbee.html#ewelink-sa-003-zigbee
This problem occurred recently, maybe a couple of dev versions ago, I can't say for sure. 
Important (!) - setting Debounce to 1 or 5 doesn't help. Power cycle - doesn't help.
Same "spam" happens when Z2M periodically "pings" the device, but there is no spam when I "configure" the device.

### What did you expect to happen?

2-3 messages instead of 16

### How to reproduce it (minimal and precise)

Just switch on or off device

### Zigbee2MQTT version

1.35.2-dev commit: 3d378922

### Adapter firmware version

20221226

### Adapter

Sonoff ZB Dongle-P

### Setup

Mac OS Catalina latest

### Debug log

Debug 2024-02-05 00:50:48Received MQTT message on 'zigbee2mqtt/Kitchen Hood/set' with data '{"state":"OFF"}'
Debug 2024-02-05 00:50:48Publishing 'set' 'state' to 'Kitchen Hood'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Debug 2024-02-05 00:50:48Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":21,"state":"OFF"}'
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":25,"state":"OFF"}'
Info 2024-02-05 00:50:48MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:50:48+03:00","linkquality":25,"state":"OFF"}'

Debug 2024-02-05 00:51:04Received MQTT message on 'zigbee2mqtt/Kitchen Hood/set' with data '{"state":"ON"}'
Debug 2024-02-05 00:51:04Publishing 'set' 'state' to 'Kitchen Hood'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
[... 4 lines cut ...]
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
[... cut ...]
Debug 2024-02-05 00:51:04Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'
Info 2024-02-05 00:51:04MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"last_seen":"2024-02-05T00:51:04+03:00","linkquality":40,"state":"ON"}'

uncle-fed, 2024-02-05:
I can confirm very similar behaviour for at least another device: [Gledopto GL-SD-001](https://www.zigbee2mqtt.io/devices/GL-SD-001.html#gledopto-gl-sd-001) dimmer. When sending ON or OFF command, it triggers a dozen or so of messages with the `last_seen`  and `state`  attributes.

Regarding the `debounce`  option, I can also confirm that it does not help fully. Only in my case, if I set debounce to 2 seconds, I start getting not multitudes of messages, but 2. One comes immediately when the state changes (so it appears that debounce is ignored), and then, after the 2 seconds of debounce interval comes another message.

mrmaximas, 2024-02-05:
@uncle-fed Did this problem occur recently or was it there before?

uncle-fed, 2024-02-05:
@mrmaximas I won't be able to tell for sure regarding "before", as I have this particular dimmer for about 3 months only and I did not check for this immediately when I got it, because it seemed to worked fine on the outside.

The reason why I started checking the logs and noticed this now is because the dimmer stopped working at all properly after the recent update of z2m to version 1.35.2. The commands sent to the dimmer (especially I suspect "brightness change") just time out for no reason. Going back to z2m version 1.35.1 immediately resolves these problems and everything works as before. I will be searching the issue tracker and if I don't find anything similar reported, I will most likely open a new issue for it. 

So this is the reason why I was looking at the logs and then, of course, I noticed that this device floods with messages (despite functioning OK). I checked this with z2m versions 1.35.1, 1.34 and 1.33  and the behaviour has not changed - still flooding. So I guess you could say that at least for the past 2-3 months the issue was already there.

And, yes, I also have Gledopto GL-C-008 and GL-C-006 that do not have this issue. Those Gledoptos, including yours 009p are LED strip dimmers. The one I have a problem with is not an LED one (see the [link](https://www.zigbee2mqtt.io/devices/GL-SD-001.html#gledopto-gl-sd-001)).

mrmaximas, 2024-02-05:
> And, yes, I also have Gledopto GL-C-008 and GL-C-006 that do not have this issue. Those Gledoptos, including yours 009p are LED strip dimmers. The one I have a problem with is not an LED one (see the [link](https://www.zigbee2mqtt.io/devices/GL-SD-001.html#gledopto-gl-sd-001)).

I see your point, in general almost everything that grows around TUYA is glitchy crooked hardware, bribed by its price.

uncle-fed, 2024-02-05:
Are you sure that the [Gledopto Dimmer](https://www.zigbee2mqtt.io/devices/GL-SD-001.html#gledopto-gl-sd-001) in question is TuYa based? I cannot see a direct evidence of this, but maybe I am not looking deep enough. Also, it is really very certain that z2m v1.35.2 breaks the normal behaviour of the dimmer when at least 3 previous releases work fine. So I wouldn't be that hasty with general conclusions about TuYa (but on the other hand, no I don't want to defend it too much, because I still largely agree :-))

mrmaximas, 2024-02-05:
> 
> Oh yeah, and the `debounce` option not working right - that has probably nothing to do with TuYa, right? See this, also: [#17984 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/17984#issuecomment-1904668326)

you are right, it doesn't, but tuya being a member of the zigbee alliance ignores standards and gledopto is forced to follow tuya's internal requirements.

uncle-fed, 2024-02-05:
Neither, it looks like. I've opened the device and it appears to be EFR32MG21 chip. Very similar to the one here: https://github.com/zigpy/zigpy/discussions/586
The exact markings on the chip are: MG21 A02011 B01XWR 2141

mrmaximas, 2024-02-06:
@uncle-fed @Koenkk after a bit of digging, I turned off the "last seen" option and miraculously the spam disappeared.
what could it be?

mrmaximas, 2024-02-06:
but in debug messages this appear
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug 2024-02-06 11:31:42Received Zigbee message from 'Kitchen Hood', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0

uncle-fed, 2024-02-06:
@mrmaximas I think you're onto something with the `last_seen`.
Have a look here: https://github.com/Koenkk/zigbee2mqtt/issues/12766#issuecomment-1823451640
According to @Koenkk certain things are "normal" when `last_seen` is enabled.
Probably... maybe... but I wonder if there are certain corner cases like the ones we are seeing here, which are not fully covered with the existing code logic.

We have two issues mentioned here actually, the mass-flooding with messages (only when `last_seen`  is enabled?).
The fact that `debounce` behaves different to how it is documented or expected (again, when `last_seen` is enabled or generally?).

mrmaximas, 2024-02-06:
@uncle-fed In my case disabling `last_seen` helps only to get rid of multiple messages into info log, but in debug log everything is the same. it is not clear who generates so many messages, we can suspect the device itself, but there was no such thing before. I deleted and added the device back in, it didn't help. well, I'm gonna try to unpack my zigbee sniffer, if I can just find a few hours of time to do it.

Steini1984, 2024-02-06:
> @mrmaximas I won't be able to tell for sure regarding "before", as I have this particular dimmer for about 3 months only and I did not check for this immediately when I got it, because it seemed to [...]

I can confirm I had issues with my Gledopto GL-SD-001 dimmers on 1.35.2. Wrong status in home assistant, setting brightness was not stable and only sometimes working and when turning off via a bound switch the light would sometimes turn back on. Was fixed by reverting to 1.35.1

uncle-fed, 2024-02-06:
@Steini1984 thanks very much for confirmation. I have now opened a separate issue for this.
[ this.](https://github.com/Koenkk/zigbee2mqtt/issues/21259)
Please confirm it there, more people is always better. Thanks!

closed 2024-07-17 by mrmaximas