Koenkk/zigbee2mqtt #21230: Tuya ZY-M100 No Converter Available

Skyhawker231, 2024-02-04:
### What happened?

Hello!
since updating to 1.35.2 I've been unable to configure any settings exposed by this presence sensor. These include presence_timeout, radar_range, move_sensitivity all resulting in No converter available. How to resolve this issue?

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff

### Setup

Home Assistant OS

### Debug log

Debug 2024-02-05 10:34:41Received MQTT message on 'zigbee2mqtt/Presence Sensor/set' with data '{"move_sensitivity":1}'
Error 2024-02-05 10:34:41No converter available for 'move_sensitivity' (1)

Debug 2024-02-05 10:36:42Received MQTT message on 'zigbee2mqtt/Presence Sensor/set' with data '{"radar_range":1.5}'
Error 2024-02-05 10:36:42No converter available for 'radar_range' (1.5)

Debug 2024-02-05 10:36:39Received MQTT message on 'zigbee2mqtt/Presence Sensor/set' with data '{"presence_timeout":51}'
Error 2024-02-05 10:36:39No converter available for 'presence_timeout' (51)

DuncanRae, 2024-02-05:
Also having this problem. 

I commented on the PR. 

https://github.com/Koenkk/zigbee-herdsman-converters/pull/6959

gustjoha, 2024-02-05:
Thanks. I also have this issue, my sensor is practically unusable right now as it is too sensitive on default settings.

Koenkk, 2024-02-05:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-05 by Koenkk

Skyhawker231, 2024-02-07:
@Koenkk Looks to be resolved however has a new error introduced with this update regarding illumination #21275

Skyhawker231, 2024-02-07:
Can also confirm that there are devices which are not setting the presence status since this update both being Tuya ZY-M100. One is working fine but the other only works in attempting to change illumination status (see issue above) but doesn't get the presence status at all.