Koenkk/zigbee2mqtt #21232: MCCGQ14LM  Firmware -1

Sonic62, 2024-02-05:
### What happened?

Firmawre invisible

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

20221226

### Adapter

Sonoff P

### Setup

ADDONS HOME ASSISTANT

### Debug log

[image]
[image]

Sonic62, 2024-02-05:
[image]
[image]
[image]

Koenkk, 2024-02-05:
we don't have an OTA for this device ATM, disabled it

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-05 by Koenkk