Koenkk/zigbee2mqtt #21233: VOCKQJK11LM Failed to configure

Sonic62, 2024-02-05:
### What happened?

Failed to configure

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

20221226

### Adapter

Sonoff P

### Setup

ADDONS HOME ASSISTANT

### Debug log

`Error 2024-02-05 02:27:59Failed to configure 'Thermomètre Salon', attempt 3 (Error: Bind 0x54ef441000357ab6/1 genPowerCfg from '0x00124b002a2ec244/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:692:28 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2924:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

ryanm352, 2024-02-06:
I had to downgrade. In my case I went to 1.34 because I wasn't sure what version I upgraded from.  This happened to me for multiple zigbee devices. They wouldn't even pair when removed. Once I downgraded, everything worked again.

cipry85, 2024-02-09:
I'm having the same problem. I have now tried downloading the version 1.34

sjorge, 2024-02-11:
Just tried with my unused one on dev (well from yesterday somewhere when I last updated)

```
Zigbee2MQTT:debug 2024-02-11 13:24:22: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"j062b-4","value":true}'
Zigbee2MQTT:info  2024-02-11 13:24:22: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2024-02-11 13:24:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"j062b-4"}'
Zigbee2MQTT:info  2024-02-11 13:24:37: Device '0x54ef441000684c5c' joined
Zigbee2MQTT:info  2024-02-11 13:24:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000684c5c","ieee_address":"0x54ef441000684c5c"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-02-11 13:24:37: Starting interview of '0x54ef441000684c5c'
Zigbee2MQTT:info  2024-02-11 13:24:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000684c5c","ieee_address":"0x54ef441000684c5c","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:debug 2024-02-11 13:24:37: Device '0x54ef441000684c5c' announced itself
Zigbee2MQTT:info  2024-02-11 13:24:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000684c5c","ieee_address":"0x54ef441000684c5c"},"type":"device_announce"}'
Zigbee2MQTT:debug 2024-02-11 13:24:43: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'genBasic', data '{"appVersion":29,"modelId":"lumi.airmonitor.acn01"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:43: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:43: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2042}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:43: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:43: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5352}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:43: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:46: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":27}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:46: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:46: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,203,10,3,40,20,4,33,168,1,5,33,93,0,6,36,4,0,0,0,0,8,33,29,1,10,33,0,0,12,32,1,100,41,251,7,101,33,254,20,102,33,9,0,103,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:46: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:49: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,162,10,3,40,20,4,33,168,1,5,33,93,0,6,36,3,0,0,0,0,8,33,29,1,10,33,0,0,12,32,1,100,41,251,7,101,33,255,20,102,33,9,0,103,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:49: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:51: Received Zigbee message from '0x54ef441000684c5c', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:51: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:52: Received Zigbee message from '0x54ef441000684c5c', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:52: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:53: Received Zigbee message from '0x54ef441000684c5c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:53: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:54: Received Zigbee message from '0x54ef441000684c5c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:54: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:55: Received Zigbee message from '0x54ef441000684c5c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:55: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:56: Received Zigbee message from '0x54ef441000684c5c', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20220222"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:56: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:24:57: Received Zigbee message from '0x54ef441000684c5c', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2020\u0000www."}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-02-11 13:24:57: Skipping message, still interviewing
Zigbee2MQTT:debug 2024-02-11 13:25:02: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'aqaraOpple', data '{"airQuality":1}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:debug 2024-02-11 13:25:40: Received Zigbee message from '0x54ef441000684c5c', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":8}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-02-11 13:25:40: MQTT publish: topic 'zigbee2mqtt/0x54ef441000684c5c', payload '{"air_quality":"excellent","battery":0,"device_temperature":20,"display_unit":"ppb_celsius","humidity":53.28,"linkquality":185,"power_outage_count":92,"temperature":20.44,"voc":8,"voltage":2722}'
```

Still seems to work fine, I'm using a zzh adaptor with 20220928 fw release.

You do have to occasionally use the button to keep it awake until it finishes everything.

sjorge, 2024-02-11:
Actually I noticed you're not pairing but z2m is probably trying to trigger a reconfigure at startup.

zh also got this change https://github.com/Koenkk/zigbee-herdsman/commit/2d3a5d1d25f45360e253946d26f4e8b95ea75e26 this release, I only have this specific device but it is missing a genPollCtrl cluster (shame) which means it z2m will now try to immediately send the messages. Before it would only do it when we got a message from the device.

The device is probably not 'awake' which explains the lack of response to the bind on genPowerCfg. I added a quirk that restores the old behavior (well it waits up to 1h for a message from the device) https://github.com/Koenkk/zigbee-herdsman-converters/pull/6840 (which got refactored later on but the result is the same).

Of course this required configure to have run at least once for the extra database entry to be present. You could try keeping the device awake by pressing the top button every 2-3 seconds and hitting the reconfigure button in the UI.

satter, 2024-03-09:
My Aqara VOCKQJK11LM sensor is being reconfigured each time after Z2M upgrade generating bunch of errors in logs:
```
Mar 09 08:51:09 homie1 npm[1921529]: Zigbee2MQTT:error 2024-03-09 08:51:09: Publish 'get' 'humidity' to 'kitchen/aqara_tvoc1' failed: 'Error: Read 0x54ef441000106dca/1 msRelativeHumidity(["measuredValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 12 - 1029 - 1 after 10000ms)'
Mar 09 08:51:32 homie1 npm[1921529]: Zigbee2MQTT:error 2024-03-09 08:51:32: Publish 'get' 'temperature' to 'kitchen/aqara_tvoc1' failed: 'Error: Read 0x54ef441000106dca/1 msTemperatureMeasurement(["measuredValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 26 - 1026 - 1 after 10000ms)'
Mar 09 08:51:55 homie1 npm[1921529]: Zigbee2MQTT:error 2024-03-09 08:51:55: Publish 'get' 'voc' to 'kitchen/aqara_tvoc1' failed: 'Error: Read 0x54ef441000106dca/1 genAnalogInput(["presentValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 27 - 12 - 1 after 10000ms)'
Mar 09 09:26:47 homie1 npm[1921529]: Zigbee2MQTT:error 2024-03-09 09:26:47: Failed to configure 'kitchen/aqara_tvoc1', attempt 1 (Error: Bind 0x54ef441000106dca/1 msTemperatureMeasurement from '0x00124b00258d3c29/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Mar 09 09:30:35 homie1 npm[1921529]: Zigbee2MQTT:error 2024-03-09 09:30:35: Failed to configure 'kitchen/aqara_tvoc1', attempt 2 (Error: Bind 0x54ef441000106dca/1 msTemperatureMeasurement from '0x00124b00258d3c29/1' failed (AREQ - ZDO - bindRsp after 10000ms)
```
Manual reconfiguration works after I wake up the sensor with button. After that no more errors until next Z2M upgrade

sjorge, 2024-03-09:
Wake up the device so it can finish a reconfigure. It never completes due to a timeout.

So z2m will try again later.

satter, 2024-03-09:
Yes, wake up and reconfigure works fine  
Is it OK for this device to be reconfigured after each Z2M upgrade?

satter, 2024-03-09:
I've also noticed missing sensors data from dashboard tile:
[image]
I'm not sure when this changed, but some time before there was other sensors like humidity and VOC

The 'exposes' tab seems to be fine:
[image]

sjorge, 2024-03-09:
The last 3 updates there were some changes to the device, so if there are no updates it should no reconfigure. But there were a few recently so that is probably why it tries to on the last few release. Some of changes:
- Add air_quality support
- Move air_quality, voc, temperature, humidity over to modernExtend (@Koenkk could this have been the trigger for them to disappear from the dashboard? although I don't see how as the names didn't change, but I don't know much about the frontend/dashboard)
- Added a quirk to inject some missing clusters (that were present but not announced, we can now bind this sensor to other things like a TRV)
- Added a quirk to work around the device not having genPollCtrl that was causing issues with the sendPolicy rework in zh
- rename Xiaomi/Aqara/Maji to Lumi on a code level (as the are all made by Lumi and sold under different brands)

All of those would have triggered a reconfigure. The latest stable added some changes where it doesn't try to reconfigure battery devices on startup. So if there are further changes that trigger a reconfigure it should not immediately result in errors at startup. That was not a Aqara specifc problem, it was causing issues for all devices that are battery powered and lacking the genPollCtrl cluster.

satter, 2024-03-09:
@sjorge thanks for the detailed explanation!
I've seen my sensors, including this one from Aqara, reconfiguring after an upgrade before, but there were no errors during this process.

I've also noticed weird battery readings on version 1.35.3, which seem to have stopped on version 1.36.0 today.  
I will replace the batteries anyway and check for errors during the next upgrade

satter, 2024-04-29:
I upgraded to version 1.36.1 today, and the dashboard still lacks every sensor except for the device temperature
There were several one-time errors shortly after startup:
```
debug 2024-04-29 13:38:44: Received MQTT message on 'zigbee2mqtt/0x54ef441000106dca/get' with data '{"humidity":0,"temperature":0,"voc":0}'
debug 2024-04-29 13:38:44: Publishing get 'get' 'humidity' to 'kitchen/aqara_tvoc1'
error 2024-04-29 13:39:07: Publish 'get' 'humidity' to 'kitchen/aqara_tvoc1' failed: 'Error: ZCL command 0x54ef441000106dca/1 msRelativeHumidity.read(["measuredValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 30 - 1029 - 1 after 10000ms)'
debug 2024-04-29 13:39:07: Error: ZCL command 0x54ef441000106dca/1 msRelativeHumidity.read(["measuredValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 30 - 1029 - 1 after 10000ms)
debug 2024-04-29 13:39:07: Publishing get 'get' 'temperature' to 'kitchen/aqara_tvoc1'
error 2024-04-29 13:39:30: Publish 'get' 'temperature' to 'kitchen/aqara_tvoc1' failed: 'Error: ZCL command 0x54ef441000106dca/1 msTemperatureMeasurement.read(["measuredValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 44 - 1026 - 1 after 10000ms)'
debug 2024-04-29 13:39:30: Error: ZCL command 0x54ef441000106dca/1 msTemperatureMeasurement.read(["measuredValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 44 - 1026 - 1 after 10000ms)
debug 2024-04-29 13:39:30: Publishing get 'get' 'voc' to 'kitchen/aqara_tvoc1'
error 2024-04-29 13:39:53: Publish 'get' 'voc' to 'kitchen/aqara_tvoc1' failed: 'Error: ZCL command 0x54ef441000106dca/1 genAnalogInput.read(["presentValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 45 - 12 - 1 after 10000ms)'
debug 2024-04-29 13:39:53: Error: ZCL command 0x54ef441000106dca/1 genAnalogInput.read(["presentValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54748 - 1 - 45 - 12 - 1 after 10000ms)
```

Koenkk, 2024-04-30:
The next z2m release will fix the dashboard issues.

closed 2024-11-28 by github-actions[bot]

xelemorf, 2025-01-07:
I just bumped into the same issue but with zstack on SLZB-06 coordinator , FW 20240710 (with baudrate 115200 and on 20240315 (with baudrate 460800) while using Z2M v1.42.0-2 and also with the latest stable v2.0.0 (also tried Edge v2.0.0).
All sensor data is zero and does not update at all, while having 3 other TVOC sensors the same type and on same FW which are working fine, the only difference is that I had to replace the old 4th with this new one, so it seems this type of device cannot be added as a new one, or I don't know what to think tbh.

```
[2025-01-07 17:57:10] error: 	z2m: Failed to configure 'bedroom-tvoc', attempt 1 (Error: Bind 0x54ef4410007953de/1 genPowerCfg from '0x00124b002b481b65/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:403:47
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZdo (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:311:16)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:30)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:3483:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13))
```