Koenkk/zigbee2mqtt #21234: Updated Moes ZS-SR Star Ring 1,2 & 3 Gang Non Dim Switch with Router Function

demcas, 2024-02-05:
### What happened?

https://moeshouse.com/products/eu-star-ring-smart-light-switch-1?variant=45282168340795

**Currently these three non dim light switches are END DEVICES**
Moes ZS-SR-EU1 - TS0011
Moes ZS-SR-EU2 - TS0012
Moes ZS-SR-EU3 - TS0013

**Moes released for testing an updated New Version of the same light switches with ROUTER functionality**  
Moes ZS-SR-EU1 - TS0001
Moes ZS-SR-EU2 - TS0002
Moes ZS-SR-EU3 - TS0003
- The new switches have been detected and functioning correctly, **but the the Backlight Mode & Power On Behaviour is missing Under Exposes** 

### What did you expect to happen?

Updated New Version of the same light switches with ROUTER functionality
Can we active the missing Exposes Backlight Mode & Power On Behaviour?

Moes ZS-SR-EU1 - TS0001
Moes ZS-SR-EU2 - TS0002
Moes ZS-SR-EU3 - TS0003
[image]
[image]

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

Latests version

### Adapter

Home Assistant SkyConnect

### Setup

Dell PC Client

### Debug log

**1Gang Non Dim End Device**
{"id":16,"type":"EndDevice","ieeeAddr":"XXXXXXXX","nwkAddr":37083,"manufId":4417,"manufName":"_TZ3000_hhiodade","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0011","manufacturerName":"_TZ3000_hhiodade","powerSource":3,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-999208499},"lastSeen":1707097931305}

**1 Gang Non Dim Router Device**
{"id":12,"type":"Router","ieeeAddr":"XXXXXXXX","nwkAddr":50472,"manufId":4417,"manufName":"_TZ3000_j8dbnrgx","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":0}},"genBasic":{"attributes":{"65503":"�sF-\u0017�sF-\u0013�sF-\u00120\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_j8dbnrgx","powerSource":1,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1707092892857}

**2Gang Non Dim End Device**
{"id":13,"type":"EndDevice","ieeeAddr":"XXXXXXXX","nwkAddr":34969,"manufId":4417,"manufName":"_TZ3000_18ejxno0","powerSource":"Mains (single phase)","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":0}},"genBasic":{"attributes":{"65534":0,"modelId":"TS0012","manufacturerName":"_TZ3000_18ejxno0","powerSource":3,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1794938824},"lastSeen":1707097811724}

**2 Gang Non Dim Router Device**
{"id":8,"type":"Router","ieeeAddr":"XXXXXXXX","nwkAddr":340,"manufId":4417,"manufName":"_TZ3000_rzw3l3ro","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0002","manufacturerName":"_TZ3000_rzw3l3ro","powerSource":1,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1030697865},"lastSeen":1707098823416}

**3Gang Non Dim End Device**
{"id":14,"type":"EndDevice","ieeeAddr":"XXXXXXXX","nwkAddr":17409,"manufId":4417,"manufName":"_TZ3000_qewo8dlz","powerSource":"Mains (single phase)","modelId":"TS0013","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011","65506":55,"65508":1,"65534":0,"modelId":"TS0013","manufacturerName":"_TZ3000_qewo8dlz","powerSource":3,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1331047771},"lastSeen":1707092156389}

**3 Gang Non Dim Router Device**
{"id":15,"type":"Router","ieeeAddr":"XXXXXXXX","nwkAddr":31367,"manufId":4417,"manufName":"_TZ3000_uilloarc","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"u�F-\u0017v�F-\u0013w�F-\u0012\u0012\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_uilloarc","powerSource":1,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":0}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAA"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"cl [... cut ...] viceIeeeAddress":"0xe0798dfffe1677c2","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1707097377661}

demcas, 2024-02-11:
@kirovilya hi, previously you helped activate the backlight mode on the dimming version of this same light switch.  Can you assist with this issue and activate the backlight mode for these new Moes switches with router function?

demcas, 2024-03-30:
Still having this issue

Hewelius, 2024-04-24:
Hello demcas,
are you pretty sure, that the version you have linked can act like a router? I would like to buy a few pieces and have asked MOUS for that, and they answer was: no router functionality.

demcas, 2024-04-24:
> Hello demcas,
> 
> are you pretty sure, that the version you have linked can act like a router? I would like to buy a few pieces and have asked MOUS for that, and they answer was: no router functionality.

Moes have built a non dim router version of this switch exclusively for me.  I had it certified for the Australian market to meet safety standards and will be distributing the switches locally here in AU.

Hewelius, 2024-04-24:
> Moes have built a non dim router version of this switch exclusively for me. I had it certified for the Australian market to meet safety standards and will be distributing the switches locally here [...]

Oh OK, Thank you for the info!

closed 2024-11-21 by github-actions[bot]