Koenkk/zigbee2mqtt #21237: OTA update of Danfoss thermostat 014G2461 fails after adapter disconnects

35gh, 2024-02-05:
### What happened?

I've tried to update my Danfoss thermostats to 01.28.0008 01.28 from 01.20.0008 01.20.

Update is quite long (roughly 80 min).

Update fails almost all the time, apparently due to Zigbee2mqtt disconnecting from my SLZB-06M coordinator. Failure can happen at any time (10%, 70%, 99%...).
- With v1.35.1, update was failing with "Update failed with reason: 'aborted by device'" error, but adapter has disconnected before according to the zigbee2mqtt uptime in the SLZB web interface
- With v1.25.2 I now get the error "Adapter disconnected, stopping"

I tried to factory reset my thermostats, restart the devices by removing the batteries as advised, restart zigbee2mqtt with the same result. I even changed the Zigbee channel and had to pair all my devices again.
I was however able to complete the OTA update for 3/8 of my devices with v1.35.1 after countless retries.

OTA works for other devices.

Connection to SLZB-06M is rock solid if I'm not using OTA. It stays connected for days. I've also pinged the adapter for hours from the docker host without any packed drop.

### What did you expect to happen?

OTA update work without failing

### How to reproduce it (minimal and precise)

Check for an OTA update
Restart the device by removing / inserting the batteries and connecting again to the Zigbee network
Wait for a cooldown period
Launch OTA update.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SLZB-06M

### Setup

Docker on PhotonOS

### Debug log

v1.35.1 logs

```
debug 2024-01-24 18:39:15OTA: Request offsets: fileOffset=347200 pageOffset=0 dataSize=63
debug 2024-01-24 18:39:15OTA: Payload offsets: start=347200 end=347250 dataSize=50
debug 2024-01-24 18:39:16OTA: Request offsets: fileOffset=347250 pageOffset=0 dataSize=63
debug 2024-01-24 18:39:16OTA: Payload offsets: start=347250 end=347300 dataSize=50
debug 2024-01-24 18:39:16OTA: Request offsets: fileOffset=347300 pageOffset=0 dataSize=63
debug 2024-01-24 18:39:16OTA: Payload offsets: start=347300 end=347350 dataSize=50
debug 2024-01-24 18:39:17OTA: Request offsets: fileOffset=347350 pageOffset=0 dataSize=63
debug 2024-01-24 18:39:17OTA: Payload offsets: start=347350 end=347400 dataSize=50
debug 2024-01-24 18:39:17OTA: Request offsets: fileOffset=347400 pageOffset=0 dataSize=63
debug 2024-01-24 18:39:17OTA: Payload offsets: start=347400 end=347450 dataSize=50
debug 2024-01-24 18:39:18OTA: Request offsets: fileOffset=347450 pageOffset=0 dataSize=63
debug 2024-01-24 18:39:18OTA: Payload offsets: start=347450 end=347500 dataSize=50
debug 2024-01-24 18:40:08OTA: Got upgrade end request for '0xccccccxxxxxxxxxx': {"status":149,"manufacturerCode":4678,"imageType":256,"fileVersion":284}
debug 2024-01-24 18:40:08OTA: Update failed with reason: 'aborted by device'
debug 2024-01-24 18:40:08Update of 'Thermostat' failed (Error: OTA: Update failed with reason: 'aborted by device')
info 2024-01-24 18:40:08MQTT publish: topic 'zigbee2mqtt/Thermostat', payload '{"adaptation_run_control":"none","adaptation_run_settings":true,"adaptation_run_status":"found","algorithm_scale_factor":1,"battery":30,"day_of_week":"thursday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","linkquality":248,"load_balancing_enable":true,"load_estimate":-7,"load_room_mean":-8000,"local_temperature":19.72,"mounted_mode_active":false,"mounted_mode_control":true,"occupied_heating_setpoint":16.5,"occupied_heating_setpoint_scheduled":16.5,"pi_heating_demand":1,"preheat_status":null,"programming_operation_mode":null,"radiator_covered":false,"regulation_setpoint_offset":null,"running_state":"idle","setpoint_change_source":"externally","system_mode":"heat","thermostat_vertical_orientation":null,"trigger_time":660,"update":{"installed_version":276,"latest_version":284,"state":"available"},"update_available":null,"viewing_direction":null,"window_open_external":false,"window_open_feature":true,"window_open_internal":"closed"}'
info 2024-01-24 18:40:08MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Thermostat"},"error":"Update of 'Thermostat' failed (OTA: Update failed with reason: 'aborted by device')","status":"error","transaction":"9ldkr-5"}'
error 2024-01-24 18:40:08Update of 'Thermostat' failed (OTA: Update failed with reason: 'aborted by device')
debug 2024-01-24 18:40:08Error: OTA: Update failed with reason: 'aborted by device' at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:569:24 at processTicksAndRejections (node:internal/process/task_queues:95:5)
```

v1.35.2:

```
debug 2024-02-05 11:47:22OTA: Request offsets: fileOffset=79800 pageOffset=0 dataSize=63
debug 2024-02-05 11:47:22OTA: Payload offsets: start=79800 end=79850 dataSize=50
debug 2024-02-05 11:47:23OTA: Request offsets: fileOffset=79850 pageOffset=0 dataSize=63
debug 2024-02-05 11:47:23OTA: Payload offsets: start=79850 end=79900 dataSize=50
debug 2024-02-05 11:47:23OTA: Request offsets: fileOffset=79900 pageOffset=0 dataSize=63
debug 2024-02-05 11:47:23OTA: Payload offsets: start=79900 end=79950 dataSize=50
debug 2024-02-05 11:47:24OTA: Request offsets: fileOffset=79950 pageOffset=0 dataSize=63
debug 2024-02-05 11:47:24OTA: Payload offsets: start=79950 end=80000 dataSize=50
error 2024-02-05 11:47:28Adapter disconnected, stopping
```

ytugarev, 2024-02-06:
+1
Just bought the device yesterday (new one with G) - eTRV0103
Firmware build date 20230914
Firmware version 00.20.0008 00.20
What is strange I have different imageType - "288"

`debug 2024-02-06 15:20:40: OTA: Got upgrade end request for '0x048727fffe1416da' (eTRV0103): {"status":149,"manufacturerCode":4678,"imageType":288,"fileVersion":28}
debug 2024-02-06 15:20:42: OTA: Update failed with reason: 'aborted by device'
`

hildebrau, 2024-02-08:
+1, sort of.. I'm having what appears to be the same issue but with an entirely different device.  I am trying to do an OTA upgrade on an Inovelli Blue 2-in-1 switch.  [I posted about it over in their community](https://community.inovelli.com/t/vzm31-sn-firmware-updates-aborting-and-hanging-via-z2m/16060).  See my details over there.

Perhaps this means it's not an Inovelli issue, but instead a zigbee2mqtt AND/OR SLZB-06M issue.

LMK how I can be of assistance with this issue.

hildebrau, 2024-02-08:
> Downgrade Z2M to 1.34.0-1 and OTA update completes fine.

I just read that part of your post and downgraded z2m and my Inovelli switch updated just perfectly.

[image]

Good to know it CAN work.

ytugarev, 2024-02-17:
After downgrading to the add-on to 1.35.1-1 update went successfully.
Or not exactly successfully cos I hit this unfamous bug when firmware ver showed as AAAABBBBCCCCDDDD
Followed steps as per guide below (one missing - press allow join after starting Z2M) and it solved the issue
[https://github.com/Koenkk/zigbee2mqtt/issues/11049#issuecomment-1072349455](url) 
[image]

35gh, 2024-03-02:
I've updated to 1.36.0 and was then able to OTA update 5 Danfoss thermostats without any issue. I guess the issue can be closed.

closed 2024-03-02 by 35gh