Koenkk/zigbee2mqtt #21239: OTA update TS011F_plug_1

hosmancevic, 2024-02-05:
### What happened?

When trying to do a OTA upgrade for my devices i am getting an error:

Z2MQTT version: 1.35.2-1
Device type: TS011F_plug_1
Error message: Failed to check if update available for '0x21d404602bd5b370' (Failed to find an endpoint which supports the OTA cluster)
Current device firmware: 0122052017

### Zigbee2MQTT version

1.35.2-1

### Adapter firmware version

20221226

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1

### Setup

Add-on on HASSIO, in Proxmox running on a NUC

oleg-d, 2024-02-29:
Same error with Sonoff usb dongle 

```
Friendly name
    zb_plug_2
Description

Last seen
    N/A
Availability
    [Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
    Router
Zigbee Model
    TS011F
Zigbee Manufacturer
    _TZ3000_okaz9tjs
Description
    Smart plug (with power monitoring by polling)
Support status

    Supported
IEEE Address
    0x70b3d52b6005043b
Network address
    0x140E
Firmware build date
    20210625
Firmware version
    500+TZSKT31BS102
Manufacturer
    [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
    [BSD29_1](https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1)
Power
Interview completed
    True
```

<code>
info 2024-02-29 14:41:32MQTT publish: topic 'zigbee2mqtt/zb_plug', payload '{"child_lock":"UNLOCK","current":0,"energy":1.72,"indicator_mode":null,"linkquality":132,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":232}
</code><code>
info 2024-02-29 14:41:32MQTT publish: topic 'zigbee2mqtt/zb_plug', payload '{"child_lock":"UNLOCK","current":0,"energy":1.72,"indicator_mode":null,"linkquality":132,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":232}'
</code><code>
info 2024-02-29 14:41:34Checking if update available for 'zb_plug_2'
</code><code>
info 2024-02-29 14:41:34MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"zb_plug_2"},"error":"Failed to check if update available for 'zb_plug_2' (Failed to find an endpoint which supports the OTA cluster)","status":"error","transaction":"cf30l-2"}'
</code><code>
error 2024-02-29 14:41:34Failed to check if update available for 'zb_plug_2' (Failed to find an endpoint which supports the OTA cluster)
</code>

yannpub, 2024-04-25:
I have similar issue regarding the OTA in Z2mqtt v1.36.1
My devices also report update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,
Reported firmware looks the same : 0122052017

The only difference is the error message which is a bit different:
Failed to check if update available for 'PlugZ_05' (OTA: Device didn't respond to OTA request)
Error: OTA: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:312:15) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:390:25) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:209:45) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

The weird thing is that I have a 20A, but like for you, it is _TZ3000_okaz9tjs and not the same manufacturer as listed in other issue reports:
Zigbee Manufacturer: _TZ3000_hdopuwv6 (16A-rated model)
Zigbee Manufacturer: _TZ3000_gvn91tmx (20A-rated model)

renslinden, 2024-06-14:
I do have the same issue. Unable to search for OTA updates.

Docker logging:
```
[2024-06-14 14:34:15] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya - MyPlug', payload '{"child_lock":"LOCK","countdown":0,"current":0,"energy":1.38,"indicator_mode":"off","last_seen":"2024-06-14T14:34:15+02:00","linkquality":176,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":234}'
[2024-06-14 14:35:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Tuya - MyPlug"},"error":"Failed to check if update available for 'Tuya - MyPlug' (OTA: Device didn't respond to OTA request)","status":"error","transaction":"k2x1e-1"}'
[2024-06-14 14:35:14] error: 	z2m: Failed to check if update available for 'Tuya - MyPlug' (OTA: Device didn't respond to OTA request)
```

Trying to reset the Energy (sum of consumed energy) isn't working either. This is the only logging i get, no failure or what so whatever. 

```
[2024-06-14 14:46:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya - MyPlug', payload '{"child_lock":"LOCK","countdown":0,"current":0,"energy":1.38,"indicator_mode":"off","last_seen":"2024-06-14T14:46:17+02:00","linkquality":176,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":236}'
[2024-06-14 14:46:17] info: 	zhc:tz: Invoked '0.0' with payload '{}'
```

It seems this particular version isn't fully supported by z2m.

-----------

Zigbee2MQTT version: 1.38.0

coordinator: SkyConnect
coordinator_version: 7.4.2 [GA]
coordinator_type: EmberZNet

zigbee_herdsman: 0.49.2
zigbee_herdsman_converters: 19.47.1

b2un0, 2024-07-01:
i paired the `_TZ3000_okaz9tjs` today with an original TuYa Gateway and there is no OTA Update available.

there is no OTA Event

[image]

on the tuya dev console i got this data points:

```json
{
  "1": "POWER1",
  "9": "Countdown Switch 1",
  "17": "Electricity",
  "18": "Current NOW",
  "19": "Power NOW",
  "20": "Voltage NOW",
  "21": "Test Bit",
  "22": "Voltage COE",
  "23": "Electric COE",
  "24": "Power COE",
  "25": "Electricity COE",
  "26": "ERROR",
  "27": "Restart status",
  "28": "Indicator mode",
  "29": "Child Lock"
}
```

b2un0, 2024-07-09:
crazy, with `1.39.0` the device is now identified as
https://www.zigbee2mqtt.io/devices/LSPA9.html

helljol, 2024-08-18:
Hello i also purchase  2 _TZ3000_okaz9tjs (Elivco LSPA9) and OTA is not working.

> i paired the `_TZ3000_okaz9tjs` today with an original TuYa Gateway and there is no OTA Update available.
> 
> there is no OTA Event
> <img alt="image" width="915" src="https://private-user-images.githubusercontent.com/2155498/344596274-1120f57b-29b2-4740-9437-56bba5635328.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM5NjgzOTcsIm5iZiI6MTcyMzk2ODA5NywicGF0aCI6Ii8yMTU1NDk4LzM0NDU5NjI3NC0xMTIwZjU3Yi0yOWIyLTQ3NDAtOTQzNy01NmJiYTU2MzUzMjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MThUMDgwMTM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmMxZTNlMWQwODcxYzc4NmUwYzUwZmEwYWU2Yzk1YWMwNGZmZDA0NTA2NjUzMWM2ZjA0YjY1ZDdlZTZhMGI3ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.SSCepYdP8M6URQnQIsFC-Z7SvsG5RkaTclQ9f_og_Mo">
> 
> on the tuya dev console i got this data points:
> 
> ```json
> {
>   "1": "POWER1",
>   "9": "Countdown Switch 1",
>   "17": "Electricity",
>   "18": "Current NOW",
>   "19": "Power NOW",
>   "20": "Voltage NOW",
>   "21": "Test Bit",
>   "22": "Voltage COE",
>   "23": "Electric COE",
>   "24": "Power COE",
>   "25": "Electricity COE",
>   "26": "ERROR",
>   "27": "Restart status",
>   "28": "Indicator mode",
>   "29": "Child Lock"
> }
> ```

Thanks for checking there was no OTA available, i also purchase  2 _TZ3000_okaz9tjs (Elivco LSPA9) and OTA is not working. I saw various issues reported and i was not sure if OTA was avaialble on a tuya gateway.

Firmware reported for the plugs is 0122052017 and they are powering off randomly like others TS011F_plug_1 before being fixed by a new firmware. This makes those plugs mostly useless so i would not recommend purchasing this brand/model.

closed 2025-01-08 by github-actions[bot]

Ivano62, 2025-04-14:
I've the same issue with TS011F (Manufacturer: Elivco - Model: LSPA9)
Thanks for any helps