Koenkk/zigbee2mqtt #21240: IKEA Tradfri LED2104R3 still showing as Not supported in homeassistant addon

Toomblr, 2024-02-05:
### What happened?

I have two IKEA Tradfri LED2104R3 lights, which were working until I upgraded to `1.35.2-dev`. This should be fixed as per #20551 . However they are still not being recognized within Z2M.
[image]

### What did you expect to happen?

They should be supported: [click](https://www.zigbee2mqtt.io/devices/LED2104R3.html)

### Zigbee2MQTT version

1.35.2-dev commit: dfda631

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Setup

Hass addon edge

Koenkk, 2024-02-05:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-05 by Koenkk

Toomblr, 2024-02-06:
@Koenkk,
This change did not fix the issue. Am I doing something wrong? I am currently running: `1.35.2-dev commit: 92e0d78`

Koenkk, 2024-02-06:
can you provide the data/database.db entry of your device?

Toomblr, 2024-02-06:
> can you provide the data/database.db entry of your device?
`{"id":29,"type":"Router","ieeeAddr":"0xf082c0fffe188a9e","nwkAddr":40413,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"\u001aTRADFRI bulb GU10 WW 345lm8","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64636],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"\u001aTRADFRI bulb GU10 WW 345lm8","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":8,"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230918","swBuildId":"1.0.36"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff06988e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230918","swBuildId":"1.0.36","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1707203798112}`

Koenkk, 2024-02-06:
Should be good now