Koenkk/zigbee2mqtt #21243: 2 important missing attributes for Sinope TH1300ZB thermostat - How to read data or add to the code ?

Montreal666, 2024-02-05:
### What happened?

I am succesfully using Z2M to interact with the Sinope TH1300ZB thermostat.

unfortunately it is missing 2 important read attributes from the "manuSpecificSinope" cluster:

FloorTemperature  (Cluster 0xff01 - attribute 0x0107)
RoomTemperature  (Cluster 0xff01 - attribute 0x010D)

### What did you expect to happen?

Currently Z2M only exposes "LocalTemperature" (either floor or room , based on active mode)
But I need to display separate room/floor temperatures in Home Assistant, no matter which mode is selected. 
I know these attributes are available as I just migrated from ZHA and they were available/identified/working (below):

[image]

Is it currently possible to read custom cluster/attribute data in HA or does it have to go through a change / pull request to the code ?

If that's the case I don't mind doing the leg work but  would need some guidance on the next steps.

Any help would be greatly appreciated.

Thank you!

### How to reproduce it (minimal and precise)

Checked with Dev console, cluster is there (manuSpecificSinope  0xff01)
but attributes are not available

FloorTemperature  (Cluster 0xff01 - attribute 0x0107)
RoomTemperature  (Cluster 0xff01 - attribute 0x010D)

or 

reportFloorTemperature
reportRoomTemperature

[image]

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

Sonoff

### Setup

HA Add-on on intel/win11

Montreal666, 2024-02-06:
Thanks for the prompt response, I am running Zigbee2mqtt as an add-on to Home assistant so I don't have direct write access to the files. What would be the best workaround in this context? Can I run a custom/dev branch version with the add-on? 
Or should I seek assistance from https://github.com/zigbee2mqtt/hassio-zigbee2mqtt ?
Thanks for your help!

Koenkk, 2024-02-06:
@Montreal666 see [here](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt?tab=readme-ov-file#testing-changes-locally) how to edit files in the addon container.

Montreal666, 2024-02-06:
Thank you, had found the link but thought it would be possible to just use the standard HA SSH terminal but it seems to require a USB drive (I guess a usb "key" would work?) and SSH keys. the procedure seems straight forward but I am running HAOS on a Virtualbox over Windows11 so not even sure if this will work. Will try later, fingers crossed. Thx

Koenkk, 2024-02-07:
I think https://community.home-assistant.io/t/hassos-docker-command-not-found/81477 can help

Montreal666, 2024-02-07:
ok I *think* I managed to login to root and gain access to the container but can't find cluster.ts;
does this look like the right place to find it?

[image]

Thank you!

Montreal666, 2024-02-08:
OK here's where I'm at:

I've edited cluster.js

[image]

restarted the docker with :

docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)

tried refreshing the browser / reboot but the changes do appear or stick; IE: if I go back to the container, they are gone.

Montreal666, 2024-02-08:
More discussion here:

https://discord.com/channels/556563650429583360/558343554246901770/1205145196329697361

Unless you have troubleshooting ideas (?) I will see if I can test on my host and report back.

Thank you!

Koenkk, 2024-02-08:
Frontend will read them from zigbee-herdsman release, so they indeed don't show up. Since you know the attribute and data type, I suggest to make a PR to `cluster.ts`

Montreal666, 2024-02-08:
Thank you for your support, will do!

Montreal666, 2024-02-08:
Done. Newbie to Github so hopefully this was done correctly. Open to learn so feel free to provide guidance. Thanks again.

Montreal666, 2024-03-13:
I have installed the latest HA version:

[image]

and checked that the new attributes can be found in the container under cluster.js 

[image]

but the 2 new attributes do not appear in the drop down list under devconsole:

[image]

What am I missing ? thanks,

Montreal666, 2024-03-18:
Hi @Koenkk, would you mind having a look at this one when you have a chance? 
can't figure out what could be wrong. Thanks again.
Cheers,

Koenkk, 2024-03-19:
@nurikk could you bump the zigbee-herdsman version [here](https://github.com/nurikk/zigbee2mqtt-frontend/blob/2b52c5ab4fb9950dcb425414b0e69f586dfb7991/package.json#L115)? Maybe we can update the version through the pipeline like we do with [z2m](https://github.com/Koenkk/zigbee2mqtt/blob/master/.github/workflows/update_dep.yml)?

Montreal666, 2024-04-11:
Thank you @Koenkk , the 2 parameters are now available in the dev console and return correct values

[image]

however if I try to add the new parameters to the reporting:

[image]

I get the following error:

[image]

Maybe I'm not using the correct approach; any advice appreciated, thanks again for your support!

Koenkk, 2024-04-13:
It looks like this device doesn't support reporting for that attribute (`UNREPORTABLE_ATTRIBUTE` error)

Montreal666, 2024-04-13:
> It looks like this device doesn't support reporting for that attribute (`UNREPORTABLE_ATTRIBUTE` error)

Since I can read the data from the dev console (above), isn't their a workaround to pull the data in HA?

 Thank you

Koenkk, 2024-04-15:
You can poll it, an example how to do that from the converter can be found [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/6fef741ae76573288c0e4be6f6e75e4db0f02d31/src/devices/busch_jaeger.ts#L119) (`endpoint.read()`)

Montreal666, 2024-04-15:
> You can poll it, an example how to do that from the converter can be found [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/6fef741ae76573288c0e4be6f6e75e4db0f02d31/src/devices/busch_jaeger.ts#L119) (`endpoint.read()`)

Thank you for the pointer.
*If* I manage to adapt the example for my use case, where should the code be located and how do I trigger it from HA?
I am using the Z2M add-on so I believe I can't modify source files.

Montreal666, 2024-04-23:
Thanks @Koenkk , this sounds promising, command goes through but I don't get any feedback from the device.
Any advice on troubleshooting / next steps ?
Thanks again for your support.

[image]

[image]

rodgorlabach, 2024-10-22:
I also get similar issues when trying to enable the "occupancy" setting under hvacThermostat.
I have a TH1300ZB
[image]

[image]

markgrovs, 2024-11-06:
@Montreal666 Did you ever figure out how to pull the Floor Temperature? I have the same goal, but I am running into the same issue you raised.

Montreal666, 2024-11-07:
Unfortunately no. As you can see above I did all I could to implement the new parameters, thanks to @Koenkk assistance, but for some reason above my pay grade I can't pull the data. It shows in the console just to tease me but can't have it report in HA. Very unfortunate as I have a dozen of these thermostats and would really need this data to manage floor temperature while using ambient control. Open to any suggestions or troubleshooting ideas.

Montreal666, 2025-01-08:
Still unresolved

Montreal666, 2025-03-11:
Still unresolved

Koenkk, 2025-07-12:
> Thanks [@Koenkk](https://github.com/Koenkk) , this sounds promising, command goes through but I don't get any feedback from the device. Any advice on troubleshooting / next steps ? Thanks again for your support.
> 
> <img alt="image" width="1124" src="https://private-user-images.githubusercontent.com/56667580/324871679-adb96acf-e716-4785-8c4f-8c4091884409.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIzMjEyMDIsIm5iZiI6MTc1MjMyMDkwMiwicGF0aCI6Ii81NjY2NzU4MC8zMjQ4NzE2NzktYWRiOTZhY2YtZTcxNi00Nzg1LThjNGYtOGM0MDkxODg0NDA5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzEyVDExNDgyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ2ODgxZDljMTI1MTVlMTM0MWQwMWI3ZjUyZGQ0YWMzYThlN2I1NDA4MzJlODA1MmRjOGFkOGM4NmU3Mzc5YzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.KigSHDJDh5gFORXeLT5_TqPwIlgmp51UbHppf-EwBdg"> <img alt="image" width="893" src="https://private-user-images.githubusercontent.com/56667580/324871900-14a9195a-ab0b-4ba7-8076-fa5a5f6d918d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIzMjEyMDIsIm5iZiI6MTc1MjMyMDkwMiwicGF0aCI6Ii81NjY2NzU4MC8zMjQ4NzE5MDAtMTRhOTE5NWEtYWIwYi00YmE3LTgwNzYtZmE1YTVmNmQ5MThkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzEyVDExNDgyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVmZGMyZmRjNmI3OThjMWEwYzYyNGE1NDlmNGVmZGJiNzE0Nzc3YjI3MTQzZTZjNDQ4Y2E0OGZiZTYxMTI5ZDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xZC4QWHyz1Sjre1WNldcl4TOla3pxK1H36-GQWaxe0o">

Could you provide the debug log of this with the latest z2m?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

rodgorlabach, 2025-11-12:
Keep it up! I'm rooting for you!

Montreal666, 2025-11-12:
> Keep it up! I'm rooting for you!

I have just upgraded to Z2M 2.x, will run more tests shortly.

Did you have a chance to test it on your end?

arsenicks, 2026-03-17:
I just bought this thermostat. Will take a look if I can help here. Will get back

arsenicks, 2026-03-26:
Well I was working on fixing the occupancy feature so I tried to include a fix for those to report. But I think I have reached a limit of external converters.. So long story short, I can query the value in the dev console right now, thanks to your PR, but I'm unable to include them in the state reporting. 

[image]

I think the device simply doesn't support reporting for those attributes, they seems to be read-only with no reporting support. It's pretty much confirmed in the [smartthings "driver"](https://raw.githubusercontent.com/Sinopetech/Smartthings/master/Sinope%20Technologies%20TH1300ZB%20V.1.0.5.txt) and the zha quirk repo [here](https://github.com/claudegel/sinope-zha/blob/master/README.md)

So like @Koenkk said you would have to poll on demand using mqtt to get update(not so nice but better than nothing I guess). I'm really no expert but I think it's possible to trigger 'onEvent' which would fire on message events. There's other devices that seems to have done something similar Ex. ZLinky but again, I'm not that comfortable with that, I'll need to put some times to learn more and I'd like to submit a PR to fix the occupancy before.

So that will need more testing and a more complicated setup as external converters are pretty easy to test,  I'll have to take a look at how can I tests bigger changes without messing with my "production stuff"..  Maybe someone here have suggestions on where can I go from here ?

Montreal666, 2026-03-28:
Everything works up to step 4.

`[2026-03-28 17:28:16] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/extensions' with data '[{"code":"export default class SinopeTemps {\n    constructor(zigbee, mqtt, state, publishEntityState, eventBus) {\n        this.eventBus = eventBus;\n        this.publishEntityState = publishEntityState;\n    }\n\n    start() {\n        this.eventBus.onDeviceMessage(this, (data) => {\n            if (data.cluster !== 'manuSpecificSinope') return;\n            const result = {};\n            if (data.data.floorTemperature !== undefined)\n                result.floor_temperature = data.data.floorTemperature / 100;\n            if (data.data.roomTemperature !== undefined)\n                result.room_temperature = data.data.roomTemperature / 100;\n            if (Object.keys(result).length > 0) {\n                this.publishEntityState(data.device, result);\n            }\n        });\n    }\n\n    stop() {\n        this.eventBus.removeListeners(this);\n    }\n}","name":"sinope_temperature.mjs"}]'
[2026-03-28 17:28:16] info: 	z2m: Loaded external extension 'sinope_temperature.mjs'.`

Where device State doesn't return the new sensors:

`{
    "current": 0,
    "energy": 1224.88,
    "floor_control_mode": "ambiant",
    "floor_limit_status": "off",
    "floor_min_heat_setpoint": "20",
    "gfci_status": "off",
    "linkquality": 189,
    "local_temperature": 23.92,
    "occupied_heating_setpoint": 24,
    "pi_heating_demand": 0,
    "power": 0,
    "power_apparent": 432,
    "running_state": "idle",
    "system_mode": "heat",
    "voltage": 240
}`

I double checked that I still can read values in the Dev console:

[3/28/2026, 5:33:04 PM] zhc:tz: Read result of 'manuSpecificSinope': {"floorTemperature":2747,"roomTemperature":2390}

Montreal666, 2026-03-31:
Sounds great. Will test and report back asap.