Koenkk/zigbee2mqtt #21247: [Wrong device]: TS0207 water leak detector

guidokieboom, 2024-02-05:
### Link

https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector_2.html#tuya-ts0207_water_leak_detector_2

### Model

TS0207 water leak detector

### Description

water leak detector

### Vendor

Tuya / Aubess

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/IH-K665.jpg

### Database entry

can't open database.db

### Notes

[image]
[image]
I have 3 same water leak detectors. First 2 (waterleak1 and waterleak2) show the correct picture. The 3rd one (waterleak3) does not show the correct picture.

guidokieboom, 2024-02-17:
It should be: https://www.zigbee2mqtt.io/devices/IH-K665.html

Koenkk, 2024-02-18:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-18 by Koenkk

guidokieboom, 2024-02-23:
> @guidokieboom <https://github.com/guidokieboom>
> Hi,
> I found this issue you send because of seeing the wrong picture of your device.
> I have also 2 devices with this code TS0207 and _TZ3000_upgcbody.
> These are the wired ones, like the picture it was before your issue. was solved.
> So the device model TS0207 form manufacturer _TZ3000_upgcbody has multiple cases and designs.
> I just wonder if it is possible to create a fingerprint in the definitions with more info to separate these different devices and get always the right picture.
> For that I need the database.db info of your device model TS0207 from manufacturer _TZ3000_upgcbody.
> Then I can compare it with the info in my database.db, and possibly create a good fingerprint fot both devices.
>
> Are you willing to shear this info of database.db?

No problem. Could you help me where to find and how to get it to you?

marcelhoogantink, 2024-02-23:
Just open Add-on 'Studio Code Server', (install it if you didn't before)
Goto config/zigbee2mqtt map. 
Find database.db file.
Right-click and download it to your system.
Put it in a ZIP file.
Drag this zip file into your reply on github, an send the reply.

guidokieboom, 2024-02-23:
That worked. As I didn't like to share info of all my devices, I copied the 3 related lines and pasted into Excel sheet.
ID27 is the one that is discussed in this issue. The other two (20 and 21) devices gave the correct picture from the start.
Hope this helps for you.
[Book1.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/14389064/Book1.xlsx)

marcelhoogantink, 2024-02-23:
Thanks!
I looked into it. The data are almost the same, found no good difference to make a separate definition for global use.
I am going to overrule the icon locally.

Thanks for your time !!