Koenkk/zigbee2mqtt #21250: Nous A1Z does not return energy consumption and bad

espeka44, 2024-02-05:
### What happened?

I did a fresh install of Z2M on my new RPi 5 and my smartplugs (NOUS A1Z :  _TZ3000_2putqrmw) doesnt show me all parameters and others are totally wrong ("current"). E.g. my freezer use to have a 35W power on 230V but Z2M display around 230A. Impossible !
All my smartplug device, don't see the "energy" which it's the sum of consumption of the plug.

Don't remember which previous Z2M version i ran on my RPi 3B+ but all theses devices properties were available and correct

### What did you expect to happen?

Works well as my old setup

### How to reproduce it (minimal and precise)

Fresh install of the latest Z2M version (1.35.2) and paring NOUS A1Z smartplug.

### Zigbee2MQTT version

1.35.2

### Adapter firmware version

 0x26720700

### Adapter

Conbee II

### Setup

Python virtual env on RPi 5

### Debug log

Failed to configure 'clothes_dryer_smartplug', attempt 1 (Error: ConfigureReporting 0xa4c1385608c58a5c/1 haElectricalMeasurement([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":50}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (11)) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:669:28) at Object.rmsCurrent (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:216:5) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3559:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))

espeka44, 2024-02-08:
I did a fresh install and same issue when i re-pairing theses smartplugs

```
Zigbee2MQTT:error 2024-02-08 09:46:35: Failed to configure 'hair_dryer_smartplug', attempt 2 (Error: ConfigureReporting 0xa4c13877a334d878/1 haElectricalMeasurement([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (14))
    at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23)
    at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:669:28)
    at Object.rmsVoltage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:220:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3595:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))
```

Something go wrong with the latest Z2M version cause the previous ones, it worked.

Koenkk, 2024-02-15:
Try with the latest  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

espeka44, 2024-02-15:
I got a error with the procedure... not sure to understand the issue
```
spk@espikabox2:/opt/zigbee2mqtt $ git checkout dev
error: pathspec 'dev' did not match any file(s) known to git
```

Koenkk, 2024-02-18:
try: git fetch origin

espeka44, 2024-03-01:
the procedure to switch to dev branch don't work for me. maybe it will be include in the next release, reachable after a upgrade ?

Koenkk, 2024-03-02:
It's available in 1.36.0

espeka44, 2024-03-03:
Still not work...
`2024-03-03 10:29:34Failed to configure '0xa4c138fad5fa7dae', attempt 1 (Error: ConfigureReporting 0xa4c138fad5fa7dae/1 haElectricalMeasurement([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":50}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (10)) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:669:28) at Object.rmsCurrent (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:216:5) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3653:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:122:13) at Immediate.<anonymous> (/opt/zigbee2mqtt/lib/extension/configure.ts:74:17))
2024-03-03 10:29:41Failed to configure 'washing_machine_smartplug', attempt 1 (Error: ConfigureReporting 0xa4c13819cd05e870/1 haElectricalMeasurement([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":50}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (13)) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at runNextTicks (node:internal/process/task_queues:60:5) at processTimers (node:internal/timers:511:9) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:669:28) at Object.rmsCurrent (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:216:5) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3653:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))`

Same data errors returned.

espeka44, 2024-03-22:
1.36.0 don't fix my issue

schnickalodeon, 2024-03-26:
Same problem here with Zigbe2mqtt 1.36.0-1 and Nous A1Z

Guik, 2024-06-05:
Hello, using 1.37.1 same issue for me: after re-pairing 3 x A1Z, they stop reporting energy.

pscoffoni, 2024-06-25:
Hi
Same problem with 1.38.0-1

Shasdo, 2024-07-03:
Dongle SONOFF Zigbee E - Ember
MQTT 1.38.0-1

No return of Power;
OK for energy

[image]
[image]

Koenkk, 2024-07-03:
Is a power value returned when attaching a load and doing a read through the dev console?

[image]

espeka44, 2024-08-18:
Still not work, with lastest version of Z2M.
Any hope to have "energy" field back and a correct "current" measurement back ?

> error 2024-08-18 15:38:54z2m: Failed to configure 'washing_machine_smartplug', attempt 1 (Error: ZCL command 0xa4c13819cd05e870/1 haElectricalMeasurement.configReport([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":50}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (6)) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19) at runNextTicks (node:internal/process/task_queues:60:5) at processTimers (node:internal/timers:511:9) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:81:20) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:866:28) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:611:9) at Object.rmsCurrent (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:217:5) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:4983:13) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:190:21) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:126:13))

jojeju9428, 2024-09-14:
Hello,

Since the update to version 1.40.1, I no longer see the current power consumption in watts for my A1Z outlets.

The `haElectricalMeasurement activePower` command returns the information correctly, but I don't have the information in the state.

[image]

[image]

[image]

[image]

Thank you for your help.

qsilver84, 2024-11-17:
Hi,
I have this version 1.41.0 but the energy returns 0. Recently I`ve update the firmware version up to 192 version

[image]

[image]

Thanks

michi2005, 2024-12-01:
Hi all

I have the same issue- newest Z2MQTT-version and newest firmware update for Nous A1Z- no power consumption visible.
Thanks in advance for looking in to this issue.

sc-ampersand-p, 2025-01-11:
I'm also affected. Power is reported as 0 while a 2W led chain of lights is on. Current is also reported as 0.

When using a stronger consumer like a space heater, accurate values are reported.

wtom, 2025-02-02:
Firmware 192.

So I have one plug with a led lamp and it shows power 0 W.

But I also have one with a device that takes 350-360W. And the power is stated as 35-36W. So I think they might just changed the calculation?

peregrines2, 2025-02-14:
Had two devices out of 41 showing only 50% of the real power consumption (all of a sudden!). Repairing them removed the issue.
Unfortunately one quit responding alltogether and I was not able to pair it anymore.

Really dont know whats going on with myNous A1Z ...

hutasje, 2025-04-06:
**[ ISSUE SOLVED (?) ]**

Dear all, 

With regards to my reported problem above, <[link](https://github.com/Koenkk/zigbee2mqtt/issues/21250#issuecomment-2761725062)> somehow I do no longer experience any issues. 

After an update warning for Z2M, I performed the update and curious to see if this fixed my A1Z issue, ... IT DID !!  

My A1Z suddenly started to report energy and power consumption again (without any physical intervention, e.g. no re-bindinding or resetting)... just a Z2M update (see below versions).   

**_.... so ... was this just a coincidence, and did I got lucky  ?_**  

(I could not find any writing about a 'fix' for the A1Z device, but a few fixes for other NOUS devices in the SW Change Log). 

**_Old:_** 
Zigbee2MQTT version 2.1.3 commit: unknown
Coordinator type zStack3x0
Coordinator revision 20230507
Coordinator IEEE Address 0x00124b002a2f0ab4
Frontend version 0.9.4
zigbee-herdsman-converters version 23.2.0
zigbee-herdsman version 3.2.7

**_Updated:_** 
Zigbee2MQTT version [2.2.1]
Coordinator type zStack3x0
Coordinator revision 20230507
Coordinator IEEE Address 0x00124b002a2f0ab4
Frontend version 0.9.7
zigbee-herdsman-converters version 23.20.1
zigbee-herdsman version 3.4.11

espeka44, 2025-04-06:
@hutasje 

Updated for me, and still don't work...

Zigbee2MQTT version: 2.2.1commit: a2debc08
Coordinator type : ConBee2/RaspBee2
Coordinator revision : 0x26720700
Coordinator IEEE Address : 0x00.....52129
Frontend version : 0.9.7
zigbee-herdsman-converters version : 23.20.1
zigbee-herdsman version : 3.4.11

espeka44, 2025-06-09:
Issue still going. A1Z products have been downgraded for a while and no Z2M update hasn't fixed it.

AndiHaas85, 2025-08-21:
Still having trouble with A1Z. Power monitoring works for minutes then returns to 0W and stays there.